fzi-forschungszentrum-informatik/cartesian_controllers #1: Use this controller on UR5e

tonkei0361, 2020-05-14:
Hello, I am trying to use this controller on UR5e and hope to get some help setting up and loading the controller to controller_manager. Felix Exner who develops Universal Robots driver told me that this controller works well with any robot that has force-torque sensor at the TCP and UR5e does have that sensor.

```
my_cartesian_compliance_controller:
    type: "position_controllers/CartesianComplianceController"
    end_effector_link: "tool0"
    robot_base_link: "base_link"
    ft_sensor_ref_link: "sensor_link"
    compliance_ref_link: "tool0"
    target_frame_topic: "target_frame"
    joints:
    - joint1
    - joint2
    - joint3
    - joint4
    - joint5
    - joint6

    stiffness:
        trans_x: 500
        trans_y: 500
        trans_z: 500
        rot_x: 100
        rot_y: 100
        rot_z: 100

    pd_gains:
        trans_x: {p: 0.05}
        trans_y: {p: 0.05}
        trans_z: {p: 0.05}
        rot_x: {p: 0.01}
        rot_y: {p: 0.01}
        rot_z: {p: 0.01}
```

Above lines of code are from README of cartesian_controllers and I'm not sure what all links  (`end_effector_link`, `robot_base_link`, ...) and `target_frame_topic` mean. The UR driver I'm using does not seem to have those parameters, or I don't know where to find them...

Any help would be appreciated!

stefanscherzinger, 2020-05-18:
When using one of the controllers for your specific robot, you have to adjust all link variables.
For using the URe, this could be something like the following:

```yaml
my_cartesian_compliance_controller:
    type: "position_controllers/CartesianComplianceController"
    end_effector_link: "tool0"
    robot_base_link: "base_link"
    ft_sensor_ref_link: "tool0"
    joints:
    - shoulder_pan_joint
    - shoulder_lift_joint
    - elbow_joint
    - wrist_1_joint
    - wrist_2_joint
    - wrist_3_joint

```
Finding your specific configuration is normally as follows:
1) Find out your robot's joint names. For example could you echo on /joint_states or similar. Those names should be mentioned in the controller.yaml configuration under *joints* (listed from base to tip) to tell the controller which joints it may take control of.

2) Inspect a TF tree of your robot, e.g. using RViz, and decide which TF links you plan to use in your controller. *end_effector_link* is the one that will follow your commanded motion, *robot_base_link* is used internally as a reference link for some algorithms. *ft_sensor_ref_link* is the reference frame in which you expect your force data to be given.

3) Test your configuration with SMALL gains and incrementally increase the values, e.g. using dynamic reconfigure that will be available for each controller.

I'll try to make a more in-depth *Getting started* section for setting these things up.

DevwratJoshi, 2020-05-22:
Hello. My question is quite similar, so I'll ask it here. Could you please tell me what _compliance_ref_link_ and _target_frame_topic_ are referring to? These are parameters for the compliance controller.

stefanscherzinger, 2020-05-28:
The *compliance_ref_link* is a user specified link of your robot that your control should "feel" compliant about. Normally, users should set this to a urdf link somewhere close to the end-effector (it can also be identical to the end-effector). Note, that it must be part of the direct kinematic chain between *robot_base_link* and *end_effector_link*. The *target_frame_topic* is the user input to command target poses to the compliance controller. Moving the robot in compliance mode requires publishing a moving target to that topic. There's a mechanism to test this with an interactive marker in Rviz. The [cartesian_motion_controller][1] works similarly (the cartesian_compliance_controller actually inherits this functionality from the cartesian_motion_controller).

[1]: https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/master/cartesian_motion_controller#getting-started

DevwratJoshi, 2020-06-24:
Thank you. Also, could I ask if it is possible to set damping constants to influence the impedance control section like it is possible to set stiffness constants?

dal88236, 2021-11-11:
Hi @stefanscherzinger 

I followed the steps above and got the following error messages when loading my_cartesian_compliance_controller:

```
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 212, in <module>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 212, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 204, in main
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 190, in main
    resp = switch_controller(loaded, [], 2, False, 0.0)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
resp = load_controller(name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    return self.call(*args, **kwds)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 532, in call
return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 532, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: rospy.servicetransport error completing service call: unable to receive data from sender, check sender's logs for details.
```
Do you know what the cause of issue might be? Thanks.

stefanscherzinger, 2021-11-11:
Hi @dal88236 

> Do you know what the cause of issue might be?

Not from what you send. Could you post the _complete_ output?

dal88236, 2021-11-11:
Hi @stefanscherzinger 

The complete output is:
```
[INFO] [1636647129.793154]: Loading controller: my_cartesian_compliance_controller
WARN /tmp/binarydeb/ros-melodic-ur-client-library-0.3.2/src/control/reverse_interface.cpp 142: Message on ReverseInterface received. The reverse interface currently does not support any message handling. This message will be ignored. 
WARN /tmp/binarydeb/ros-melodic-ur-client-library-0.3.2/src/control/reverse_interface.cpp 142: Message on ReverseInterface received. The reverse interface currently does not support any message handling. This message will be ignored. 
[INFO] [1636647129.797283]: Loading controller: scaled_pos_joint_traj_controller
WARN /tmp/binarydeb/ros-melodic-ur-client-library-0.3.2/src/control/reverse_interface.cpp 142: Message on ReverseInterface received. The reverse interface currently does not support any message handling. This message will be ignored. 
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 212, in <module>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 212, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 190, in main
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 190, in main
    resp = load_controller(name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    resp = load_controller(name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    return self.call(*args, **kwds)
      File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 532, in call
return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 532, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceExceptionrospy.service: .transport error completing service call: unable to receive data from sender, check sender's logs for detailsServiceException
: transport error completing service call: unable to receive data from sender, check sender's logs for details
[INFO] [1636647129.882820]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1636647129.882820]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1636647129.884357]: Stopping all controllers...
[INFO] [1636647129.884540]: Stopping all controllers...
[WARN] [1636647129.887189]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 111] Connection refused
[WARN] [1636647129.887690]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 111] Connection refused
================================================================================REQUIRED process [ur_hardware_interface-3] has died!
process has died [pid 14455, exit code -11, cmd /home/user/Desktop/UIUC_HRC/ur5_ws-master/devel/lib/ur_robot_driver/ur_robot_driver_node __name:=ur_hardware_interface __log:=/home/user/.ros/log/1d547c0e-430a-11ec-84e9-b0a4605b8227/ur_hardware_interface-3.log].
log file: /home/user/.ros/log/1d547c0e-430a-11ec-84e9-b0a4605b8227/ur_hardware_interface-3*.log
Initiating shutdown!
================================================================================
[tf2_parts_broadcaster-17] killing on exit
[status_listener-15] killing on exit
[joint_currents_talker-16] killing on exit
[ur_kinematics-14] killing on exit
[free_drive-11] killing on exit
[connect-10] killing on exit
[move_group-9] killing on exit
[controller_stopper-7] killing on exit
[ros_control_stopped_spawner-6] killing on exit
[ur_hardware_interface/ur_robot_state_helper-8] killing on exit
[ros_control_controller_spawner-5] killing on exit
[ur_hardware_interface-3] killing on exit
[robot_state_publisher-2] killing on exit
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...

You can start planning now!

[ WARN] [1636647134.386367863]: service '/get_planning_scene' not advertised yet. Continue waiting...
[ INFO] [1636647134.387518859]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ur_kinematics-14] escalating to SIGTERM
[controller_stopper-7] escalating to SIGTERM
[ur_kinematics-14] escalating to SIGKILL
[rosout-1] killing on exit
[master] killing on exit
Shutdown errors:
 * process[ur_kinematics-14, pid 14489]: required SIGKILL. May still be running.
shutting down processing monitor...
... shutting down processing monitor complete
done
```

stefanscherzinger, 2021-11-12:
@dal88236 

I can't figure out from the output what the problem might be.
Let's go one step back and load the URe in its default configuration without any of the _cartesian_ controllers.

I would suggest the following:
1) Load your UR-setup without any of the _cartesian_ controllers
2) In a different launch file, load only the Cartesian controllers configuration to the parameter server. Don't spawn them yet.
   Something similar to
   ```bash
        <!-- Load ROS controller configurations to parameter server -->
        <rosparam file="$(find your_package)/config/cartesian_controllers.yaml" command="load"/>

   ```
   I'm assuming that `cartesian_controllers.yaml` contains your parameterization of the `my_cartesian_compliance_controller`. You might need to put that into the *correct* namespace of the controller manager. When your stack of 1) runs normally, you can find that with `rosservice list | grep manager` and check the namespace. Then load your controllers.yaml in that namespace.
3) In rqt, load the `my_cartesian_compliance_controller`. It should be available, if loaded in the right namespace.
4) If that works, stop the other controller that currently claim the joints and
5) Start the `my_cartesian_compliance_controller`.

That will get us more specific error outputs.

stefanscherzinger, 2021-11-27:
@dal88236 Did you find some time to test this?

stefanscherzinger, 2022-01-13:
@dal88236 Feel free to reopen if this is still relevant.