fzi-forschungszentrum-informatik/cartesian_controllers #2: Using this controller with gazebo

DevwratJoshi, 2020-06-08:
Hello. I have been trying to use this controller in the gazebo simulator with gravity, trying in essence to recreate in simulation the experiment where an object is moved with a sliding motion along a surface while maintaining a constant force against the surface. I start this controller when the object is touching the surface. Initially, the force is ramped up to the required quantity. This is followed by motion in a direction orthogonal to the axis along which force is applied, trying to keep the force constant. However, when run the program, there is a large overshoot in the force during force ramping.

[image]

Is there something different that must be done when using this controller with gravity? How should I control the prevent this overshoot?

stefanscherzinger, 2020-06-15:
When working on real hardware, if you don't want the controller to react to the gravity forces of your grasped objects, then you must bias the sensor accordingly. I'm not familiar with how this works in Gazebo, though. Apart from that, the controllers command either joint positions or joint velocities, which are realized in underlying control loops in Gazebo. They should work both in gravity and without gravity. The overshoot you are referring to looks like loosing stability. Could you post your configuration for the cartesian_compliance_controller?

DevwratJoshi, 2020-06-23:
The configuration looks like this: 

 cartesian_compliance_controller:
    type: "position_controllers/CartesianComplianceController"
    end_effector_link: "robotiq_85_tip_link" 
    robot_base_link: "base_link"
    ft_sensor_ref_link: "wrist_3_link"
    compliance_ref_link: "ee_link"
    target_frame_topic: "target_frame"
    joints:
                - shoulder_pan_joint
                - shoulder_lift_joint
                - elbow_joint
                - wrist_1_joint
                - wrist_2_joint
                - wrist_3_joint 
 stiffness:  # w.r.t. compliance_ref_link
        trans_x: 500
        trans_y: 500
        trans_z: 500
        rot_x: 100
        rot_y: 100
        rot_z: 100
    
pd_gains:
        trans_x: {p: 0.05}
        trans_y: {p: 0.05}
        trans_z: {p: 0.05}
        rot_x: {p: 0.01}
        rot_y: {p: 0.01}
        rot_z: {p: 0.01}

yangxy415, 2020-07-14:
Hello,my question is similar. I am trying to use the force controller on the realhardware UR10e, but there is also a overshot in the force (ft_sensor_wrench). And in the Gazebo without gravity can the robot implement normaly.  And I use the built_in ft_sensor to measure the force in the TCP on the real UR10e. So how to bias the sensor accordingly? or there may be other solutions?

stefanscherzinger, 2020-07-15:
There's a special service to bias the sensor, offered by the UR driver. It's explained [here][1].
You might want to rostopic echo on the sensor message from URe to check if zeroing was successful.

Especially with stiff environments, such as workbenches etc., force control in general has the disadvantages of overshooting. That's closely related to how fast the inner control loops. This is an inherent problem of position-based (or velocity-based) force control and cannot be avoided completely. However, when moving with moderate velocity, those overshoots should not exceed certain ranges and the robot should stay stable. The URe series is much faster than the UR series in this aspect, but for the precise moment of making contact with the environment, most force control loops will still be too slow to detect the impact and react fast enough to prevent overshooting. The control gains that @DevwratJoshi shared seem reasonable for testing. What I personally always do (for these robots) is check how the robot moves when I guide it manually in free space. This helps me to get a feeling how responsive the system behaves for the control parameters.

Also, the links *end_effector_link*, *robot_base_link*, *ft_sensor_ref_link* and *compliance_ref_link* must be part of the same kinematic chain. If that's not the case, then the force control may misbehave. I normally spot errors with this config when manually guiding the robot.

[1]: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#zero_ftsensor-std_srvstrigger

Csp0920, 2020-11-02:
Hello. I have a problem when using my_cartesian_force_controller in gazebo. I have successfully spawned this controller. But when I publish a geometry.msgs/WrenchStamped to the topic /my_cartesian_force_controler/target_wrench, there is no reaction in gazebo. 
I don't know what the problem is. Could you give me a hand.

Csp0920, 2020-11-03:
I have solved this problem. Thank you.

gavanderhoorn, 2020-11-03:
It would be very much appreciated if you could tell us *how* you've "solved this problem".

Csp0920, 2020-11-04:
It is mainly because of my carelessness.
I loaded the controller but I didn't spawn it.
And I forgot to copy the pid config code to the controller.yaml...

yaogengcheng, 2021-03-23:
Hello, I also want to simulate in gazebo and  I added the following codes in cartesian_controller_examples/example.launch.
however, it happens an error "ERROR: Could not start controller with name my_cartesian_motion_controller because no controller with this name exists" [WARN] [1616504764.092994, 23.922000]: Controller Spawner couldn't find the expected controller_manager ROS interface. Can you tell me what should I do? Thank you

The added code:
<include file="$(find gazebo_ros)/launch/empty_world.launch">
      <arg name="paused" value="false" />
      <arg name="debug" value="0" />
      <arg name="gui" value="true" />
      <arg name="use_sim_time" value="true" />
      <arg name="headless" value="false" />
      <arg name="world_name" value="worlds/empty.world" />
    </include>

      <param name="robot_description" command="$(find xacro)/xacro --inorder '$(find ur_description)/urdf/ur3_robot.urdf.xacro'" /> 
        <node pkg="rostopic" type="rostopic" name="fake_joint_calibration"
        args="pub /calibrated std_msgs/Bool true" />

       <node name="urdf_spawner" pkg="gazebo_ros" type="spawn_model"
        args="-urdf -model ur3 -param robot_description -x 0 -y 0 -z 0" /> -->

yaogengcheng, 2021-03-23:
Hello, I also want to simulate in gazebo and  I added the following codes in cartesian_controller_examples/example.launch.
however, it happens an error "ERROR: Could not start controller with name my_cartesian_motion_controller because no controller with this name exists" [WARN] [1616504764.092994, 23.922000]: Controller Spawner couldn't find the expected controller_manager ROS interface. Can you tell me what should I do? Thank you
The added code:
`<include file="$(find gazebo_ros)/launch/empty_world.launch">
      <arg name="paused" value="false" />
      <arg name="debug" value="0" />
      <arg name="gui" value="true" />
      <arg name="use_sim_time" value="true" />
      <arg name="headless" value="false" />
      <arg name="world_name" value="worlds/empty.world" />
    </include>`
`      <param name="robot_description" command="$(find xacro)/xacro --inorder '$(find ur_description)/urdf/ur3_robot.urdf.xacro'" />
        <node pkg="rostopic" type="rostopic" name="fake_joint_calibration"
        args="pub /calibrated std_msgs/Bool true" />
       <node name="urdf_spawner" pkg="gazebo_ros" type="spawn_model"
        args="-urdf -model ur3 -param robot_description -x 0 -y 0 -z 0" /> -->`

closed 2021-11-15 by stefanscherzinger

ZZWang21, 2023-07-13:
> I have solved this problem. Thank you.

Hi, how did you solve it, could you copy your controller.yaml here? Thanks.