fzi-forschungszentrum-informatik/cartesian_controllers #3: use cartesian controller on UR10e

yangxy415, 2020-06-25:
Hello,I have a question. I use the cartesian force controller both in Gazebo and on real UR10e. I publish a target wrench(eg. 5N)on the topic and when the ft_sensor is equal to the target_wrench(eg.-5N), the robot will stop in the simulation Gazebo. It can implement the function. But on the real robot, it will not stop and always add the force until the robot have a protective stop. I can not find the reason about it,.

stefanscherzinger, 2020-07-15:
Can you confirm that the real robot yields forces correctly when pushing it manually? You should be able to manually guide the robot in free space. If it does not react to your pushing, then we have a first start for finding what's wrong.

Csp0920, 2020-09-15:
Hello, I have a same problem. When I pub a target wrench(1N and 0.05N) on the topic,  the real robot will not stop when I try to stop it with my hand.

Csp0920, 2020-09-15:
By the way, if the ft_sensor should be add into the robot.urdf?
Thank you very much.

stefanscherzinger, 2020-11-05:
Yes, for the robot.urdf of your specific use case you need to add a link for a force-torque sensor. You must then specify this link as **ft_sensor_ref_link** in the according .yaml configuration of the controller somewhere in between base and end-effector. The reason is that all Cartesian controllers parse that urdf between the specified **robot_base_link** and **end_effector_link** and build kinematic transformations, so that measurements from the sensor are transformed correctly. Also check that the measurements from this sensor are connected to your controller.
```bash
rostopic list | grep ft_sensor_wrench
```
should list the right input. You might need to remap some topics to achieve this.

Csp0920, 2020-11-05:
> 是的，对于特定用例的robot.urdf，您需要添加一个力扭矩传感器的链接。然后，您必须在控制器的.yaml配置中，将此链接指定为**ft_sensor_ref_link**，位于基础执行器和最终执行器之间。原因是所有笛卡尔控制器都会在指定的**robot_base_link**和end_effector_link之间解析**urdf**并建立运动学转换，以便正确地转换传感器的测量值。还要检查该传感器的测量值是否已连接到控制器。
> 
> ```shell
> rostopic列表| grep ft_sensor_wrench
> ```
> 
> 应该列出正确的输入。您可能需要重新映射一些主题才能实现此目的。

Thank you for your reply.

Csp0920, 2020-11-05:
> 是的，对于特定用例的robot.urdf，您需要添加一个力扭矩传感器的链接。然后，您必须在控制器的.yaml配置中，将此链接指定为**ft_sensor_ref_link**，位于基础执行器和最终执行器之间。原因是所有笛卡尔控制器都会在指定的**robot_base_link**和end_effector_link之间解析**urdf**并建立运动学转换，以便正确地转换传感器的测量值。还要检查该传感器的测量值是否已连接到控制器。
> 
> ```shell
> rostopic列表| grep ft_sensor_wrench
> ```
> 
> 应该列出正确的输入。您可能需要重新映射一些主题才能实现此目的。

I find that there is a problem in force torque sensor in the ur3e, so I can't use it in the real robot.
Now I am trying to use it in gazebo, but also get a wrong data in sim force torque sensor.

closed 2021-09-24 by stefanscherzinger