fzi-forschungszentrum-informatik/cartesian_controllers #4: Integration with MoveIt

madhan001, 2020-08-26:
Is it possible to integrate the compliance controller with moveit for use with a UR10 ? If so, how?

stefanscherzinger, 2020-10-22:
(Sorry for the late reply)
It is kind of possible, but this comes at some expense.
You could use [this controller adapter](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/master/joint_to_cartesian_controller) from this package. That's a mechanism to take conventional joint-based trajectories and turn them into a moving Pose in Cartesian space. You could then track this pose with e.g. the CartesianComplianceController.

However, be aware that you cannot control the joints during execution. The Cartesian working controllers might "find" different joint constellations than you expect.

madhan001, 2020-10-22:
Thank you so much for your reply. Would the ur10's built in ft sensor offer satisfactory performance for this controller or is a dedicated load cell absolutely necessary ? could the moveit controller adapter work with the cartesian planner inside moveit ? I want to use this controller to make the end effector follow a 6D cartesian path while making contact with the surface, the paths are dynamically generated from a computer vision program in Python.

stefanscherzinger, 2020-11-05:
Concerning your first question:
Interesting idea. As far as I remember, those force-torque signals are based on a calibrated estimation using motor currents. They are surely not as accurate as the measurements from a force-torque sensor, but you could give it a try. Just make sure to rename the topic accordingly to connect it to the CartesianComplianceController. Also, you should find out the frame in which they are given and specify this in the according .yaml controller configuration. This frame must be part of the kinematic chain from base to tip. I'm not sure how a base_frame is handled, though. I'm looking forward to your feedback.

Concerning your second question:
Right now I'm not too familiar with the Cartesian planner in Moveit. But you mention dynamically generated in Python: This should work without Moveit, right? If you manage to stream PoseStamped messages from these paths, then you can follow those with the CartesianComplianceController.

madhan001, 2020-11-05:
I was able to get the controller load up and I published a target wrench value and the robot moved downwards and this movement is faster when the stiffness is set to a low value. Can you please tell me how to get the end effector move on a surface while not applying a force above a certain threshold, I have a PoseArray object with a series of 6D poses that need to be followed while being in contact with the surface. Also regarding MoveIt usage, I need it to enforce a "safe" workcell for the robot. Can you please help me out with this ? I'm new to ROS.

closed 2020-12-19 by madhan001

ZZWang21, 2022-10-12:
> I was able to get the controller load up and I published a target wrench value and the robot moved downwards and this movement is faster when the stiffness is set to a low value. Can you please tell me how to get the end effector move on a surface while not applying a force above a certain threshold, I have a PoseArray object with a series of 6D poses that need to be followed while being in contact with the surface. Also regarding MoveIt usage, I need it to enforce a "safe" workcell for the robot. Can you please help me out with this ? I'm new to ROS.
> 
> Edit: I was able to figure out the way it gets target poses and piped the force value from the built in force sensor of the UR10 into the compliance controller, the result was really bad with the end effector bobbing around and being generally unstable but, what seems to me a mystery is that the built in force mode works much better than this one despite using the same hardware . The robot could smoothly follow the points being published with the motion controller though.
> 
> **Update**: Got access to a ft sensor, the instability is gone and works fine now. Thank you so much for this controller, it works really well for my use case

Excuse me @madhan001 ,  did you use compliance_controller , or "cartesian_motion_controller"? Thank you.