fzi-forschungszentrum-informatik/cartesian_controllers #6: How to connect ft sensor to the CartesianForceController

AbdElRahmanFarhan, 2020-12-16:
I have an ati force torque sensor that publishes Wrench messages on a certain topic. My question is. how to let CartesianForceController take that feedback from the sensor to close the control loop.

madhan001, 2020-12-16:
You need to do a remap from the sensor's wrench topic to the controller's
wrench topic inside the top level launch file of your robot. Use rostopic
list | grep wrench to see all the wrench topics.

AbdElRahmanFarhan, 2020-12-16:
Thank you for your response. But, what is the name of the controller wrench topic?

AbdElRahmanFarhan, 2020-12-16:
Thanks I have figured out what is the problem. The ft topic is /your_controller_name/ft_sensor_wrench.

madhan001, 2020-12-19:
You're welcome, you may close the issue if resolved

AbdElRahmanFarhan, 2020-12-21:
Solved

closed 2020-12-21 by AbdElRahmanFarhan