fzi-forschungszentrum-informatik/cartesian_controllers #8: Problem in using "my_cartesian_compliance_control".

Csp0920, 2020-12-22:
Firstly, I add the controller config to my yaml:
`my_cartesian_compliance_controller:
    type: "position_controllers/CartesianComplianceController"
    end_effector_link: "tool0"    # All links below must come before this link
    robot_base_link: "base_link"
    ft_sensor_ref_link: "wrist_3_link"
    compliance_ref_link: "tool0"
    target_frame_topic: "tool0"
    joints:
     - shoulder_pan_joint
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint

    stiffness:  # w.r.t. compliance_ref_link
        trans_x: 500
        trans_y: 500
        trans_z: 500
        rot_x: 100
        rot_y: 100
        rot_z: 100

    pd_gains:
        trans_x: {p: 0.03}
        trans_y: {p: 0.03}
        trans_z: {p: 0.05}
        rot_x: {p: 0.01}
        rot_y: {p: 0.01}
        rot_z: {p: 0.01}`

Then I spawn this controller, and pub a msg to "my_cartesian_compliance_controller/tool" but I get a WARN:
`[ WARN] [1608607333.332922936]: Got target pose in wrong reference frame. Expected: base_link but got 
`
And my robot will not move.
I have no idea what cause this, could you help me?
Thanks.

stefanscherzinger, 2020-12-22:
Two things:
1) That warning says that the pose message that you are commanding is not given in the right reference frame. All poses must be given in the robot_base_link, which you parameterized with "base_link". Try setting the header's frame_id of your command message accordingly. But be careful, your pose must really be given in the coordinate axes of that reference. If that's not the case, then you must transfer your target before commanding this to the controller. Getting a pose from some camera and just forwarding this will not work (it's not a TF).

2) Setting ft_sensor_ref_link with "wrist_3_link" seems strange. What sensor are you using? This sensor will have its own reference frame somewhere in your robot's urdf. You should pick that one, else the sensor's measurements will not be correctly transformed.

Csp0920, 2020-12-26:
Yes, after I change the frame_id, it can move.
And here is another question. 
If I can change the moving speed when I publish a position/orientation?  I publish a position/orientation, I find the robot arm move too fast.

Csp0920, 2020-12-26:
I modify the iteration, and I found the robot arm shaking a lot.

Csp0920, 2020-12-28:
> Two things:
>
> 1. That warning says that the pose message that you are commanding is not given in the right reference frame. All poses must be given in the robot_base_link, which you parameterized [...]

Because I am using the UR3e robot, the force/torque sensor is integrated in last link. I set the ft_sensor_ref_link to "wrist_3_link".

stefanscherzinger, 2021-01-14:
@Csp0920 :
> I modify the iteration, and I found the robot arm shaking a lot.

Note the philosophy behind the motion controllers: They will directly drive to the target frames. A higher _error_scale_ makes them drive faster in each control cycle and higher  _iterations_  give them more (internal) control cycles to reach that target. By setting both to high values, you continuously make this controller an Inverse Kinematics solver, i.e. it computes the joint configuration for your target only, not in between. In order to not shake too much, make sure to interpolate your targets at a decent rate. I would recommend publishing target frames in the neighborhood of 100 Hz for this configuration. If the controller is set to behave "smooth", a lower target rate of few Hz might be sufficient.

stefanscherzinger, 2021-01-14:
> Because I am using the UR3e robot, the force/torque sensor is integrated in last link. I set the ft_sensor_ref_link to "wrist_3_link".

From experience, I would set the sensor to tool0 in that case. The axis of wrist_3_link are rotated with respect to the end-effector flange and will not be aligned with the sensor's axes. Force control will behave unexpected if you ignore this.

Csp0920, 2021-01-15:
> @Csp0920 :
>
>
> Note the philosophy behind the motion controllers: They will directly drive to the target frames. A higher _error_scale_ makes them drive faster in each control cycle and higher [...]

1. Thanks very much for your reply. I have tried to change the iteration and error_scale to see the different motion result. Changing the error_scale, the robot will move faster. Changing the iteration, robot will need more time to calculate and move more slowly. Is it correct?
2. And you say that I should set the sensor to tool0` (Tip link? I am sorry that I am not clear the difference between ee_link and tool0 in the ur robot.urdf)` . If there is a bias between force sensor link and tool0, the force sensor will get a torque, but in this situation I may believe that the torque is added in tool0? 

3. Sorry that I am a beginner in ros adn robot study. I used to use moveit to get the information of my robot. But it seems that But you can't use moveit with your controller. I read your code carefully, and I want to know how to use the api in your controller  code, such as `const KDL::Frame& getEndEffectorPose() const;`.

I'm sorry I have too many questions. And thank you for your guidance.