fzi-forschungszentrum-informatik/cartesian_controllers #9: Use cartesian_motion_controller to reach target_frame with UR5

FrederikPlahl, 2021-01-11:
Hey everyone,

I'm trying to use the cartesian_motion_controller and the Universal_Robots_ROS_Driver to reach a target frame. I try to launch the controller with the launch file:

```
<launch>
        <!--Load robot_description to parameter server-->
        <param name="/robot_description"
        command="$(find xacro)/xacro --inorder '$(find ur5_controller)/urdf/ur5.urdf.xacro'" />

        <!--Load hardware configuration-->
        <rosparam file="$(find ur5_controller)/config/ur5_hw_config.yaml" command="load"></rosparam>

        <!--Load hardware interface
        <node name="sim_hardware_interface" pkg="ros_control_boilerplate" type="sim_hw_main"
        output="screen" /> 
        I don't know which hardware interface to use with the UR5 or where to find it-->

        <!-- Robot state publisher -->
        <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher">
                <param name="publish_frequency" type="double" value="50.0" />
                <param name="tf_prefix" type="string" value="" />
        </node> 

        <!-- Joint state publisher -->
        <node name="joint_state_controller_spawner" pkg="controller_manager" type="spawner" args="joint_state_controller" respawn="false" output="screen"/>

        <!-- Load controller configuration -->
        <rosparam file="$(find ur5_controller)/config/ur5_controllers.yaml" command="load"></rosparam>

        <!-- Spawn controllers -->
        <node name="controller_spawner" pkg="controller_manager" type="spawner"
                args="--stopped
                my_cartesian_motion_controller
                my_motion_control_handle
                my_joint_to_cartesian_controller
                joint_trajectory_controller
                "
        />

        <!-- Spawn joint controller for testing my_joint_to_cartesian_controller  -->
        <!-- Note: This controller should be loaded and started -->
        <group ns="my_joint_to_cartesian_controller" >
                <node name="controller_spawner" pkg="controller_manager" type="spawner" args="joint_trajectory_controller" />
        </group>

        <!-- Visualization -->
        <arg name="my_rviz" value="-d $(find ur5_controller)/etc/ur5.rviz" />
        <node name="rviz" pkg="rviz" type="rviz" respawn="false" args="$(arg my_rviz)" output="screen" >
        </node>
</launch>
```
For the hardware Interface I have no idea which to use and where to find that. So for now I commented it out
Launching the file gives me following errors:

```
plfr1011@w-robolab-p11:~/ros_workspace$ roslaunch ur5_controller ur5_controller.launch 
... logging to /home/plfr1011/.ros/log/eb1c18ba-540c-11eb-b160-00032d460848/roslaunch-w-robolab-p11-20115.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://10.166.32.166:42275/

SUMMARY
========

PARAMETERS
 * /generic_hw_control_loop/cycle_time_error_threshold: 0.03
[... 81 lines cut ...]
[robot_state_publisher-1] process has died [pid 20128, exit code 1, cmd /opt/ros/melodic/lib/robot_state_publisher/robot_state_publisher __name:=robot_state_publisher __log:=/home/plfr1011/.ros/log/eb1c18ba-540c-11eb-b160-00032d460848/robot_state_publisher-1.log].
log file: /home/plfr1011/.ros/log/eb1c18ba-540c-11eb-b160-00032d460848/robot_state_publisher-1*.log
[ INFO] [1610381591.405189378]: Stereo is NOT SUPPORTED
[ INFO] [1610381591.405242041]: OpenGl version: 3 (GLSL 1.3).
[INFO] [1610381591.457935]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1610381591.470110]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1610381591.481615]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1610381591.490804]: Loading controller: joint_state_controller
[ERROR] [1610381591.501797067]: No name given for the robot.
[ERROR] [1610381592.507898]: Failed to load joint_state_controller
[INFO] [1610381592.515561]: Controller Spawner: Loaded controllers: 
[ERROR] [1610381592.526647]: Failed to load my_cartesian_motion_controller
[INFO] [1610381592.536981]: Started controllers: 
[ERROR] [1610381593.550563]: Failed to load my_motion_control_handle
[ERROR] [1610381594.582498]: Failed to load my_joint_to_cartesian_controller
[ERROR] [1610381595.611245]: Failed to load joint_trajectory_controller
[WARN] [1610381621.585879]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[my_joint_to_cartesian_controller/controller_spawner-4] process has finished cleanly
log file: /home/plfr1011/.ros/log/eb1c18ba-540c-11eb-b160-00032d460848/my_joint_to_cartesian_controller-controller_spawner-4*.log
```

I'm not really sure how to correctly use these controllers...

The final goal would be the TCP of my robot holding the target_frame position while I'm moving the mobile platform on which the robot is mounted on.
Currently I'm only able to move the robot based on joint values I send via a client to the /scaled_pos_joint_traj_controller/follow_joint_trajectory action.

FrederikPlahl, 2021-01-14:
Okay I made it to solve some of the errors. I took the wrong urdf file by accident. 

Now I still have the problem that my controllers will not load ```because controller type does not exist.```

```
[ERROR] [1610640493.583018763]: Could not load controller 'my_cartesian_motion_controller' because controller type 'position_controllers/CartesianMotionController' does not exist.
[ERROR] [1610640493.583077701]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1610640494.610224942]: Could not load controller 'my_motion_control_handle' because controller type 'cartesian_controllers/MotionControlHandle' does not exist.
[ERROR] [1610640494.610275677]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1610640495.638195301]: Could not load controller 'my_joint_to_cartesian_controller' because controller type 'cartesian_controllers/JointControllerAdapter' does not exist.
[ERROR] [1610640495.638241208]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
```
 
Using ```rosservice call contoller_manager/list_controller_types``` gives me:
```
neobotix@mmo-500-1:~$ rosservice call controller_manager/list_controller_types
types: [diff_drive_controller/DiffDriveController, effort_controllers/JointEffortController,
  effort_controllers/JointGroupEffortController, effort_controllers/JointGroupPositionController,
  effort_controllers/JointPositionController, effort_controllers/JointTrajectoryController,
  effort_controllers/JointVelocityController, force_torque_sensor_controller/ForceTorqueSensorController,
  industrial_robot_status_controller/IndustrialRobotStatusController, joint_state_controller/JointStateController,
  pos_vel_acc_controllers/JointTrajectoryController, pos_vel_controllers/JointTrajectoryController,
  position_controllers/JointGroupPositionController, position_controllers/JointPositionController,
  position_controllers/JointTrajectoryController, position_controllers/ScaledJointTrajectoryController,
  ur_controllers/SpeedScalingStateController, velocity_controllers/JointGroupVelocityController,
  velocity_controllers/JointPositionController, velocity_controllers/JointTrajectoryController,
  velocity_controllers/JointVelocityController, velocity_controllers/ScaledJointTrajectoryController]
base_classes: ['controller_interface::ControllerBase', 'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase']
```
@stefanscherzinger or @fmauch could you help me with that?

stefanscherzinger, 2021-01-14:
Hi @FrederikPlahl, using ROS controllers is normally quite simple. Don't launch the _examples.launch_ file from this package for your use case. That is just to get you started in simulation. For your use case, it's expected that you put similar controller configuration into the respective launch files of your application. For simplicity, I would recommend that you change two files of the UR driver package directly:

1)  In _Universal_Robots_ROS_Driver/ur_robot_driver/config/ur5_controllers.yaml_ append the following lines:

``` yaml
# A joint-based Cartesian motion controller.
my_cartesian_motion_controller:
    type: "position_controllers/CartesianMotionController"

    # Needed for Inverse Kinematics
    robot_base_link: "base_link"
    end_effector_link: "tool0"
    joints: *robot_joints
    solver:
        error_scale: 10
        iterations: 10
    pd_gains:
        trans_x: {p: 10.0}
        trans_y: {p: 10.0}
        trans_z: {p: 10.0}
        rot_x: {p: 1.0}
        rot_y: {p: 1.0}
        rot_z: {p: 1.0}
```
 
That should make this instance of the CartesianMotionController "known" to your setup. Of course you can name this instance differently, _my_cartesian_motion_controller_ is just a default. Once that's done, you can load, start, stop and unload this controller at runtime via the respective controller manager. I personally find the rqt plugin handy for this during initial testing. Note the [different states](http://wiki.ros.org/controller_manager) a controller can go through.

2) It's handy (albeit not required) to load this controller already at startup. You can decide whether it should load and start directly or load in a stopped state. I'll recommend the stopped state for now. You would then replace the line that says _stopped_controllers_  in _Universal_Robots_ROS_Driver/ur_robot_driver/launch/ur5_bringup.launch_  with:
```yaml
  <arg name="stopped_controllers" default="my_cartesian_motion_controller pos_joint_traj_controller joint_group_vel_controller" doc="Controllers that are initally loaded, but not started."/>

```

That should spawn your controller. You can then start and use it.

To get a better feeling of how things work, it might be better to go through [some basics](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/master/cartesian_motion_controller#getting-started) in simulation first.

FrederikPlahl, 2021-01-18:
Thanks for the help @stefanscherzinger!
Now I am able to move the robot with the cartesian_motion controller. The problem is that it doesn't move where I'd expect it to go. I think it could have something to do with the fact that the robot is mounted on a mobile platform and the tf frames are called differently (```base``` instead of ```base_link``` while ```base_link``` is somewhere else) as you can see on the picture.

[image]

When I run a program that should send the TCP 10cm up in z-direction from the current pose (not the one in the picture) with respect to the ```base``` frame it says:
```[ WARN] [1610967574.302286566]: Got target pose in wrong reference frame. Expected: base_link but got base```

So I changed the header frame to base_link. Now the robot is moving but in a completely different direction than it is supposed to do.

Here is the publisher of the target_frame which I'm sending:

```c++
#include <ros/ros.h>
#include <geometry_msgs/PoseStamped.h>
#include <tf2/LinearMath/Quaternion.h>

int main(int argc, char** argv){
  ros::init(argc, argv, "target_pub");
  ros::NodeHandle node;
  ros::Publisher pub = node.advertise<geometry_msgs::PoseStamped>("/ur5_cartesian_motion_controller/target_frame", 10);

  ros::Rate rate(50.0);

  geometry_msgs::PoseStamped poseStamped;
  tf2::Quaternion q;

  poseStamped.header.stamp = ros::Time::now();
  poseStamped.header.frame_id = "base_link";

  poseStamped.pose.position.x = 0;
  poseStamped.pose.position.y = -400;
  poseStamped.pose.position.z = 600;

  q.setRPY(180, 0, 0);
  poseStamped.pose.orientation.x = q.getX();
  poseStamped.pose.orientation.y = q.getY();
  poseStamped.pose.orientation.z = q.getZ();
  poseStamped.pose.orientation.w = q.getW();

  while (node.ok()){  
        pub.publish(poseStamped);
        rate.sleep();
    }
  
  return 0;
}
```
Do you know what I am doing wrong?

stefanscherzinger, 2021-01-18:
@FrederikPlahl just two things:
In the _..controllers.yaml_ change the _robot_base_link_ so that it reflects your setup:
```yaml
    # Needed for Inverse Kinematics
    robot_base_link: "base"
```
The controller will then expect its targets in that frame.
And in your publisher, adapt the _frame_id_  to comply with the controller's safety check:
```cpp
poseStamped.header.frame_id = "base";
```

Note that you are firing up one single point and ask the controller to go there. If the controller is parameterized to be an Inverse Kinematics solver (high gains, solver/iterations = big number, solver/error_scale = big number) your robot will respond with a jump, because it will only command the final joint state. If you want more trajectory control, try publishing a "moving target pose" from start pose to target pose (as in sampled with approx. 50 Hz).

FrederikPlahl, 2021-01-18:
Thanks! Actually, that was a quite dumb mistake since I did try this in the _..controllers.yaml_ at first but I must have messed something up by fixing that. Anyway no it is working so thanks for helping me out and for the tips!

closed 2021-01-18 by FrederikPlahl