fzi-forschungszentrum-informatik/cartesian_controllers #12: Control of multiple robots

ManueU, 2021-02-15:
Hi, 
I am working with two UR5. 
Master controller.yaml: 

```
# Settings for ros_control control loop
hardware_control_loop:
   loop_hz: &loop_hz 125

# Settings for ros_control hardware interface
ur_hardware_interface:
   joints: &robot_joints
     - master_shoulder_pan_joint
     - master_shoulder_lift_joint
     - master_elbow_joint
     - master_wrist_1_joint
     - master_wrist_2_joint
     - master_wrist_3_joint

# Publish all joint states ----------------------------------
joint_state_controller:
   type: joint_state_controller/JointStateController
   publish_rate: *loop_hz

```

Slave controller.yaml: 

```
# Settings for ros_control control loop
hardware_control_loop:
   loop_hz: &loop_hz 125

 
# Settings for ros_control hardware interface
ur_hardware_interface:
   joints: &robot_joints
     - slave_shoulder_pan_joint
     - slave_shoulder_lift_joint
     - slave_elbow_joint
     - slave_wrist_1_joint
     - slave_wrist_2_joint
     - slave_wrist_3_joint

# Publish all joint states ----------------------------------
joint_state_controller: 
   type: joint_state_controller/JointStateController
   publish_rate: *loop_hz

# Cartesian motion controller   
my_cartesian_motion_controller:
    type: position_controllers/CartesianMotionController
    end_effector_link: "tool0"
    robot_base_link: "base"
    target_frame_topic: "target_frame"
    joints: *robot_joints
    pd_gains:
        trans_x: {p: 15.0, d: 3.0}
        trans_y: {p: 15.0, d: 3.0}
        trans_z: {p: 15.0, d: 3.0}
        rot_x: {p: 15.0, d: 3.0} 
        rot_y: {p: 15.0, d: 3.0}
        rot_z: {p: 15.0, d: 3.0}

```
All works without the my_cartesian_motion_controller: 

[image]

When I introduce the my_cartesian_motion_controller in the .launch file, I see the following error: 

[image]

**Could you help me?** 

My .launch file: 
```
<?xml version="1.0"?>
<launch>
<group ns="master">
<include file="$(find ur_robot_driver)/launch/ur5_bringup.launch">
	<arg name="robot_ip" value="192.168.1.4"/>
	<arg name="reverse_port" value="50001"/>
	<arg name="script_sender_port" value="50002"/>
	<arg name="tf_prefix" value="master_"/>
	<arg name="controllers" value="joint_state_controller"/>
	<arg name="stopped_controllers" value=""/>
	<arg name="controller_config_file" value="$(find ur_robot_driver)/config/ur5_controllers_master.yaml"/> 
	<arg name="robot_description_file" value="$(find ur_description)/launch/ur5_upload_master.launch"/>
       <arg name="kinematics_config" value="$(find ur_description)/config/URcalib/master_calibration.yaml"/>  
</include> 
</group>

 <group ns="slave">
<include file="$(find ur_robot_driver)/launch/ur5_bringup.launch">
	<arg name="robot_ip" value="192.168.1.2"/>
	<arg name="reverse_port" value="50003"/>
	<arg name="script_sender_port" value="50004"/>
	<arg name="tf_prefix" value="slave_"/>
	<arg name="controllers" value="joint_state_controller my_cartesian_motion_controller"/>
	<arg name="stopped_controllers" value="my_cartesian_motion_controller"/>
	<arg name="controller_config_file" value="$(find ur_robot_driver)/config/ur5_controllers_slave.yaml"/>
	<arg name="robot_description_file" value="$(find ur_description)/launch/ur5_upload_slave.launch"/>
        <arg name="kinematics_config" value="$(find ur_description)/config/URcalib/slave_calibration.yaml"/> 
</include>
</group> 
</launch>
```

Thank you so much!

ManueU, 2021-02-16:
I solved the problem thanks to the following modifications: 

1. slave_controller.yaml: 

```
# Settings for ros_control control loop
hardware_control_loop:
   loop_hz: &loop_hz 125

 
# Settings for ros_control hardware interface
ur_hardware_interface:
   joints: &robot_joints
     - slave_shoulder_pan_joint
     - slave_shoulder_lift_joint
     - slave_elbow_joint
     - slave_wrist_1_joint
     - slave_wrist_2_joint
     - slave_wrist_3_joint

# Publish all joint states ----------------------------------
joint_state_controller: 
   type: joint_state_controller/JointStateController
   publish_rate: *loop_hz

# Cartesian motion controller   
my_cartesian_motion_controller:
    type: position_controllers/CartesianMotionController
    end_effector_link: "slave_tool0"
    robot_base_link: "slave_base"
    target_frame_topic: "target_frame"
    joints: *robot_joints
    pd_gains:
        trans_x: {p: 15.0, d: 3.0}
        trans_y: {p: 15.0, d: 3.0}
        trans_z: {p: 15.0, d: 3.0}
        rot_x: {p: 15.0, d: 3.0} 
        rot_y: {p: 15.0, d: 3.0}
        rot_z: {p: 15.0, d: 3.0}

```

2. cartesian_controller_base.hpp: 
```

// Get controller specific configuration
  if (!nh.getParam("/slave/robot_description",robot_description))
  {
    ROS_ERROR("Failed to load '/robot_description' from parameter server");
    return false;
  }
  if (!nh.getParam("/slave/my_cartesian_motion_controller/robot_base_link",m_robot_base_link))
  {
    ROS_ERROR_STREAM("Failed to load " << nh.getNamespace() + "/robot_base_link" << " from parameter server");
    return false;
  }
  if (!nh.getParam("/slave/my_cartesian_motion_controller/end_effector_link",m_end_effector_link))
  {
    ROS_ERROR_STREAM("Failed to load " << nh.getNamespace() + "/end_effector_link" << " from parameter server");
    return false;
  } 
```

**Is it correct?**
**If I have to use the my_catesian_motion_controller for both the robots, which is the correct way to do changes?**

Thank you!

gavanderhoorn, 2021-02-16:
It'd probably already work if you remove the `/` before `robot_description` in the parameter name.

with the `/`, it's always expected the parameter is in that exact location. This makes remapping and pushing down into namespaces impossible.

I would strongly suggest to *not* embed names of your current setup into a generic controller from this repository. So don't prefix `/slave` to `robot_description` fi.

ManueU, 2021-02-16:
I agree with you, but it doesn't work removing only  the `/` before `robot_description` in the parameter name.
Have you others advices? 

Thank you so much!

gavanderhoorn, 2021-02-16:
> I agree with you, but it doesn't work removing only the / before robot_description in the parameter name.

it should, if you have a `robot_description` parameter in each of your namespaces, which is what I assumed. And `nh` cannot be a private `NodeHandle`.

I haven't checked the code here, so my suggestion was purely based on behaviour of other controllers.

stefanscherzinger, 2021-02-17:
Yes, this needs adjustments. See #13

commit ce2966d6 (2021-07-21): Search for robot_description upwards in namespaces

This allows controllers to use local robot_descriptions and supports
to control various robots in parallel through namespacing.

Fixes #12 #13.
commit 94fdce7a (2021-07-21): Search for robot_description upwards in namespaces

This allows controllers to use local robot_descriptions and supports
to control various robots in parallel through namespacing.

Fixes #12 #13.
PR #24 (2021-07-21): Search for robot_description upwards in namespaces
This allows controllers to use local robot_descriptions and supports
to control various robots in parallel through namespacing.

Fixes #12
Fixes #13
closed 2021-07-21 by stefanscherzinger