fzi-forschungszentrum-informatik/cartesian_controllers #14: ur3 keeps moving when using my_cartesian_compliance_controller

jsbyysheng, 2021-02-20:
I used this config in my real world ur3 robot:
```
my_cartesian_compliance_controller:
    type: "position_controllers/CartesianComplianceController"
    end_effector_link: "tool_holder_link"    # All links below must come before this link
    robot_base_link: "base_link"
    ft_sensor_ref_link: "ft_sensor_link"
    compliance_ref_link: "tool_holder_link"
    target_frame_topic: "target_frame"
    joints: *robot_joints
    solver:
        error_scale: 50
        iterations: 50
    pd_gains:
        trans_x: {p: 10.0}
        trans_y: {p: 10.0}
        trans_z: {p: 10.0}
        rot_x: {p: 1.0}
        rot_y: {p: 1.0}
        rot_z: {p: 1.0}
```
When I publish the msg as below,  it was very strange that the robot kept moving to a very strange style with very high speed. (stiffness: 1000, 1000, 1000, 50, 50, 50)
```
rostopic pub -1 /my_cartesian_compliance_controller/target_wrench geometry_msgs/WrenchStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
wrench:
  force:
    x: 0.1            
    y: 0.0            
    z: 0.0          
  torque:
    x: 0.0              
    y: 0.0             
    z: 0.0" 
```

stefanscherzinger, 2021-10-18:
Hi @jsbyysheng 

I'm keeping up with old issues. Is this still relevant for you?

stefanscherzinger, 2021-11-19:
Could be related to #32

stefanscherzinger, 2022-01-13:
@jsbyysheng

Feel free to reopen if this is still an issue.

closed 2022-01-13 by stefanscherzinger