fzi-forschungszentrum-informatik/cartesian_controllers #18: Using  my_cartesian_force_controller in ur3

yaogengcheng, 2021-03-18:
Hello,
    Now I want to use my_cartesian_force_controller to achieve moving following the trajectory I plan, and at the mean time the end-effector should keep the constant force in the Z axis. I want to use it to finish sanding and polishing by admittance control. How can I do it?
  Thank you!

stefanscherzinger, 2021-10-18:
Hi @yaogengcheng 

Sorry, this package was somewhat unsupported for the last months. Did you solve your use case?

stefanscherzinger, 2021-12-04:
@yaogengcheng 

Until I test this myself, you could have a look at [this controller.](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/master/joint_to_cartesian_controller#joint-to-cartesian-controller). The idea is to connect Moveit to the `joint_trajectory_controller` and turn that into a target pose for the `cartesian_compliance_controller`.

PR #41 (2024-01-09): Improve compatibility of `CartesianComplianceController` with Moveit
## Goal
Users can plan robot end-effector motion with Moveit and execute that with Cartesian compliance in contact.

## Approach
Moveit's output is a `FollowJointTrajectoryGoal` that best integrates with the `joint_trajectory_controller`. The *cartesian_controllers* repository already provides a `JointControllerAdapter` that allows to connect such a `joint_trajectory_controller` and turn the joint-based trajectory into Cartesian targets via robot forward kinematics. For motion that is sufficiently away from robot singularitie [... cut]
closed 2022-07-01 by stefanscherzinger