fzi-forschungszentrum-informatik/cartesian_controllers #20: Using position_controllers/CartesianMotionController on custom urdf in Simulation

ehthas, 2021-05-09:
Hi,

I am trying to use the package for my custom model, a human model with 7 DOF arm. I installed the package using the instruction given in Readme touse catkin build with Release mode.  I started by implementing with my_joint_to_cartesian_controller, used the template and simply loaded my_joint_to_cartesian_controller/joint_trajectory_controller in launch file and I did get the following topics which I am supposed to get for my use case

/my_joint_to_cartesian_controller/joint_trajectory_controller/command
/my_joint_to_cartesian_controller/joint_trajectory_controller/follow_joint_trajectory/cancel
/my_joint_to_cartesian_controller/joint_trajectory_controller/follow_joint_trajectory/feedback
/my_joint_to_cartesian_controller/joint_trajectory_controller/follow_joint_trajectory/goal
/my_joint_to_cartesian_controller/joint_trajectory_controller/follow_joint_trajectory/result
/my_joint_to_cartesian_controller/joint_trajectory_controller/follow_joint_trajectory/status
/my_joint_to_cartesian_controller/joint_trajectory_controller/state

I followed the same pattern. I now made a yaml file with the template of my_cartesian_motion_controller, replaced end-effector  link, base link and joints with the ones of my robot and keeping target_frame_topic as target_frame(generic as in template). But I keep on getting the error: **Could not load controller 'my_cartesian_motion_controller' because controller type 'position_controllers/CartesianMotionController' does not exist.**.

I am unable to understand the issue, please guide. I have to eventually implement cartesian_compliance_controller. But not being able to understand the basic controller it is a bit difficult for me at the moment to implement them for my use case. Thank you.
Best Regards

stefanscherzinger, 2021-05-10:
Hi @ehthas 

> But I keep on getting the error: Could not load controller 'my_cartesian_motion_controller' because controller type 'position_controllers/CartesianMotionController' does not exist..

That looks as if the plugin did not get exported correctly.
What's the output of calling
```bash
    rospack plugins --attrib=plugin controller_interface
```
in a sourced shell? That should list the _cartesian_motion_controller_ among other controllers.

> 1. As my robot is JointPositionInterface based, so do I need to comment the JointVelocityInterface through out header and cpp files of the controller I am using.

No, the controllers are templated for the special cases of position and velocity interfaces. By using
```yaml
position_controllers/CartesianMotionController
```
in the .yaml file, you specify that you want to use the _position based version_. The velocity based one would be `velocity_controllers/CartesianMotionController`.

> 2. When I implemented my_joint_to_cartesain_controller I expected I will be able to send pose commands but since its joint_trajectory_type so if I still have to tell joint names and joint positions then how different it is from position_controllers/Joint TrajectoryController ?

Drop the _joint_to_cartesian_controller_. That converts joint-based trajectories to moving Cartesian targets. That's not what you want. I guess for your use case, the CartesianMotionController (alone) is a good choice. You can send it Cartesian poses and the controller will send joint positions to your robot.

ehthas, 2021-05-10:
Hi @stefanscherzinger, Thank you for the kind response. Ok I understand

 rospack plugins --attrib=plugin controller_interface

I did try it earlier and I tried it again now. Since I have the package inside my catkin_ws so until I am not inside my catkin_ws and source it I do not see the cartesian_motion_controller. This is the command I am running and below is the output I get

**Inside my catkin_ws I use the following command** 
**~/catkin_ws$** rospack plugins --attrib=plugin controller_interface | grep cartesian_motion_controller

**Result**
cartesian_motion_controller /home/has/catkin_ws/src/cartesian_controllers/cartesian_motion_controller/cartesian_motion_controller_plugin.xml

ehthas, 2021-05-10:
Yes, I am getting this error in my sourced catkin_ws.

stefanscherzinger, 2021-05-10:
Could you post the .yaml file of your controller configuration and the respective part in the launch file?

ehthas, 2021-05-10:
I have attached the yaml file and launch file. The lines in bold inside launch file shows where I load the controller.
[yaml_and_launch_files.docx](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/files/6452558/yaml_and_launch_files.docx)

stefanscherzinger, 2021-05-10:
I'll post the content here.
.yaml file:
```yaml
joint_state_controller:
    type: joint_state_controller/JointStateController
    publish_rate: 50

my_cartesian_motion_controller:
    type: "position_controllers/CartesianMotionController"
    end_effector_link: "right_wrist"
    robot_base_link: "world"
    target_frame_topic: "target_frame"
    joints:
    - jright_shoulder_rotz   
    - jright_shoulder_roty
    - jright_shoulder_rotx
    - jright_forearm_rotz
    - jright_forearm_roty 
    - jright_wrist_rotz
    - jright_wrist_rotx

    pd_gains:
        trans_x: {p: 10.0}
        trans_y: {p: 10.0}
        trans_z: {p: 10.0}
        rot_x: {p: 1.0}
        rot_y: {p: 1.0}
        rot_z: {p: 1.0}
      
gripper_controller:
  type: "position_controllers/JointTrajectoryController"
  joints:
    - jright_grip_rhand
  constraints: 
      goaltime: 0.5
      stopped_velocity_tolerance: 0.02
      jright_grip_rhand: {trajectory: 0.1, goal: 0.0}
  gains:  
    jright_grip_rhand: {p: 100.0, i: 0.01, d: 10.0, i_clamp: 1}

  stop_trajectory_duration: 0.5
  state_publish_rate: 25
  action_monitor_rate: 10

```

and launch file:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<launch>

        <!-- Robot Model -->   
        <param name="robot_description" command="$(find xacro)/xacro --inorder '$(find humsub_proj)/urdf/humarm_6dof.xacro'" /> 
        
        <arg name="x" default="0"/>
        <arg name="y" default="0"/>
        <arg name="z" default="0"/>
  
        <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" /> 
     
        
        <!-- Spawn the robot model -->
        <node name="mhuman_spawn" pkg="gazebo_ros" type="spawn_model"  output="screen"  
              args="-urdf -param robot_description -model mhuman -x $(arg x) -y $(arg y) -z $(arg z)" /> 

        <!-- Load Controllers -->
        <rosparam command="load" file="$(find humsub_proj)/config/cartesianmotion.yaml" ns="/mhuman"/> 
        <rosparam command="load" file="$(find humsub_proj)/config/pid_jointtocartesian.yaml" ns="/mhuman"/>  

        <!-- Controllers -->
        <node name="controller_spawner" pkg="controller_manager" type="spawner"   
              respawn="false" output="screen" ns="/mhuman" 
              args="--namespace=/mhuman 
                     joint_state_controller 
                    gripper_controller 
                    my_cartesian_motion_controller"/> 

</launch>
```

stefanscherzinger, 2021-05-10:
Looks alright. What's the output of asking for the available types?
```bash
rosservice call /mhuman/controller_manager/list_controller_types
```

stefanscherzinger, 2021-11-15:
Please reopen if this issue still persists.

closed 2021-11-15 by stefanscherzinger