fzi-forschungszentrum-informatik/cartesian_controllers #21: compliance controller bounce and does not stop in hard surfaces

leitoxve, 2021-06-10:
when I use the compliance controller against a rigid contact surface it starts to bounce and does not stop, it is as if the movement is not dampened
is there any damping parameter i need to modify???
what are the good parameters of pd and stiffness???
I'm using
p = 0.005
d = 0.0001
stiffness = 140

real time influences in some way ???
I have not modified the kernel to work in real time, I should do it??

leitoxve, 2021-06-11:
I am using universal robots 10e and the application in principle is like paint with a roller, I will try next week and report my results, thanks a lot for a quick response

leitoxve, 2021-07-01:
hello, indeed by lowering the `error_scale` and by decreasing the number of `iterations` from `5` the vibrations go down, it looks very nice =) but very slow = (
release mode effectively removes the red print " pipeline overflow"
what do the error and the iterations mean?
you know if this driver works for the Kuka iiwa???

closed 2021-11-15 by stefanscherzinger

shrutic12, 2024-04-08:
Hi @le, I am using the catesian compliance controller with UR10e robot on ROS2 Humble.  With the cartesian compliance controller, I am testing with different stiffnesses in XYZ translation direction and I am expecting that this will add "compliance" to the robot in these axes, and expect that the robot will be less stiff in these directions. For instance, reducing stiffness in Z, would mean I should be able to push the robot up (physically) and it should restabilize (like if it had a spring damper attached to the end effector). But this is not what happens in my tests. After reducing stiffness in XYZ, I command the robot to move vertically down to a position which has an obstacle, but instead of stopping on contact and stabalizing, the robot continue to move down.

Can you help me out? I have reduced the stiffness to about 90 in X,Y,Z translation for the tests. Are there any recommended stiffness values for UR10e from your experience/tests?