fzi-forschungszentrum-informatik/cartesian_controllers #25: Using this controller with iiwa Kuka

leitoxve, 2021-07-29:
Its posible to made a compliant control in iiwa kuka with this controller???

stefanscherzinger, 2021-10-18:
Hi @leitoxve 

I haven't worked with the Kuka iiwa myself, but at least the `RobotHW` seems to have the needed [`PositionJointInterface`](https://github.com/ros-industrial/kuka_experimental/blob/melodic-devel/kuka_rsi_hw_interface/include/kuka_rsi_hw_interface/kuka_hardware_interface.h#L118). So, my assumption is yes, it _should_ work.

But, the Kuka iiwa is known for its joint-torque sensors and superior impedance control in stiff contacts. If you want to command Cartesian motion, I suggest you use the `CartesianMotionController` instead and try to use _low-level_ joint-based Impedance control in parallel (at least I hope that's possible somehow to activate).

closed 2021-11-15 by stefanscherzinger

matthias-mayr, 2023-01-02:
We tried this controller package on a KUKA iiwa without a dedicated FT sensor. The results are here: https://lup.lub.lu.se/student-papers/search/publication/9064756
tl;dr: Emulating the FT sensor that is needed by this approach from the external torque signals from the iiwa made it challenging to get both compliance and stability.

We then implemented Cartesian impedance control that send torque commands and published it here: https://github.com/matthias-mayr/Cartesian-Impedance-Controller with a paper providing a high-level description and introduction to the control theory (https://arxiv.org/abs/2212.11215)
Both repositories allow end-effector control & force control and therefore complement each other. In fact, we use this one with our URs and impedance control with Panda & iiwas.