fzi-forschungszentrum-informatik/cartesian_controllers #26: Apply effort_controllers to this package

graziegrazie, 2021-09-17:
I'm planning to apply cartesian_compliance_controller in this package to Nextage Open, upper-body humanoid. The robot originally supports only effort_controllers. And, it seems difficult to add other controllers. I know this package supports position_controllers and velocity_controllers, not effort_controllers. On the other hand, both of position_controllers and effort_controllers contains joint_position_controller. Therefore, I am expecting there is possibility my plan will works well with some code change.

Could you give me any suggestion about my plan? Especially, @stefanscherzinger and @fmauch

stefanscherzinger, 2021-09-20:
Hi @graziegrazie 
> If I can apply effort_controller / EffortJointInterface to this package, does it seem to work well?

Unfortunately, that won't work. The controllers are really designed for velocity or position interfaces that most industrial-like robots provide, and use a forward dynamics-based approach to turn end-effector force/torque vectors into joint-based robot motion.

I see mainly two options, both of which require own implementation with some knowledge of the robot's dynamics:

1) You could implement a `PositionJointInterface` for the robot, in which you use some form of PID regulator for each axis. If you don't require excellent tracking performance, you might get away without needing feed-forward gravity compensation.

   > And, it seems difficult to define PositionJointInterface and VelocityJointInterface.

   Maybe it sounds more complicated than it is. Here's an implementation of the `joint_trajectory_controller`'s [hardware interface adapter](https://github.com/ros-controls/ros_controllers/blob/noetic-devel/joint_trajectory_controller/include/joint_trajectory_controller/hardware_interface_adapter.h#L283) for an effort interface.

2) You could implement your own compliance controller, an `impedance controller` in that case. Here's a implementation within Franka's [impedance example controller](https://github.com/frankaemika/franka_ros/blob/develop/franka_example_controllers/src/cartesian_impedance_example_controller.cpp#L173) to give you some ideas.

I didn't find ROS-control support in the [Next Age robot's package](https://github.com/tork-a/rtmros_nextage) at a first glance. Can you point me to the EffortJointInterface you plan to use?

graziegrazie, 2021-09-26:
Thank you for your comment @stefanscherzinger!

I understand well that effort_controller does not fit for this package.

> I didn't find ROS-control support in the Next Age robot's package at a first glance. Can you point me to the EffortJointInterface you plan to use?

I am really sorry. I understand wrongly about Nextage control system architecture. Nextage does not support any Hardware Interface. EffortJointInterface is not defined. moveit_commaner and ROS_client, which inherits moveit_commander, are [officially supported](http://wiki.ros.org/rtmros_nextage/Tutorials/Programming_Hiro_NEXTAGE_OPEN_MOVEIT). As far as I understand, lower layer is composed with different framework, not ROS.

So, since I'd like to ask you different question, let me close this one.

closed 2021-09-26 by graziegrazie