fzi-forschungszentrum-informatik/cartesian_controllers #31: Drift in x-y plane

r-ym, 2021-11-16:
Hello,

Thank you for open-sourcing the package. It has been super helpful with prototyping.

Recently I have been noticing a drift in my end-effector position when using the motion controllers i.e say I send a Linear command in Z while my end-effector is effectively planar to X-Y plane I notice a constant increase in the Y-axis. Will tweaking the controller gains and solver parameters address this issue?

stefanscherzinger, 2021-11-19:
@r-ym Thanks for your feedback. You're welcome :)

> I send a Linear command in Z while my end-effector is effectively planar to X-Y plane

That depends on what controller you are using: In case you are using the `CartesianForceController`, that drift is part of the algorithm's way of simulating reaction motion from external force-torque vectors (*forward dynamics*). The control input *guides* the robot, rather than enforcing motion via perfectly decoupled Cartesian axes.

In Case you are using the `CartesianComplianceController` with a wrist-mounted force-torque sensor, the drift could come from changing end-effector loads during motion.

In Case you are using the `CartesianMotionController`, the Cartesian axes should be decoupled, i.e. no drift should arise. That would indicate a behavior I would not expect, and I would like to hear about your controller parameterization.

> my end-effector is effectively planar to X-Y plane

To be sure that your measurement is *exact*, could you lookup the end-effector's relative pose to the robot base (e.g. with tf) before sending the command?
Something like
```bash
rosrun tf tf_echo base_link tool0  # adjust to your robot
```

stefanscherzinger, 2021-11-27:
@r-ym I rethought your question.

> I send a Linear command

That let me think somehow in end-effector coordinates. But the `CartesianMotionController` gets its target poses in the specified base frame, so the end-effector's orientation should not matter. I'm still assuming, however, that you are measuring this drift with `tf`. Could you confirm that?

I can think of two reasons, why the controller could still drift:
- If there happens to be a singular configuration on the way to your target, the controller will be forced to leave the *direct* path to your target. It will still try to stay as close as the robot's kinematics allows for it.
- If you command a single distant target via publishing the respective `PoseStamped` only once, the controller might drift a little. The algorithm uses an internal linearization to decouple the Cartesian axes as good as possible but you get more exact results if you publish the `PoseStamped` along the path you want to follow with a rate of, say `100Hz`.

> Will tweaking the controller gains and solver parameters address this issue?

If it's the first point, then no. If it's the second point, then adjusting the solver parameters and the *path to your target* by suitable publishing should remove the drift.

stefanscherzinger, 2021-12-04:
Please reopen if more clarification is needed.

closed 2021-12-04 by stefanscherzinger