fzi-forschungszentrum-informatik/cartesian_controllers #32: Document kinematic URDF requirements for controller parameterization

stefanscherzinger, 2021-11-19:
## Problem
All `URDF` links in the controllers configuration must be part of the same kinematic chain.
E.g.
```bash
    end_effector_link: "tool0"    # All links below must come before this link
    robot_base_link: "base_link"
    ft_sensor_ref_link: "sensor_link"
    compliance_ref_link: "tool0"
```
Assumes that there's a chain from `tool0` to `base_link`, with `sensor_link` in between, i.e.
```txt
base_link --- ... -- sensor_link -- ... --- tool0
```
The `end_effector_link` must be the last link. It's used as end point in KDL's solvers.

Users might configure this:
```txt
base_link --- ... -- some_other_link --- tool0
                                    \--- sensor_link 
```
which will lead to a wrong force propagation and thus wrong behavior with the `CartesianForceController` and `CartesianComplianceController`.

stefanscherzinger, 2021-11-19:
Thanks to @chriseichmann who ran into this recently.

captain-yoshi, 2022-08-18:
This could be easily solved by keeping two KDL tree's,  the only requirement is that the `sensor_link` and the `end_effector_link` must be part of the `robot_base_link`. I don't know if this would change a lot of things in the driver though...

That way this sort of configuration would be valid and the user would not have to change the URDF.
```
base_link --- ... -- some_other_link --- tool0
                                    \--- sensor_link
```

stefanscherzinger, 2022-09-20:
This is now enforced through #75 at controller initialization. Invalid configurations will fail to initialize and emit an error +  explanation. I think that's sufficiently addressed at the moment.

closed 2022-09-20 by stefanscherzinger