fzi-forschungszentrum-informatik/cartesian_controllers #39: Force and Compliance Controller: Gravity Compensation

KKSTB, 2021-12-02:
I can see there is gravity compensation feature. But how should I use it properly?
In the controller yaml, I've added `gravity` xyz and `tool` com_x, com_y, com_z, mass. But I see there is something called `signal_taring`. How should I deal with it? Thanks!

stefanscherzinger, 2021-12-04:
@KKSTB 

I have to admit that that's a somewhat hidden feature. I'll try to add documentation and probably rework that a little to make it more usable. After going through the code and trying to remember what I did there, I think the user was supposed to
specify tool mass and center of mass as you did in the controller's .yaml file. The `signal_taring` service call was optional. It should allow users to *bias away* additional force-torque measurements in the current robot's position. If I remember correctly, the users could call that controller's service right before biasing the sensor to zero (with a sensor-specific call).

But even after going through the code, I'm not completely sure how that worked or how the procedure was meant. Use that with care if you must. I'll work on a PR with integration test to rethink this feature on a higher level.

stefanscherzinger, 2021-12-09:
Small addition: I found an older explanation [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/commit/82d2d1d58ecba68a709116cd13b531c1b15e528b#commitcomment-38085756).

KKSTB, 2022-09-07:
I understand the algorithm now after trying it with force torque sensor. The algorithm need some fixes in order to get the compensation correct.
1. In `signalTaringCallback()`, there are calls to `Base::displayInTipLink()` and `Base::displayInBaseLink()`. These functions use `m_ik_solver->getPositions()`. So the tare function does not get the correct joint positions if the controller is not running. So `if (!Base::isRunning())`, it should call `Base::m_ik_solver->setStartState(Base::m_joint_handles, Base::m_joint_handle_map);` to update joint positions.
2. In `signalTaringCallback()` and `compensateGravity()`, the wrench calculated and assigned to `tmp` is referenced at ft_sensor_ref_link. I tested that their value actually behave very close to FT sensor wrench if plotted in graph. So their calculation should be treated similar to FT sensor wrench. Therefore they should also be frame-transformed with `m_ft_sensor_transform` before `Base::displayInBaseLink()`. Because of this requirement, I optimized the calculation by not frame-transforming in `ftSensorWrenchCallback()`, and not rotate to base link in `signalTaringCallback()` and `compensateGravity()`. Instead inside `computeForceError()` I added the raw FT sensor wrench + model weight wrench + gravity comp during taring together, and then frame-transform by `m_ft_sensor_transform` and rotate to base link by `Base::displayInBaseLink()` altogether to get the compensated value.

The compensation works after the above modifications.

stefanscherzinger, 2022-09-09:
@KKSTB Many thanks for sticking with this and taking the time to understand the feature!

> The compensation works after the above modifications.

That's good to hear! Could you maybe provide a PR? That would be awesome for me to review and integrate.

KKSTB, 2022-09-30:
I cherry-picked the fix to create the PR. Although I believe it should work, it would be great if someone could do some verification. Thanks for the great work of cartesian controller.