fzi-forschungszentrum-informatik/cartesian_controllers #42: ROS2 Version

Souphis, 2021-12-05:
Hi!
Are there any plans for porting this controller to ROS2? In our project, we have rewritten the controller to the ROS2 but without the ros control part (just plain calculation). Maybe we could help you with that after New Year?

stefanscherzinger, 2021-12-06:
Hi @Souphis 

Yes, there are plans to port this to ROS2. Unfortunately, it has never been *urgent* enough to do that. We recently used the `CartesianForceController` for a small assembly demo at a national fair. That state is on [this branch](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2-devel), albeit a little `UR`-driver specific. I plan to port both the `CartesianMotionController` and the `CartesianComplianceController` soon. That should not be too much effort, although I need to reiterate on some basic control philosophies with the new ROS2 possibilities.

Support is highly welcome, especially through testing on real hardware and through opening issues!

What robot are you working on?

Souphis, 2021-12-06:
Hi @stefanscherzinger

In our lab we are using UR and ABB robots. Until now we developed our own driver for ABB (without hardware interface), but we are going to port this https://github.com/ros-industrial/abb_robot_driver to ROS2 soon. Of course, we will be happy to help you with testing ROS2 version.

jdlangs, 2022-05-04:
@stefanscherzinger we're interested in the status of the ROS2 version and in supporting if it makes sense for you. Where would you say things are now in terms of the `ros2` branch?

stefanscherzinger, 2022-05-17:
@jdlangs The `ros2` branch has a working version of the *cartesian_force_controller*. We used that together with the UR10e (ROS2 galactic driver). I'm now back from a longer holiday and am taking care of porting the rest in the next days/weeks.

Supporting makes sense for me with testing on real hardware. I'm setting up a simulation for helping me with the ROS2 port, but that won't cover everything. If you want to support, it would be awesome to test the [clean-up-ros2-control](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/clean-up-ros2-control) branch. That should become the basis also for the `cartesian_motion_controller` and `cartesian_compliance_controller`.

stefanscherzinger, 2022-07-05:
@jdlangs @gbartyzel 

The `cartesian_controllers` are now mostly ported to **ROS2**. The current status is on the [ros2-devel](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2-devel) branch. What's missing at the moment is mostly documentation. But If you know your way around ROS2-control, that should not be a problem. I'm tremendously grateful for any additional feedback and use cases on real hardware :)

jdlangs, 2022-07-08:
@stefanscherzinger thanks for the update! FYI today is my last day at SwRI and I most likely won't be working on this in the future. Lily (@lcbw) will be taking the lead on working with the new ROS 2 controllers and she is actively working on testing them now.

stefanscherzinger, 2022-07-08:
@jdlangs Alright, thanks for the info and good look for your next steps!

stefanscherzinger, 2022-09-20:
See latest [ros2](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2) version. Documentation is now updated.

closed 2022-09-20 by stefanscherzinger