fzi-forschungszentrum-informatik/cartesian_controllers #45: I can not find Controller Manager plugin in rqt.

jack162, 2022-01-07:
hello sir,
i follow the tutorial of cartesian_motion_controller.

At step2
"In another sourced terminal, open rqt and navigate to the Controller Manager plugin under Robot Tools. Select /controller_manager as namespace and activate my_cartesian_motion_controller. Also activate my_motion_control_handle."

I can't find where is Controller Manager,and i don't know how to activate my_cartesian_motion_controller and my_motion_control_handle also.

here is the logging and my rqt.

_auto-starting new master
process[master]: started with pid [12565]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 746ac8ba-6f86-11ec-81cc-0f25607d5329
process[rosout-1]: started with pid [12578]
started core service [/rosout]
process[sim_hardware_interface-2]: started with pid [12585]
process[robot_state_publisher-3]: started with pid [12586]
process[joint_state_controller_spawner-4]: started with pid [12587]
process[controller_spawner-5]: started with pid [12592]
[ INFO] [1641538382.147323898]: Waiting for model URDF on the ROS param server at location: //robot_description
process[my_joint_to_cartesian_controller/controller_spawner-6]: started with pid [12597]
process[rviz-7]: started with pid [12603]
[ INFO] [1641538382.255676508]: GenericHWInterface Ready.
[ INFO] [1641538382.255712603]: SimHWInterface Ready.
[ INFO] [1641538382.318936093]: rviz version 1.14.10
[ INFO] [1641538382.318973029]: compiled against Qt version 5.12.8
[ INFO] [1641538382.318981617]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1641538382.327060737]: Forcing OpenGl version 0.
[INFO] [1641538382.555254]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1641538382.558662]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1641538382.560923]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1641538382.563101]: Loading controller: joint_state_controller
[INFO] [1641538382.570345]: Controller Spawner: Loaded controllers: joint_state_controller
[ INFO] [1641538382.579871257]: Stereo is NOT SUPPORTED
[ INFO] [1641538382.579928669]: OpenGL device: NV117
[ INFO] [1641538382.579947916]: OpenGl version: 4.3 (GLSL 4.3) limited to GLSL 1.4 on Mesa system.
[ INFO] [1641538382.582083913]: Forward dynamics solver initialized
[ INFO] [1641538382.582171796]: Forward dynamics solver has control over 6 joints
[INFO] [1641538382.630409]: Started controllers: joint_state_controller
[ INFO] [1641538382.638396159]: Forward dynamics solver initialized
[ INFO] [1641538382.638424694]: Forward dynamics solver has control over 6 joints
[ INFO] [1641538382.673071918]: Failed to load /my_cartesian_force_controller/gravity from parameter server
[ INFO] [1641538382.673472969]: Failed to load /my_cartesian_force_controller/tool from parameter server
[ INFO] [1641538382.691275593]: Forward dynamics solver initialized
[ INFO] [1641538382.691370680]: Forward dynamics solver has control over 6 joints
[ INFO] [1641538382.718041897]: Failed to load /my_cartesian_compliance_controller/gravity from parameter server
[ INFO] [1641538382.718337260]: Failed to load /my_cartesian_compliance_controller/tool from parameter server
[image]

graziegrazie, 2022-01-07:
I also faced same situation. Could you try  `rosrun rqt_controller_manager rqt_controller_manager` ? Then, you will see a window like an attached image. I hope this helps you.

[image]

stefanscherzinger, 2022-01-07:
@jack162 The `rqt_controller_manager` is not part of the *default* installation. Try installing that as a system dependency like a normal ROS package, e.g.:
```bash
sudo apt install ros-noetic-rqt-controller-manager 
```
You will then be able to use it in rqt under `Plugins -> Robot Tools` or run it directly as a ROS node, as @graziegrazie mentioned.

jack162, 2022-01-07:
oh! it is work for me!
Thank you a lot!

closed 2022-01-07 by jack162

kurshakuz, 2022-08-18:
> @jack162 The `rqt_controller_manager` is not part of the _default_ installation. 

If that is the case, then adding a note to install that package in the [README](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers#getting-started) would have been nice imo.

crane-1, 2025-04-16:
thanks a lot!