fzi-forschungszentrum-informatik/cartesian_controllers #46: [ERROR] Could not load class cartesian_controllers/MotionControlHandle:

jack162, 2022-01-09:
hello sir,
I am sorry to disturb you,and thank to your reply.

i have run the example.launch,and already work in simulation.
Now I want to run this package on my real robot.
I bring up my robot and load the robot model by my self.
After I modify the example.launch and  config file,
I got these errors.

`process[sim_hardware_interface-1]: started with pid [16828]
process[robot_state_publisher-2]: started with pid [16829]
process[joint_state_controller_spawner-3]: started with pid [16830]
process[controller_spawner-4]: started with pid [16831]
process[my_joint_to_cartesian_controller/controller_spawner-5]: started with pid [16834]
[ INFO] [1641719585.555838361]: Waiting for model URDF on the ROS param server at location: //robot_description
[ INFO] [1641719585.715890492]: GenericHWInterface Ready.
[ INFO] [1641719585.715934264]: SimHWInterface Ready.
[INFO] [1641719585.858335]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1641719585.862199]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1641719585.865203]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1641719585.870465]: Loading controller: joint_state_controller
[ INFO] [1641719585.932273288]: Forward dynamics solver initialized
[ INFO] [1641719585.932329071]: Forward dynamics solver has control over 6 joints
[INFO] [1641719586.107606]: Controller Spawner: Loaded controllers: joint_state_controller
[ INFO] [1641719586.142645515]: Forward dynamics solver initialized
[ INFO] [1641719586.142699369]: Forward dynamics solver has control over 6 joints
[ INFO] [1641719586.313108168]: Failed to load /my_cartesian_force_controller/gravity from parameter server
[ INFO] [1641719586.314744461]: Failed to load /my_cartesian_force_controller/tool from parameter server
[INFO] [1641719586.337597]: Started controllers: joint_state_controller
[ INFO] [1641719586.365877227]: Forward dynamics solver initialized
[ INFO] [1641719586.365937714]: Forward dynamics solver has control over 6 joints
[ INFO] [1641719586.503257427]: Failed to load /my_cartesian_compliance_controller/gravity from parameter server
[ INFO] [1641719586.504587068]: Failed to load /my_cartesian_compliance_controller/tool from parameter server
**[ERROR] [1641719586.553613922]: Could not load class cartesian_controllers/MotionControlHandle: Could not find library corresponding to plugin cartesian_controllers/MotionControlHandle. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1641719586.553707571]: Could not load controller 'my_motion_control_handle' because controller type 'cartesian_controllers/MotionControlHandle' does not exist.
[ERROR] [1641719586.553749021]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1641719587.555364]: Failed to load my_motion_control_handle
[ERROR] [1641719587.562036925]: Could not load class cartesian_controllers/JointControllerAdapter: Could not find library corresponding to plugin cartesian_controllers/JointControllerAdapter. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1641719587.562158849]: Could not load controller 'my_joint_to_cartesian_controller' because controller type 'cartesian_controllers/JointControllerAdapter' does not exist.
[ERROR] [1641719587.562209026]: Use 'rosservice call controIller_manager/list_controller_types' to get the available types
[ERROR] [1641719588.563634]: Failed to load my_joint_to_cartesian_controller**
[WARN] [1641719616.126391]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[my_joint_to_cartesian_controller/controller_spawner-5] process has finished cleanly
`

I already install controller_manager.
is their has anything i forget to change?

stefanscherzinger, 2022-01-09:
@jack162 

> [ERROR] [1641719586.553707571]: Could not load controller 'my_motion_control_handle' because controller type 'cartesian_controllers/MotionControlHandle' does not exist.

Seems as if the `controller_manager`'s terminal window doesn't know the `cartesian_controllers` library. Just in case you forgot, you need to source your workspace after building.
If that doesn't solve it, what's the output of
```bash
rospack plugins --attrib=plugin controller_interface
```
and
```bash
ldd -r devel/lib/libcartesian_controller_handles.so
```
?

jack162, 2022-01-10:
@stefanscherzinger 
I already source my workspace.
The output of rospack plugins --attrib=plugin controller_interface is 
```
cartesian_motion_controller /home/kyyoung/josh/detect_patient_ws/src/cartesian_controllers/cartesian_motion_controller/cartesian_motion_controller_plugin.xml
joint_to_cartesian_controller /home/kyyoung/josh/detect_patient_ws/src/cartesian_controllers/joint_to_cartesian_controller/joint_to_cartesian_controller_plugin.xml
cartesian_controller_handles /home/kyyoung/josh/detect_patient_ws/src/cartesian_controllers/cartesian_controller_handles/cartesian_controller_handles_plugin.xml
cartesian_force_controller /home/kyyoung/josh/detect_patient_ws/src/cartesian_controllers/cartesian_force_controller/cartesian_force_controller_plugin.xml
cartesian_compliance_controller /home/kyyoung/josh/detect_patient_ws/src/cartesian_controllers/cartesian_compliance_controller/cartesian_compliance_controller_plugin.xml
joint_trajectory_controller /opt/ros/kinetic/share/joint_trajectory_controller/ros_control_plugins.xml
joint_state_controller /opt/ros/kinetic/share/joint_state_controller/joint_state_plugin.xml
```
and the output of is 
`ldd: devel/lib/libcartesian_controller_handles.so: No such file or directory`

stefanscherzinger, 2022-01-10:
@jack162 

> so I just remake it by
catkin_make --only-pkg-with-deps cartesian_controller_handles
and it seems can find the controller !

Good, that's a start. But I'm a bit confused about your workspaces. You seem to be working in *detect_patient_ws* and *my_ws*. Make sure that you build and source everything appropriately and don't mix workspaces. Also, you seem to be using **kinetic**, which was *end of life* in May last year. You should consider upgrading your system.

> my_joint_to_cartesian_controller. Cannot enforce limits for joint 'tm_shoulder_1_joint'. It has no velocity limits specification. 

If you don't explicitly need it, I would recommend to drop the `my_joint_to_cartesian_controller` for now. To achieve this, remove [this line](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_examples/launch/examples.launch#L47) and [this line](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_examples/launch/examples.launch#L48), and remove [this group](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_examples/launch/examples.launch#L54) in your launch file.

Could you upload or paste your `.launch` file here for further debugging?

jack162, 2022-01-11:
@stefanscherzinger 
Thank you for your reply. I really appricate it.
Actually, my_ws is detect_patient_ws,they are same workspace.
I am sorry to make you confused.

The OS of this robot is kinetic I know it is too old.
But if upgrade have some problems,so I don't upgrade it yet.

After I follow your instructions.
My node and topic graphic is this.
[image]
I think the problems is joint_state_publisher.
Here are my bring up launch file and the example file for my robot.
https://github.com/kyyoung54402/josh

jack162, 2022-01-12:
@stefanscherzinger 
1. I think my tm_driver is same as [this](https://github.com/viirya/ros-driver-techman-robot/tree/master/tm_driver),and it looks same as [the repo of @kentsai0319 ] (https://github.com/kentsai0319/techman_robot/tree/master/tm_driver).Therefore,I think my tm_drive is support ros_control.

stefanscherzinger, 2022-01-12:
@jack162 

> Therefore,I think my tm_driver is support ros_control.

Yes, that looks good.

> Can I drop joint_state_controller and keep joint_state_publisher?

No, that won't work. You need the joint state feedback from the real hardware as provided by the `joint_state_controller`. The `joint_state_publisher` is just a helper tool for simulation that gives you sliders to easily move joints. In ROS-control, you would normally use a `joint_trajectory_controller` and the `ros-<version>-rqt-joint-trajectory-controller` plugin for that task.

> I have remove sim_hardware_interface in mars.launch,but when i run it.The output is

That's probably because the real robot's controller manager is in a different namespace.

> Should I write my own hardware_interface like This tutorial?

That's probably a good tutorial to get an overall idea of what's going on with ROS-control. But no, you should not write a `hardware_interface` for your robot. That's already done by the driver you are using. It's more a question of how to configure everything properly with launch files.

---

I recommend the following steps:

1. Start your robot driver normally and leave the `cartesian_controllers` aside for now. Find the robot's `controller_manager` namespace by calling
    ```bash
    rosservice list | grep controller_manager
    ```
    That should list services, such as e.g. `/tm_700/controller_manager/list_controllers`. In this example, `tm_700` would be the namespace.
2. Now load the `cartesian_controllers` to the parameter server in that namespace. That would be something similar to
    ```bash
        <group ns="tm_700" >
                <rosparam file="$(find cartesian_controller_examples)/config/example_controllers.yaml" command="load"/>
        </group>
    ```
    I guess you could put that code [here](https://github.com/kyyoung54402/josh/blob/main/mars_lite_bringup.launch#L16) in your launch file.
3. Next, restart your driver with the launch file and use the [rqt-controller-manager](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/45#issuecomment-1007307823) to *load* and *start* the controllers.
4. After loading and starting the `joint_state_controller`, your TF tree should be working again.
5. If all of that works, try to investigate/google how to automatically start controllers in your launch files.

jack162, 2022-01-16:
@stefanscherzinger  I almost made it! 

i succeeded load and start joint_state_controller. and check what is the robot's controller_manager namespace by calling  
`rosservice list | grep controller_manager`
and the output is 
```
/controller_manager/list_controller_types
/controller_manager/list_controllers
/controller_manager/load_controller
/controller_manager/reload_controller_libraries
/controller_manager/switch_controller
/controller_manager/unload_controller
```
it seems no specify namespace,
now, I already start `my_cartesian_motion_controller` and `my_motion_control_handle`
My rqt graph as below. 
[image]
But when I drag the interactive marker ,the robot didn't move at all.(I have remap target_frame in tm_driver node
```
<remap from="my_motion_control_handle/target_frame" to="target_frame" />
<remap from="my_cartesian_motion_controller/target_frame" to="target_frame" />
```
[image]

I also try to publish value to /target_frame by
```
kyyoung@mars-nctu:~$ rostopic pub  /target_frame geometry_msgs/PoseStamped \ '{header: {frame_id: "base_link"},pose: {position:{x: 0.23,y: -0.5,z: 0.85},orientation: {x: 0.71242103881,y: 0.0263192043967,z: 0.0267772663724,w: 0.700747130531}}}'
publishing and latching message. Press ctrl-C to terminate
```
But it seems fail.

stefanscherzinger, 2022-01-17:
@jack162 
Great! That's a progress.

> I also try to publish value to /target_frame by

Careful with that. Depending on how you specified your `my_cartesian_motion_controller`, this might result in a quick jump straight towards that goal. Try to use the `my_motion_control_handle` in the beginning to get a feeling for the controller and its parameters.

Your controller specification seems ok. I'm not sure at the moment why it is not moving. Could you perform some basic test first? Try stopping the `my_cartesian_motion_controller` and `my_motion_control_handle` and *load* and *start* the `joint_trajectory_controller` instead. You can then use the `rqt` gui (ros-kinetic-rqt-joint-trajectory-controller) to move the joints with sliders. That's usually the first thing I like to do on a new setup to check if ROS-control does what I would expect.

If that's working in your case, let's see if the topics are really connected correctly with
```bash
rostopic list | grep target
```
and then `rostopic info <topic-of-interest>` to confirm that the driver is both sending and receiving via the correct target topic.

It looks like so in your `rqt graph` screenshot, but let's check anyway.

Are you sure it's not moving, or is it maybe moving *very*, *very* slowly?

jack162, 2022-01-19:
I already solve it! 
the controller type In tm_driver [tm_hardware_interface.cpp](https://github.com/viirya/ros-driver-techman-robot/blob/master/tm_driver/src/tm_hardware_interface.cpp#:~:text=%7D-,else%20if%20(controller_it%2D%3Etype%20%3D%3D%20%22hardware_interface%3A%3APositionJointInterface%22)%20%7B,-ROS_DEBUG(%22Stopping%20position) of tm_driver is `hardware_interface::PositionJointInterface`.
but in [example_controllers.yaml](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_examples/config/example_controllers.yaml#:~:text=%2D%20joint6-,joint_trajectory_controller%3A,-type%3A%20%22position_controllers/JointTrajectoryController) is `position_controllers/JointTrajectoryController`.
After I make them same name the robot can move by joint_trajectory_controller! 
tomorrow I will try to use your `cartesian_controllers`.
Thank you!!

jack162, 2022-01-20:
hi @stefanscherzinger  , I am sorry to disturb you.
I try to change `hardware_interface::PositionJointInterface` In  [tm_hardware_interface.cpp] to [position_controllers/CartesianMotionController](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_examples/config/example_controllers.yaml#L2) and 
[cartesian_controllers/MotionControlHandle](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_examples/config/example_controllers.yaml#L75),But the robot didn't move also.

Here is some information
```
kyyoung@mars-nctu:~$ rostopic list | grep target
/target_frame
```
```
kyyoung@mars-nctu:~$ rostopic info /target_frame 
Type: geometry_msgs/PoseStamped

Publishers: 
 * /tm_driver (http://192.168.50.77:38297/)

Subscribers: 
 * /tm_driver (http://192.168.50.77:38297/)
```
```
kyyoung@mars-nctu:~$ rostopic info /my_motion_control_handle/update_full 
Type: visualization_msgs/InteractiveMarkerInit

Publishers: 
 * /tm_driver (http://192.168.50.77:38297/)

Subscribers: None
```
```
kyyoung@mars-nctu:~$ rostopic info /my_motion_control_handle/feedback 
Type: visualization_msgs/InteractiveMarkerFeedback

Publishers: 
 * /rviz_1642671973461984459 (http://192.168.50.5:35371/)

Subscribers: 
 * /tm_driver (http://192.168.50.77:38297/)
```

stefanscherzinger, 2022-01-20:
@jack162 

I'm referring to your earlier posts first:

> After I change hardware_interface::PositionJointInterface to position_controllers/JointTrajectoryController, the robot can move by joint_trajectory_controller!

That's definitely not the right approach to work with ROS-control. You should not need to change source code. That indicates that something in general is not working. That's very weird. All ROS-controllers are normally interfaced with `pluginlib` via base class pointers. And their types should be `hardware_interface::PositionJointInterface` for controllers that command joint positions.

> The OS of this robot is kinetic I know it is too old.
But if upgrade have some problems,so I don't upgrade it yet.

Maybe that's part of the problem. But I'm not sure where to start looking (system-wide ROS installed libraries, packages built from source, deprecated ROS Kinetic, ...)

> I try to change hardware_interface::PositionJointInterface In tm_hardware_interface.cpp to either position_controllers/CartesianMotionController or
cartesian_controllers/MotionControlHandle,But the robot didn't move also.

At this point, it's not really an issue of the `cartesian_controllers`. As a very first test, you should be able to get the combination of `robot driver` + `joint_trajectory_controller` + `rqt gui` up and running **without** changing source code. If that's not working, try to open an issue at the driver's repository and tell them that ROS-control does not work in your case.

> Can tm_driver do IK by itself when my_motion_control_handle just send a pose to tm_driver?

I don't think so. I didn't find any interface or controller that would allow doing that in their repository. But that's something to ask the driver's maintainers.

jack162, 2022-01-22:
@stefanscherzinger  Thank you!

> the output of dpkg -l | grep pluginlib

`kyyoung@mars-nctu:~$ dpkg -l | grep pluginlib
ii  ros-ardent-class-loader                           1.0.0-13xenial-20171209-014146-0800                              amd64        The class_loader package is a ROS-independent package for loading plugins during runtime and the foundation of the higher level ROS &quot;pluginlib&quot; library.
ii  ros-ardent-pluginlib                              3.0.0-0xenial-20180208-133902-0800                               amd64        The pluginlib package provides tools for writing and dynamically loading plugins using the ROS build infrastructure.
ii  ros-kinetic-class-loader                          0.3.9-0xenial-20191214-003009+0000                               amd64        The class_loader package is a ROS-independent package for loading plugins during runtime and the foundation of the higher level ROS &quot;pluginlib&quot; library.
ii  ros-kinetic-pluginlib                             1.11.3-0xenial-20210503-102752-0800                              amd64        The pluginlib package provides tools for writing and dynamically loading plugins using the ROS build infrastructure.
ii  ros-kinetic-pluginlib-tutorials                   0.1.10-0xenial-20210503-103710-0800                              amd64        The pluginlib_tutorials package
ii  ros-kinetic-ros-base                              1.3.2-0xenial-20210503-151705-0800                               amd64        A metapackage which extends ros_core and includes other basic non-robot tools like actionlib, dynamic reconfigure, nodelets, and pluginlib.`

jack162, 2022-02-14:
After remake cartesian_controllers,it has work!
But I am still change the souce code in [tm_hardware_interface.cpp](https://github.com/viirya/ros-driver-techman-robot/blob/master/tm_driver/src/tm_hardware_interface.cpp#L225).
Thank you！

closed 2022-02-19 by jack162