fzi-forschungszentrum-informatik/cartesian_controllers #49: Question about 'cartesian_force_controller'

yukkwan-yuan, 2022-02-14:
Dear Sir,
Hello, I am a beginner in robotics, I would like to ask if your 'cartesian_force_controlle' toolkit uses impedance control? Or other control methods, where do I need to find the relevant description? thanks

stefanscherzinger, 2022-02-14:
Hello @yukkwan-yuan 

The `cartesian_force_controller` is an *admittance* type controller. That means it tracks a given *force-torque* reference profile and commands the robot joints with *motion* (position or velocities).

Here's a [high-level overview](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/resources/doc/Solver_details.md#the-control-loop) of the control cycle. And here's more information on [the theory](https://arxiv.org/pdf/2009.11888.pdf) (Fig. 2).

yukkwan-yuan, 2022-02-14:
Thanks for the quick answer, it helped me a lot, thank you very much!

closed 2022-02-14 by stefanscherzinger