fzi-forschungszentrum-informatik/cartesian_controllers #50: can I using cartesian_motion_controller by sending a command?

jack162, 2022-02-14:
Hello sir! 
I am the owner of this [issue](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/46).
Finally, I successfully using cartesian_controllers in my real robot!
Both motion control and force control work well,that is so amazing.
Thank you so much!

But In motion control.
Instead of using the interactive marker to guide the end effector of the robot,
I want to send a target pose by hand. It is possible?

I have try to subscribe /target_frame to get current pose of end effector,
and send a target pose to /my_motion_control_handle/update.

But the robot didn't move.

stefanscherzinger, 2022-02-14:
Hi @jack162 

> Both cartesian_motion_controller and cartesian_force_controller work well,that is so amazing.

Great! I'm happy that it works for you now!

> I want to send a target pose by hand. Is it possible?

Yes, that's possible. Sending target poses to the robot via a script (ROS node) is the intended behavior. The `cartesian_controller_handle` is just a simple tool to test things and teleoperate the robot to get to know the controllers.

> I have try to subscribe /target_frame to get current pose of end effector,
and send a target pose to /my_motion_control_handle/update.

Try it the other way: First, stop the `my_motion_control_handle` so that it does not compete with the target that you send manually. Then, publish directly to `/target_frame`, either from the command line, or from within some python script. But be careful! Try that maybe in simulation first. If you send a target that's *far* away from your current position, the robot will try to get there very quickly. You can also lower the [error_scale](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/47#issuecomment-1033979286) to e.g.. `0.01` to make the robot move slowly at first. After you have more experience with sending targets (from python scripts), you can increase that value again.

stefanscherzinger, 2022-02-17:
@jack162 

> [This video](https://www.youtube.com/watch?v=prFZojc9Q10) is the result of using cartesian_controllers for remote ausculation.

Thank you for sharing this! This might also help others some day.

> Is it possible change the controller at script?

Yes, sure. That's possible. That is not specific to the `cartesian_controllers`. That is with the normal `switch_controllers` service provided by the `controller_manager`. [Here's an example](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_tests/integration_tests/test_startup.py#L58) as part of an integration test from this package. Maybe you can use that as inspiration for your use case.

> or using two controller at same time?

It's not possible to use both `carteian_motion_controller` and `cartesian_force_controller` in parallel. They will compete for writing joint commands and will conflict. That's what the [`cartesian_compliance_controller`](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/master/cartesian_compliance_controller#cartesian-compliance-controller) is for. That's a hybrid between the two and exposes *both* control interfaces: target motion and target wrench.

jack162, 2022-02-20:
@stefanscherzinger  Thank you!
I have implement [cartesian_compliance_controller](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/master/cartesian_compliance_controller#cartesian-compliance-controller) on my robot. Here is the [result](https://studio.youtube.com/video/BMS932HiO_s/edit) 
It looks so good!
[cartesian_controllers](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers) is so amazing!

closed 2022-02-20 by jack162