fzi-forschungszentrum-informatik/cartesian_controllers #58: Getting started example Stuck

WackoCommander, 2022-04-07:
I am following the getting started instructions for the cartesian compilance controller. 
When I reach step 3, it requires me to publish a geometry_msgs/WrenchStamped to /my_cartesian_compliance_controller/target_wrench. However, for some reason the topic does not appear, therefore I am unable to publish to it. 

When I do first run the launch file, I get the following errors. Not sure if it is relevant. 

`[ INFO] [1649329496.614889386]: Failed to load /my_cartesian_force_controller/gravity from parameter server
[ INFO] [1649329496.615109900]: Failed to load /my_cartesian_force_controller/tool from parameter server
[ INFO] [1649329496.629339174]: Forward dynamics solver initialized
[ INFO] [1649329496.629366176]: Forward dynamics solver has control over 6 joints
[ INFO] [1649329496.648531181]: Failed to load /my_cartesian_compliance_controller/gravity from parameter server
[ INFO] [1649329496.648751449]: Failed to load /my_cartesian_compliance_controller/tool from parameter server
`

Will appreciate any help.

closed 2022-05-02 by WackoCommander

stefanscherzinger, 2022-05-17:
@WackoCommander Sorry for the late response. I was on an extended vacation with zero-email-policy. Did things work out for you?

WackoCommander, 2022-05-18:
Yes, everything worked out. Thank you.