fzi-forschungszentrum-informatik/cartesian_controllers #59: Joint continuity

caranx-medical, 2022-04-20:
Hi, 

I would like to start using this "robot agnostic" Cartesian controller module - which seems pretty promising. 
Before installing and trying it I would like first to understand if and how jumps in joint space are avoided. 

According to the robot kinematics and joint limits the robot, while tracking a pose, might get stuck or it might be required to fully reconfigure in joint space. How these two scenarios are handled by this module?

Thanks for your help!

stefanscherzinger, 2022-05-17:
Sorry @caranx-medical for the late reply. I'm catching up after my vacation.
Nice to hear that you are interested in the controllers!

To explain what happens in your two scenarios, let me give you a *visual* explanation of how the solver controls the robot end-effector for motion tracking:
The solver turns the difference to your target pose into a 6-dimensional force (translation and rotation) and simulates (and commands to the driver) how the robot moves by that force. Imagine you sit in the target pose as a tiny observer and pull the robot (e.g. with a rope that you attached to the end-effector) towards yourself. This is essentially what happens. Now, if you reach joint limitations on your way, the robot might no longer move straight towards you, but instead in a *best effort* kind of way, i.e as best as the kinematics allows for it. 

> **might get stuck**: Yes. You might end up in local minima, even when *going another way* would have brought you closer to the target. The controllers will not trigger reconfigurations themselves.

> **it might be required to fully reconfigure in joint space**: That's required before you operate in Cartesian control. A rule of thumb is: Move with joint-space control (possibly planned), whenever you have big motion with traversing big parts of the robot's workspace, and Cartesian control when you are in *the right spot* in you workspace and don't expect configuration changes for your application.

Giggus, 2022-05-18:
No worries and thanks for your reply!
Another question, while moving toward the target is self-collision and/or collision with the environment avoided?

Thanks again