fzi-forschungszentrum-informatik/cartesian_controllers #60: applying damping to compliance controller for full impedance?

sroubert, 2022-05-11:
I notice the original paper implements damping. However, in the controller configurations I only see options for linear stiffness. How can I include damping?

stefanscherzinger, 2022-05-17:
Hi @sroubert 

> How can I include damping?

Are you asking because of contact instability?

I split the philosophy of damping into two sets of parameters. For *impedance control*, damping gives you an option to adjust how *viscous* (as in velocity-proportional damping) the end-effector feels and also determines oscillations of the impedance element (together with the stiffness) and contact stability. For admittance control, which we have with *position-* or *velocity controlled* robots, I decided that this is better realized with an alternate mechanism. I also wanted to get rid of spring-damper oscillations, for which I couldn't find meaningful use cases.

In the current implementation, making the system less *viscous* (that is, more responsive) is done with increasing the [`error_scale`](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/resources/doc/Solver_details.md#solver-parameters). There's an experimental limit to that responsiveness. You will note oscillations in contact if the system is too responsive. In addition, contact stability can also be tweaked (in a certain margin) with the PD gains in each of the six Cartesian axes.

stefanscherzinger, 2022-07-01:
@sroubert

> From my understanding, the error_scale implicitly controls the damping?

Yes, it also influences the damping behavior.

> Is there an explicit relationship between this error_scale and viscosity / damping parameters as in figures 7 and 8 of the FDCC IROS 2017 article?

No, there's no longer such explicit relationship.

closed 2022-07-01 by stefanscherzinger

Daan0103, 2024-01-19:
Just a question, what is exactly the difference between cartesian compliance control and cartesian impedance control? Because it is very comparable with each other right? I'm researching which of the two is the best solution in my project, using an UR5e.

stefanscherzinger, 2024-01-19:
@Daan0103

Impedance and Admittance control (compliance control = sub category of Admittance control) use different Algorithms to realize compliant behavior on robotic manipulators. Roughly citing Hogan, one of the original authors to draw an explicit distinction:

- Impedance control = Mapping *motion* to *effort* , e.g. Cartesian *motion offset* to *joint torques*
- Admittance control = Mapping *effort* to *motion*, e.g. Cartesian *force-torque signals* to *joint velocity*

If you have some minutes, I suggest you read into [this paper](https://fileadmin.cs.lth.se/ai/Proceedings/ICRA2010/MainConference/data/papers/1462.pdf) by Christian Ott et al. It nicely recapitulates the concepts and provides great insight.

Your use case requires Admittance control. Until Universal Robots provide access to their low-level torque control interface, you command that thing with either joint position commands or joint velocity commands. If you use signals from a force-torque sensor to realize compliant behavior, that's Admittance control.