fzi-forschungszentrum-informatik/cartesian_controllers #61: Adding cartesian_force_controller to controller_manager

cschempp, 2022-06-15:
Hello,

i tried adding the cartesian_force_controller from the ros2 branch into the [ur_control.launch.py](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_bringup/launch/ur_control.launch.py) of the [UR ROS2 Driver package](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver).
Therefore i cloned this repo and build the packages cartesian_controller_base and cartesian_force_controller in my workspace.
I then added 

```
cartesian_force_controller_spawner_stopped = Node(
     package="controller_manager",
     executable="spawner",
     arguments=["cartesian_force_controller", "-c", "/controller_manager", "--stopped"],
    )
```

to the launch_setup in ur_control.launch.py and  `cartesian_force_controller_spawner_stopped` in the nodes_to_start = [ .... ].

In the [ur_controllers.yaml ](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_bringup/config/ur_controllers.yaml) where the controllers are defined, i added 

```
cartesian_force_controller:
  ros__parameters:
    end_effector_link: "tool0"
    robot_base_link: "base_link"
    ft_sensor_ref_link: "sensor_link"
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint
      
    pd_gains:
        trans_x: {p: 0.05}
        trans_y: {p: 0.05}
        trans_z: {p: 0.05}
        rot_x: {p: 0.01}
        rot_y: {p: 0.01}
        rot_z: {p: 0.01}
```

and under controller_manager: ros__parameters: i added the line 
```
cartesian_force_controller:
      type: position_controllers/CartesianForceController
```

If i list the controller types with `ros2 control list_controller_types`, the cartesian_force_controller/CartesianForceController is there.
But loading the controller with `ros2 control load_controller cartesian_force_controller` yields the error message:
**Error loading controller, check controller_manager logs**. The logs say **"Loader for controller 'cartesian_force_controller' not found"**.

What is the correct way to register a new controller to the controller_manager started in an existing .launch file?

Any help would be really appreciated.

stefanscherzinger, 2022-06-15:
Hi @cschempp

The port to ROS2 is currently under development. Please use the [ros2-devel](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2-devel) branch for the meantime.
You can build the complete `cartesian_controllers` package inside a ROS2 workspace with colcon. Unported packages are *colcon-ignored*.

> But loading the controller with ros2 control load_controller cartesian_force_controller yields the error message:

Yes, the controller's type changed to `cartesian_force_controller/CartesianForceController`. That would now be:
```bash
controller_manager:
  ros__parameters:

...

    cartesian_force_controller:
      type: cartesian_force_controller/CartesianForceController

... 
```

Also note that the controller gains might need some smaller adaptation to ROS2. This is currently experimental. Anything you can share about your experience on your system (+ use case an good controller parameters) is greatly appreciated!

cschempp, 2022-06-15:
Hi @stefanscherzinger ,

thank you for the quick reply. Ok, i will use the ros2-devel branch.
We are going to use an UR5e and other industrial manipulators for automated assembly.
Therefore i'm highly interested in getting the cartesian controllers ported to ros2.
Of course, i will be happy to help in regards of testing on real-hardware.

TheFalcoGuy, 2022-07-01:
Hi, I am also using a UR5e robot for force control. I followed @cschempp 's steps to run the controller along with @stefanscherzinger 's suggestions. I ran the launch command 

`ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.0.88 launch_rviz:=false`
However, I get the following error:
`[ros2_control_node-1] [INFO] [1656640062.114959053] [cartesian_force_controller]: Forward dynamics solver initialized
[ros2_control_node-1] [INFO] [1656640062.114976674] [cartesian_force_controller]: Forward dynamics solver has control over 6 joints
[ros2_control_node-1] [ERROR] [1656640062.115359766] [cartesian_force_controller]: No command_interfaces specified
[ros2_control_node-1] [ERROR] [1656640062.115416897] [controller_manager]: After configuring, controller 'cartesian_force_controller' is in state 'finalized' , expected inactive.
[spawner.py-13] [INFO] [1656640062.115944629] [spawner_cartesian_force_controller]: Failed to configure controller
`
The controller successfully loaded, but failed to be configure and was unable to start. I am wondering what the solution for this issue is? Below is the full error message (FYI I am porting the compliance controller to ROS 2). Thanks.
`
[INFO] [launch]: All log files can be found below /home/d-lab/.ros/log/2022-06-30-21-47-39-364486-dlab-Alienware-m15-R6-393766
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [393772]
[INFO] [controller_stopper_node-3]: process started with pid [393774]
[INFO] [ros2_control_node-1]: process started with pid [393770]
[INFO] [robot_state_publisher-4]: process started with pid [393776]
[INFO] [spawner.py-5]: process started with pid [393778]
[INFO] [spawner.py-6]: process started with pid [393780]
[INFO] [spawner.py-7]: process started with pid [393782]
[INFO] [spawner.py-8]: process started with pid [393784]
[INFO] [spawner.py-9]: process started with pid [393786]
[INFO] [torque_broadcaster-10]: process started with pid [393788]
[INFO] [gripper-11]: process started with pid [393790]
[INFO] [spawner.py-12]: process started with pid [393792]
[INFO] [spawner.py-13]: process started with pid [393794]
[dashboard_client-2] [INFO] [1656640059.657408369] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[robot_state_publisher-4] Parsing robot urdf xml string.
[controller_stopper_node-3] [INFO] [1656640059.664828509] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[robot_state_publisher-4] Link base had 0 children
[robot_state_publisher-4] Link base_link_inertia had 1 children
[robot_state_publisher-4] Link shoulder_link had 1 children
[robot_state_publisher-4] Link upper_arm_link had 1 children
[robot_state_publisher-4] Link forearm_link had 1 children
[robot_state_publisher-4] Link wrist_1_link had 1 children
[robot_state_publisher-4] Link wrist_2_link had 1 children
[robot_state_publisher-4] Link wrist_3_link had 1 children
[robot_state_publisher-4] Link flange had 1 children
[robot_state_publisher-4] Link tool0 had 0 children
[robot_state_publisher-4] [INFO] [1656640059.665007402] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1656640059.665082223] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1656640059.665091114] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1656640059.665096295] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1656640059.665101301] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1656640059.665106679] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1656640059.665112080] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1656640059.665117462] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1656640059.665122835] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1656640059.665128531] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1656640059.665134235] [robot_state_publisher]: got segment wrist_3_link
[ros2_control_node-1] [INFO] [1656640059.677263551] [URPositionHardwareInterface]: Starting ...please wait...
[ros2_control_node-1] [INFO] [1656640059.677317411] [URPositionHardwareInterface]: Initializing driver...
[ros2_control_node-1] [WARN] [1656640059.678489939] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[ros2_control_node-1] [INFO] [1656640059.763582347] [UR_Client_Library]: Negotiated RTDE protocol version to 2.
[ros2_control_node-1] [INFO] [1656640059.764214414] [UR_Client_Library]: Setting up RTDE communication with frequency 500.000000
[spawner.py-7] [INFO] [1656640059.789348221] [spawner_speed_scaling_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-12] [INFO] [1656640059.792310462] [spawner_cartesian_compliance_controller]: Waiting for /controller_manager services
[spawner.py-8] [INFO] [1656640059.792652087] [spawner_force_torque_sensor_broadcaster]: Waiting for /controller_manager services
[spawner.py-9] [INFO] [1656640059.793529044] [spawner_joint_trajectory_controller]: Waiting for /controller_manager services
[... cut ...]
[ros2_control_node-1] [ERROR] [1656640062.115416897] [controller_manager]: After configuring, controller 'cartesian_force_controller' is in state 'finalized' , expected inactive.
[spawner.py-13] [INFO] [1656640062.115944629] [spawner_cartesian_force_controller]: Failed to configure controller
[INFO] [spawner.py-7]: process has finished cleanly [pid 393782]
[INFO] [spawner.py-8]: process has finished cleanly [pid 393784]
[ERROR] [spawner.py-13]: process has died [pid 393794, exit code 1, cmd '/home/d-lab/workspace/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner.py cartesian_force_controller -c /controller_manager --stopped --ros-args'].
`

stefanscherzinger, 2022-07-01:
@TheFalcoGuy 

> An error happens when I try building the compliance controller on line 83 of cartesian_compliance_controller.cpp

Thanks, that pointed me to something important. The `ros2-devel` branch is *ROS2 Galactic* only at the moment.
I had the CI-pipeline deactivated for *Foxy* and missed that implementation change with `init -> on_init`.
Unfortunately, that's not the only change.

I'll try to support `Foxy` this evening and get back to you soon. (That includes your change for *2)*, which is correct).

Concerning your point *1)*: I'll personally like using *gdb* inside a screen session. Note the line with the `prefix`:
```python
    control_node = Node(
        package="controller_manager",
        executable="ros2_control_node",
        parameters=[robot_description, robot_controllers],
        #prefix="screen -d -m gdb -command=/home/scherzin/.ros/my_debug_log --ex run --args",
        output={
            "stdout": "screen",
            "stderr": "screen",
        },
    )
```
After picking up the *gdb* session with `screen -r`, I inspect the backtrace inside *gdb* with `bt`. (I also like to have some *breakpoints* ready in `/home/scherzin/.ros/my_debug_log` that help me stop at the right spots while debugging.) That should give some insights. But let's fix the support for *Foxy* first before digging deeper.

stefanscherzinger, 2022-07-02:
@TheFalcoGuy 

I updated the [ros2-devel](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2-devel) branch with the latest fixes for the *Foxy* support. You might want to check that out and try again.

TheFalcoGuy, 2022-07-02:
Thanks for being so quick. I tried running the compliance controller with my ur5e. While I did not get any errors, I did not get any compliant motion either. My setup is still the same as last time, 

I ran the following commands:
`ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.0.88 launch_rviz:=false`
`ros2 control switch_controllers --start cartesian_compliance_controller --stop joint_trajectory_controller`
and 
`ros2 control list_controllers
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active    
cartesian_force_controller[cartesian_force_controller/CartesianForceController] inactive  
force_torque_sensor_broadcaster[ur_controllers/ForceTorqueStateBroadcaster] active    
cartesian_compliance_controller[cartesian_compliance_controller/CartesianComplianceController] active    
io_and_status_controller[ur_controllers/GPIOController] active    
joint_trajectory_controller[joint_trajectory_controller/JointTrajectoryController] inactive`

My yaml file is as follows:
`cartesian_compliance_controller:
  ros__parameters:
    end_effector_link: "tool0"
    robot_base_link: "base_link"
    ft_sensor_ref_link: "tool0"
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint
      
    pd_gains:
        trans_x: {p: 0.05}
        trans_y: {p: 0.05}
        trans_z: {p: 0.05}
        rot_x: {p: 0.01}
        rot_y: {p: 0.01}
        rot_z: {p: 0.01}
    
    stiffness:  # w.r.t. compliance_ref_link
        trans_x: 50.0
        trans_y: 50.0
        trans_z: 50.0
        rot_x: 10.0
        rot_y: 10.0
        rot_z: 10.0
    
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
    state_publish_rate: 100.0
    action_monitor_rate: 20.0
    allow_partial_joints_goal: false
    constraints:
      stopped_velocity_tolerance: 0.0
      goal_time: 0.0`

And my rqt graph is as follows:
[image]

I noticed in the rqt graph that I am facing a similar issue in #15 where the cartesian compliance controller is getting information from the incorrect topic. Unlike ROS1, however, parameter remapping from the launch file is slightly trickier as there is no global parameter anymore. I was wondering if you can give some directions on how to remap /m_ft_sensor_wrench to a different topic (the topic I want to remap to is the /ft_data topic but for some reason it is not listed in rqt_graph). Thanks

stefanscherzinger, 2022-07-03:
@TheFalcoGuy 

> Unlike ROS1, however, parameter remapping from the launch file is slightly trickier as there is no global parameter anymore.

At runtime, ROS2 controllers are part of the `controller_manager`. So, remapping seems to work for that node. For instance:
```python
    control_node = Node(
        package="controller_manager",
        executable="ros2_control_node",
        parameters=[robot_description, robot_controllers],
        #prefix="screen -d -m gdb -command=/home/scherzin/.ros/my_debug_log --ex run --args",
        output={
            "stdout": "screen",
            "stderr": "screen",
        },
        remappings=[
            ('m_ft_sensor_wrench', 'the_measured_ur_sensor_wrench'),
            ]
    )
```
remaps the `cartesian_compliance_controller`'s `m_ft_sensor_wrench` to `the_measured_ur_sensor_wrench` (I'm not sure right now how that's called on the real system).

> I got motion, though I am not entirely sure if this is expected behavior

Thanks for sharing the video! That's not expected behavior. For some axes, the robot moves *towards* the directions where forces are applied. I assume that this is because the `compliance_ref_link` parameter is missing in the controller's specification. Try add this to your controller:

```yaml
cartesian_compliance_controller:
  ros__parameters:
    end_effector_link: "tool0"
    robot_base_link: "base_link"
    ft_sensor_ref_link: "tool0"
    compliance_ref_link: "tool0"  # This one is important
    ...
```
You might also remove the following parameters form the `cartesian_compliance_controller`:
```yaml
state_interfaces:
  - position
  - velocity
state_publish_rate: 100.0
action_monitor_rate: 20.0
allow_partial_joints_goal: false

constraints:
  stopped_velocity_tolerance: 0.0
  goal_time: 0.0`
```
Those parameters won't be used. I don't know where you got them from.

I also noticed that your controller is very responsive. That won't be stable in contact with stiff surfaces. Keep in mind that you might need to tweak the controller's `p` gains a little (for instance increase the rotational gains quite a bit in comparison to the translational ones). The [`error_scale`](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/ros2-devel/resources/doc/Solver_details.md#solver-parameters) parameter is still used in ROS2, e.g. with
```yaml
cartesian_compliance_controller:
  ros__parameters:
  ...

  solver:
    error_scale: 0.1
```

> My goal is to use the compliance controller as an admittance controller. How would I go about tuning this to get that behavior? 

Can you be more specific what do you mean by that?

TheFalcoGuy, 2022-07-08:
Hi @stefanscherzinger, sorry for the late reply. I took your advice and have started tuning my controller. 

After adding the compliace reference link, the issue you pointed out still remains. When I inverted the proportional gains, however, the robot behaves as expected. The error scaling you mentioned helped adjusting the controller's sensitivity. During the tuning process, I noticed that applying a large amount of force on the arm results in the arm drifiting in the direction of where the force was applied. I was wondering if you had any experience with dealing with this issue? Thanks again for the advice!

stefanscherzinger, 2022-07-08:
@TheFalcoGuy 

> When I inverted the proportional gains, however, the robot behaves as expected

You mean like `trans_x = -0.05`? I believe that's not the right way. Let's check again if the kinematic setup is correct.
Could you try `base` instead of `base_link` in your controller's config?

```yaml
cartesian_compliance_controller:
  ros__parameters:
    end_effector_link: "tool0"
    robot_base_link: "base"  # The force sensor reference frame for the UR driver in Foxy
    ft_sensor_ref_link: "tool0"
    compliance_ref_link: "tool0"
    ...
 ```
I remember that the UR driver used `base` for its force torque sensor signals.

>  I tried adding a k_i term as I thought this was a result of some steady state error

That's not required. The control law has an integrator already when integrating the simulated accelerations to velocities and positions.

> I noticed that applying a large amount of force on the arm results in the arm drifiting in the direction of where the force was applied. 

After fixing the kinematics config from above, is that still the case?

lcbw, 2022-07-12:
ubuntu 20.04 | fzi cart_controllers on branch foxy | universal robots ros2 driver on branch foxy | 
trying to implement the cartesian_force_controller:
I am also encountering this issue and changing my robot_base_link to base instead of base_link did work - thank you!

stefanscherzinger, 2022-08-09:
Hi @lcbw 

Sorry, I didn't check your post after your edit and missed your question. I suggest using the `ros2-devel` branch for now. That has most features and also fixed some deficiencies when switching controllers. After merging some cleanup, everything goes into the `ros2` branch.

closed 2022-08-09 by stefanscherzinger

shrutichakraborty, 2024-01-31:
> cartesian_force_controller_spawner_stopped = Node(
>      package="controller_manager",
>      executable="spawner",
>      arguments=["cartesian_force_controller", "-c", "/controller_manager", "--stopped"],
>     )

Hello, I'm following this response to setup the cartesian force controller on a UR10e on ros2 and am a bit confused with the terminology. What is the significance of the "stopped" in `cartesian_force_controller_spawner_stopped = Node(
     package="controller_manager",
     executable="spawner",
     arguments=["cartesian_force_controller", "-c", "/controller_manager", "--stopped"],
    )` ?