fzi-forschungszentrum-informatik/cartesian_controllers #67: Add example for command line interface

TomGrimwood, 2022-07-20:
Could an example for sending a position to the position_controllers/CartesianMotionController through a ROS node or Command line be added? I am working on integrating it into my system but I am a newbie when it comes to this :D

stefanscherzinger, 2022-07-22:
@TomGrimwood 

I'll see if I can setup an example this weekend.

That would be for instance:
- Bring the robot into some desired joint configuration (with the `joint_trajectory_controller`)
- Make a TF lookup from the specified `end_effector_link` to the specified `robot_base_link`
- Fill that into a `geometry_msgs/PoseStamped` message and publish that to the `position_controllers/CartesianMotionController`'s input topic. Remember to set the header's `frame_id` with the specified `robot_base_link`.
- Now change the position in that message a little like e.g. 0.1 (10cm) in x.
- The controller will interpolate to that target
- Lower the solver's `error_scale` for smooth motion. Higher it for faster *jumps* to the target.

TomGrimwood, 2022-07-23:
Thats really helpful thank you.

I am trying to incorporate this into a system similar to that of the https://github.com/DFKI-NI/mobipick mobipick which is a UR5 mounted on a MiR100 base. but I have having issues loading the controller, However I was able to get it working with just a standalone UR5 model in gazebo. Do you see any obvious reasons why this implementation wouldn't work? does the robot_base_link set in the .yaml file have to be the actual base link of my robots URDF? Or can it be the base_link of the UR5, which is not the origin of my URDF.

stefanscherzinger, 2022-07-25:
@TomGrimwood 

> does the robot_base_link set in the .yaml file have to be the actual base link of my robots URDF? Or can it be the base_link of the UR5, which is not the origin of my URDF

It *should* be the base link of the UR5 manipulator in this case. The reason for setting both a `robot_base_link` and an `end_effector_link` is that the controllers use those endpoints to find the active joints they can use for control. Internally, they build a kinematics chain from those joints in between. I guess parsing fails if you specify a more distant (as in URDF hierarchical distance) link somewhere in the mobile base. I will definitely fail if there's a non-fixed joint in the URDF part that will be included in the manipulators kinematics chain but that the controller can not control, such as odometry-related joints.

> which is a UR5 mounted on a MiR100 base.

That should work well. We are doing something very similar in our lab.

stefanscherzinger, 2022-08-16:
@TomGrimwood 

I mentioned I would try to setup an example. Unfortunately, there hasn't been much time to do this. Did you make progress in the meantime?

TomGrimwood, 2022-08-16:
Hey, Yes I got it working with the MiR + UR5. The issue was basically the controllers for the arm were loaded in a  <group> tag a namespace - adding the cartesian controller to the list of nodes launched within the group gave an error "Searched enclosing namespaces for robot_description but nothing found", even though the other controllers launched fine. I fixed this by removing the group tag and just having the one namespace.

I am not great with C++, but am trying to learn -  so I was unable to figure out the TF lookup method you suggested.
Do you know if there is an easy way to have the motion_control_handle relative to the world frame of a mobile manipulator?

Thanks for your work and help Stefan!

stefanscherzinger, 2022-08-17:
Thanks for your feedback!

>  I fixed this by removing the group tag and just having the one namespace.

It should be possible to have namespaces, though. It fails, if there is no `robot_description` upwards from the sub namespace (where you launch the controllers) to the root namespace. But if you found something that works for you without too much inconvenience, you could of course skip debugging this and leave it as is.

> I am not great with C++, but am trying to learn - so I was unable to figure out the TF lookup method you suggested.

I just found a promising node for following TF frames directly inside this package [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_handles/nodes/moving_target.py). This is the respective [launch file](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_handles/launch/moving_target.launch). Maybe that's easier to use. When trying this, use a low `error_scale` of the controller at first. This will let the controller move slowly so that you can test things, get used to the behavior, and hit the emergency stop in case the motion is not what you expect.

> Do you know if there is an easy way to have the motion_control_handle relative to the world frame of a mobile manipulator?

Unfortunately not right out of the box. I remember implementing something like this, but I have to find and commit it. I think that could be a relevant feature. I thus opened a request in #71.

closed 2022-09-09 by stefanscherzinger