fzi-forschungszentrum-informatik/cartesian_controllers #70: Tool COM frame

captain-yoshi, 2022-08-16:
In what frame does the tool COM belongs to ? I presume it is from the `end_effector_link` parameter. 

My end effector configuration is a dual gripper. I would like the closed gripper fingertips (e.g. the `robotiq_2f140_tcp` link) to apply a force in a direction. If I set it that way, the COM frame is invalid (if my assumption is correct).

``` yaml
my_cartesian_force_controller:
    type: "position_controllers/CartesianForceController"
    com_link: "tool0" # <-- should another link be added for COM like so
    end_effector_link: "robotiq_2f140_tcp"
    robot_base_link: "base_link"
    ft_sensor_ref_link: "robotiq_ft300_sensor"

    tool:
      mass: 2.12 # kg
      com_x: -0.024
      com_y: 0.0
      com_z: 0.0923

```

[image]

stefanscherzinger, 2022-08-16:
Here's an example (also specify the gravity). Adjust the values to your use case:

```bash
my_cartesian_force_controller:
    type: "position_controllers/CartesianForceController"
    end_effector_link: "robotiq_2f140_tcp"
    robot_base_link: "base_link"
    ft_sensor_ref_link: "robotiq_ft300_sensor"

    # Controller gains and solver specifics ...

    # Gravity should point towards the center of earth.
    # In robot_base_link.
    gravity:
        x: 0
        y: 0
        z: -9.81

    tool:
        # Mass of everything that's mounted after the sensor.
        # If unknown, try to estimate this as good as possible.
        mass: 2.12

        # The center of mass in the ft_sensor_ref_link.
        # Take a measure and try to estimate that as good as possible.
        com_x: 0.0
        com_y: 0.0
        com_z: 0.235

```

captain-yoshi, 2022-08-16:
Thx for your detailed answer!

If someone had a sensor near the gripper, would the mass between `tool0` and the `sensor` be compensated somehow ? This sort of configuration may be rare.

**Somewhat related question**: Are the URDF inertial tags (for the tool) used when simulating the dynamics ?

stefanscherzinger, 2022-08-17:
You are welcome!

> If someone had a sensor near the gripper, would the mass between tool0 and the sensor be compensated somehow ?

No, that wouldn't be compensated. Only the parts that come *after* the sensor (so that the sensor can measure their weight) are taken into consideration. The idea and purpose of this compensation mechanism is as follows (adapting myself from [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/commit/82d2d1d58ecba68a709116cd13b531c1b15e528b#commitcomment-38085756)):

> When using force control in some application, the force-torque sensor is usually zero-ed in this new configuration to enable an unbiased force control in contact. This sensor zeroing, however, will not be persistent to changes in orientation of the tool, especially if the tool has a considerable mass at a certain distance to the end-effector. With the specification of gravity and a center of mass (com), users can make sensor biasing invariant to rotations.

Everything that comes *before* the sensor will not affect the sensor measurements directly. The dynamic effects of these components will be suppressed by the high-gain joint controllers of the robotic manipulators. Most robot's OEM softeare allows to specify a separate tool mass for improved motion tracking if that's known to the user.

> Are the URDF inertial tags (for the tool) used when simulating the dynamics ?

No, they aren't. But that could be implemented as an improvement. Currently, the users need to estimate the sum of all component masses *after* the sensor and specify that along with the corresponding center of mass in the controller's config. Thinking of this, it could be a nice feature to parse the tool inertial tags from the URDF and automate the computation of the overall center of gravity. You could open a feature request for this :)

closed 2022-08-17 by stefanscherzinger