fzi-forschungszentrum-informatik/cartesian_controllers #73: Force Controller: intended behaviour

captain-yoshi, 2022-08-17:
Testing the force controller with real hardware. I want the tip of my gripper to apply a constant force of `-4N
` in the Z axis (wrt. the fingertips of the gripper) on a box without moving in any other directions. The robot starts moving in the right direction, but when it hits the box it starts sliding out of the box (almost like searching for a way to continue it's movement without exceeding the wrench force). Is this the intended behaviour ?

I am feeding the `/my_cartesian_force_controller/ft_sensor_wrench` with my ft sensor (wrt. the `ft_sensor_ref_link` frame).

To achieve what I want is it better to use the compliance controller and to crank up stiffness in the other axes ? Or maybe I could set the pd gains to zero for the other axis...

Step by step guide:
- Bringup the robot
- Reset the FTS
- Signal taring (`/my_cartesian_force_controller/signal_taring`)
- Start the controller `my_cartesian_force_controller` (initially stopped)
- Apply the `-4N` on the Z axis to the `/my_cartesian_force_controller/target_wrench` topic.

stefanscherzinger, 2022-08-17:
> Is this the intended behaviour ?

It depends. Maybe you could provide a screenshot in Rviz or some schematic sketch for better illustration of the frames. Assuming you refer to the [image](https://user-images.githubusercontent.com/32679594/184980632-45cf869e-fec5-4132-b9ce-63f438676fa4.png) from #70, I think what you describe is what I would expect. A constant force of `-4N` in the Z axis (blue axis) would cause a horizontal motion parallel to the surface. For making controlled normal contact, I would have commanded `+4N` in the X axis (red axis).

> The robot starts moving in the right direction,

That's strange. I would expect horizontal motion right away. Maybe it's something with the gravity compensation. Remember that your kinematic chain must meet some requirements as briefly explained [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/32). Could you post here your `tool` and `gravity` entries from the `.yaml` configuration?

> To achieve what I want is it better to use the compliance controller and to crank up stiffness in the other axes ? 

You could do that, yes. The philosophy behind the compliance controller is controlling the robot primarily with target poses and adjusting contact forces a little in a volume governed by the restoring stiffness forces.

> Or maybe I could set the pd gains to zero for the other axis...

I wouldn't do that. I would go for compliance instead.

captain-yoshi, 2022-08-17:
@stefanscherzinger Here is the setup, slightly different from #70.

[image]

stefanscherzinger, 2022-08-17:
Thanks for the image and for the config file!

When in contact, does it leave the box to one side? And during leaving the box, does the robot change the end-effector orientation?

Could you confirm that the robot does not rotate when guiding the robot slowly by hand in the end-effector (by grasping the colored coordinate system you drew), It shouldn't. If it does, there's something wrong.

Could you also deactivate the gravity compensation (by setting the gravity constant to `0.0`)? Let's see if that has some effect.

captain-yoshi, 2022-08-18:
> When in contact, does it leave the box to one side? And during leaving the box, does the robot change the end-effector orientation?

Yes and the robot end-effector orientation **does not** change. I have attached a [video](https://drive.google.com/drive/folders/1kNhk7iVsJjqg-gAHRThBnH4p1LU73pz-?usp=sharing) that demonstrate this. The video does not compensate for gravity (set to `0.0`).

> Could you confirm that the robot does not rotate when guiding the robot slowly by hand in the end-effector (by grasping the colored coordinate system you drew), It shouldn't. If it does, there's something wrong.

The robot does not rotate when slowly guided by hand from the gripper tcp. However the coordinate does not seem correct (e.g. pulling-down the gripper pushes the gripper in another axis...). Is the sensor frame changed to match the end effector frame ?

> Could you also deactivate the gravity compensation (by setting the gravity constant to 0.0)? Let's see if that has some effect.

Disabling the gravity compensation does not seem to have any effects. This should be expected as the tcp orientation does not change.

> It's kind of prone to little drifts and the controller best combines with some entity that corrects this intuitively, such as a human operator via teleoperation, manual guidance or some deep learned skill that controls all axes to not diverge too much.

That makes sense so I think the controller behaviour is appropriate. So my options are:
- Enforce desired axes not to diverge too much with the `force_controller`. Not sure how to do that.
- Use the `compliance_controller` and set desired axes to be stiffed and set a desired wrench. This should apply a constant force to a desired contact point to be completely bound ?

My 2 use cases are:
- Apply a constant force in one axis without diverging too much (what we discussed until now).
- Touch N Go: Have a grasped stylus to click something on a touch screen. I think using the `compliance_controller` and changing the target_frame (once the desired force contact has been achieved) would work.

stefanscherzinger, 2022-08-18:
Thanks again for the detailed information!

There's something wrong with the setup.

> However the coordinate does not seem correct (e.g. pulling-down the gripper pushes the gripper in another axis...)

That's *not* correct behavior. We need to fix this. It will also break the `cartesian_compliance_controller`.
Your posted config seems alright, but it does not necessarily show if the kinematics is valid. Could you check #32 and see if all specified links are part of the same kinematic chain?

> Is the sensor frame changed to match the end effector frame ?

Measurements are transformed automatically to the right frames. No need to change anything manually besides assuring a valid setup in the config file.

> That makes sense so I think the controller behaviour is appropriate.

No, it's not (yet) appropriate. The *wrong coordinates* you experienced by hand are also commanding the robot in some arbitrary direction when in contact. That's why I wanted to check by hand. Both the `cartesian_force_controller` and the `cartesian_compliance_controller` must feel intuitive and plausible when moved by hand. That's usually the first thing I check to confirm my setup.

Let's reflect on your options for your use cases after fixing this.

---

Btw: What ROS version are you using?

captain-yoshi, 2022-08-18:
> Your posted config seems alright, but it does not necessarily show if the kinematics is valid

You were right, I assumed it was part of the same kinematic tree... Will update this thread when I can test the modified tree on the real robot.

[image]

captain-yoshi, 2022-08-19:
It is working as expected now. It feels intuitive and plausible when moved by hand. Also it stays at the desired target wrench contact point with a very small drift (the contact point surface of the box is a little bit slippery).

> Btw: What ROS version are you using?

Noetic.

captain-yoshi, 2022-08-19:
Trying to debug the `CartesianComplianceController` now.

> The cartesian_compliance_controller must feel intuitive and plausible when moved by hand

I am not able to move the robot by hand. I have reduced the translation stiffness of the robot to the minimum of `100 N/m`. There is no target wrench applied nor target frame. 

Should I be able to move the robot like the ForceController with these conditions ? This config is the same used for the ForceController and is working as expected...

``` yaml
my_cartesian_compliance_controller:
  type: "position_controllers/CartesianComplianceController"
  robot_base_link: "ur3_base_link"
  end_effector_link: "robotiq_2f140_tcp"
  ft_sensor_ref_link: "robotiq_ft300_frame_id"
  compliance_ref_link: "robotiq_2f140_tcp"
  target_frame_topic: "target_frame"
  joints: *robot_joints

  stiffness:
    trans_x: 100
    trans_y: 100
    trans_z: 100
    rot_x: 100
    rot_y: 100
    rot_z: 100

  pd_gains:
    trans_x: {p: 0.05}
    trans_y: {p: 0.05}
    trans_z: {p: 0.05}
    rot_x: {p: 0.01}
    rot_y: {p: 0.01}
    rot_z: {p: 0.01}

  gravity:
    x: 0
    y: 0
    z: -9.81

  tool:
    mass: 2.12
    com_x: -0.0242
    com_y: 0.0
    com_z: 0.0923
```

captain-yoshi, 2022-08-19:
One thing that is weird is that applying a `target wrench` of `+2N in the Y axis`, the robot moves wrt. the `sensor ref` frame and not the `compliance ref` frame.

[image]

kurshakuz, 2022-08-22:
> But when I do this and start the compliance controller the robot starts shaking. Maybe I got to set the target frame to the actual end effector link...

Did you try to decrease the `error_scale` in the solver configuration? Then you can try increasing it step by step and look for the value that suits you. It helped me to reduce the shaking as well. Take a look at that [comment](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/47#issuecomment-1033979286).

In the meantime, I would like to also understand how to constraint the robot movement to one axis only. I was too able to set up the force control properly (proper compliance in desired direction), but the end effector drifts away in orthogonal direction when pushing a hard and flat object.

captain-yoshi, 2022-08-23:
> Did you try to decrease the error_scale in the solver configuration?

Thx that did help! I guess the error_scale controls the speed of the robot for equilibrium.

There is no problem with setting the `error_scale` to 1 in my system. But strangely, the first time that I start the 
`compliance_controller` there is a large spike of force before being at equilibrium. Subsequent stop and start of the controller does not have this behaviour. This seems like the mass/com is not properly configured. When pressing the UR3 Free Drive mode button, the robot does not move (so these parameters should be good, only the reference changes with the compliance_controller). So by setting the `error_scale` near 0 for initialisation it can get past those large force spikes. 

@stefanscherzinger Did you noticed large spikes when starting the compliance controller for the first time ?

> In the meantime, I would like to also understand how to constraint the robot movement to one axis only. I was too able to set up the force control properly (proper compliance in desired direction), but the end effector drifts away in orthogonal direction when pushing a hard and flat object.

I tested with a cardboard box and successfully applied constant force up to 20 Newtons without drifting. I put stiffness to 100 for the desired axis and 5000 in the other 2 axis. Torque was left with default values.

kurshakuz, 2022-08-23:
> There is no problem with setting the `error_scale` to 1 in my system. But strangely, the first time that I start the `compliance_controller` there is a large spike of force before being at [...]

I think the initial shaking behavior could also be due to the system going into unstable state without an ability to recover. I got into similar edge cases from time to time and tuning the `pd_values` could also help. I am not sure though how to properly configure them, but adding a small `d` value instead of zero helps a bit too. I think this should be the next resort after tuning the `error_scale` and `iterations` variables. Let me know if you have ideas how to config pd values.

> I tested with a cardboard box and successfully applied constant force up to 20 Newtons without drifting. I put stiffness to 100 for the desired axis and 5000 in the other 2 axis. Torque was left with default values.

I have a brief question regarding your experiments. Did you notice your end effector position shift (goes down) over a couple of touch-and-go-back iterations? I made a simple script that goes to a goal pose, applies some force and goes back to initial position, and noticed that over time the robot does not go to initial position exactly (drifted downward). Just curious if something is not correct with my setup

KKSTB, 2022-09-03:
> There is no problem with setting the error_scale to 1 in my system. But strangely, the first time that I start the
compliance_controller there is a large spike of force before being at equilibrium. Subsequent stop and start of the controller does not have this behaviour. This seems like the mass/com is not properly configured. When pressing the UR3 Free Drive mode button, the robot does not move (so these parameters should be good, only the reference changes with the compliance_controller). So by setting the error_scale near 0 for initialisation it can get past those large force spikes.

The initial jump on UR robot is caused by the joint command position of 0,0,0,0,0,0 left in the UR ROS driver, or any value that was left behind. To look into the problem you need to understand both the UR ROS driver and `ros_control/controller_manager`.

First, take a look at UR ROS driver node:
https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/hardware_interface_node.cpp
Near the end is a while loop performing ordinary `RobotHW` task. Its doing the job in this sequence:
1. `read()`
2. `cm.update()` (cm is `ControllerManager`)
3. `write()`

Then lets dig into controller_manager.cpp `ControllerManager::update()`:
https://github.com/ros-controls/ros_control/blob/noetic-devel/controller_manager/src/controller_manager.cpp
Its doing the job in this sequence:
1. reset controller if necessary (not related to our problem)
4. calls `updateRequest()` which calls `update()` of the active controllers (i.e. `update()` of any active cartesian controllers)
5. calls `manageSwitch()` which calls `starting()` or `stopping()` of controllers (i.e. `starting()` or `stopping()` of cartesian controllers)

Because of this sequence, if the controller does not write position command in `starting()`, the leftover value in UR ROS driver will be `write()` to the robot. And since `joint_position_command_` in UR ROS driver are initialized to 0,0,0,0,0,0, the robot tries to move to 0,0,0,0,0,0 for the first sample, and then back to the current position at the next sample. This causes a jump. Or if the robot was manually moved to another position before starting controller, the robot jumps to the previous `stopping()` position, and then back to the current position at the next sample.

To solve it, I think we may add `writeJointControlCmds()` at the end of `CartesianControllerBase::starting()` to write the starting positions from `m_ik_solver->setStartState()` to the command positions.

captain-yoshi, 2022-09-04:
Wow thanks for your detailed hypothesis ! 
My use case was manually moving the robot, and then starting the compliance controller so I think you're right.

> To solve it, I think we may add writeJointControlCmds() at the end of CartesianControllerBase::starting() to write the starting positions from m_ik_solver->setStartState() to the command positions.

Il try this next week and report back.

> I have a brief question regarding your experiments. Did you notice your end effector position shift (goes down) over a couple of touch-and-go-back iterations? I made a simple script that goes to a [...]

Did not have time to test this yet, but will report back when I do. Could it be that over time your force torque sensor drifts ?

kurshakuz, 2022-09-04:
> Did not have time to test this yet, but will report back when I do. Could it be that over time your force torque sensor drifts ?

Yes, it indeed drifts over time and I have not figure out how to overcome it when working with the compliance controller. I instead switched to using a combination of MoveIt and force control for my application, which works as expected.

stefanscherzinger, 2022-09-09:
Thanks @KKSTB for these insights!

> To solve it, I think we may add writeJointControlCmds() at the end of CartesianControllerBase::starting() to write the starting positions from m_ik_solver->setStartState() to the command positions.

That sounds like a safe default indeed. I remember that I did that in the hardware interfaces for the driver's that I developed recently. But it does not harm to have that additionally in the controllers. Would you mind opening a separate issue for this and add your detailed explanation so that is conserved and better visible for other users? I'll then target this with a PR.

stefanscherzinger, 2022-09-09:
@captain-yoshi @kurshakuz Thanks for continuing to get things running! I'm back from vacation now.

> Yes, it indeed drifts over time and I have not figure out how to overcome it when working with the compliance controller.

What we normally do is working is short sessions of several seconds to few minutes with compliance control, then switch it off (do some motion planning etc.,), then zero the sensor and go back into compliance. It's relatively important to always zero the sensor measurements before switching force-based control on. Just don't do that in contact or you will add the contact forces as offset into the reference profile.

kurshakuz, 2022-09-09:
> What we normally do is working is short sessions of several seconds to few minutes with compliance control, then switch it off (do some motion planning etc.,), 

That's exactly what I try to do now, thanks for reaffirming :)

> then zero the sensor 

Can you please suggest what is the best way to do that? Is there a corresponding service that I can call during controller switching? I am using UR10e with an FT sensor for my tasks.

stefanscherzinger, 2022-09-09:
@kurshakuz If I'm not mistaken, we normally use [this service](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#zero_ftsensor-std_srvstrigger) provided by the driver. Also note the answers from [this issue](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/444) about the different control modes.

captain-yoshi, 2022-09-09:
Just a quick snippet for the ur3 with a robotiq FTS-300. There maybe some additional error checks missing

```cpp
#include <ros/ros.h>

#include <robotiq_ft_sensor/sensor_accessor.h>
#include <std_srvs/Trigger.h>
#include <controller_manager_msgs/SwitchController.h>

int main(int argc, char** argv)
{
  ros::init(argc, argv, "open_lid_biorad_t100");
  ros::AsyncSpinner spinner(1);
  spinner.start();

  ros::NodeHandle nh;

  // initialization
  ros::ServiceClient sensor_client =
      nh.serviceClient<robotiq_ft_sensor::sensor_accessor>("/robotiq_ft_sensor/robotiq_ft_sensor_acc");
  ros::ServiceClient controller_client =
      nh.serviceClient<std_srvs::Trigger>("/my_cartesian_force_controller/signal_taring");
  ros::ServiceClient ctrl_manager_client =
      nh.serviceClient<controller_manager_msgs::SwitchController>("/controller_manager/switch_controller");

  sensor_client.waitForExistence(ros::Duration(1));
  controller_client.waitForExistence(ros::Duration(1));
  ctrl_manager_client.waitForExistence(ros::Duration(1));

  // reset fts
  robotiq_ft_sensor::sensor_accessor req;
  robotiq_ft_sensor::sensor_accessor resp;
  req.request.command_id = req.request.COMMAND_SET_ZERO;

  if (!sensor_client.call(req) && !req.response.success)
  {
    ROS_ERROR("Failed to reset sensor");
    return 0;
  }

  // tare signal from force controller
  std_srvs::Trigger trigger;
  if (!controller_client.call(trigger) && !trigger.response.success)
  {
    ROS_ERROR("Failed to tare signal from force controller");
    return 0;
  }

  // start force controller
  controller_manager_msgs::SwitchController switch_ctrl;
  switch_ctrl.request.start_controllers = { "my_cartesian_force_controller" };
  switch_ctrl.request.timeout = 1.0;
  switch_ctrl.request.strictness = switch_ctrl.request.STRICT;

  if (!ctrl_manager_client.call(switch_ctrl) && !switch_ctrl.response.ok)
  {
    ROS_ERROR("Failed to start force controller '%s'", switch_ctrl.request.stop_controllers[0].c_str());
    return 0;
  }

  ROS_INFO("Successfully started !");
}
```

@stefanscherzinger Is it possible to tare the signal and reset the fts while the force/complaince controller is running ? Granted that the user does so when not in contact.

stefanscherzinger, 2022-09-12:
@captain-yoshi 

> Is it possible to tare the signal and reset the fts while the force/compliance controller is running ? Granted that the user does so when not in contact.

Yes, that should be possible. The force/compliance controller won't receive new messages (sensor readings) for some seconds, but that's no problem. As you mentioned, you should not be in contact because then the robot might move unhindered into the surface and might trigger the robot's protective stop.

stefanscherzinger, 2022-09-20:
@captain-yoshi 

I think we addressed everything in this issue. Feel free to re-open if desired.

captain-yoshi, 2022-09-28:
@stefanscherzinger

Retried the compliance controller with #78 and even #41 like some others suggested but the ur3 goes into protective stop. One thing I noticed is I think I have time to stop the robot with the emergency button now... The force controller does not have this behaviour.

There is no data going through `/my_cartesian_compliance_controller/target_frame` neither `/my_cartesian_compliance_controller/target_wrench`.

Is it possible that the robot needs a little time to adjust to the unusal end-effector configuration which has 2.23 kg of mass and COM ? Maybe with my setup it is normal to set the `error_scale` lower then the default 1 ?

stefanscherzinger, 2022-09-28:
> Is it possible that the robot needs a little time to adjust to the unusal end-effector configuration which has 2.23 kg of mass and COM ?

I wouldn't think so, no.

> Maybe with my setup it is normal to set the error_scale lower then the default 1 ?

Sure, that might be necessary for some configurations. Start with `0.0` and steadily increase that until the system becomes unstable. This gives you a good feeling for the value. Then set it to something that works in your case. I saw stable controllers with `error_scale = 1.5` and unstable ones with `error_scale = 0.05`. It all depends on the robot and driver. It's on my list, though, to make that more *universal*. 

Note that you can set the `error_scale` in the `.yaml` file for a safe start:

```bash
  ...
  target_frame_topic: "target_frame"
  joints: *robot_joints

  solver:
      error_scale: 0.0
```

captain-yoshi, 2022-09-28:
Would you recommand to keep the gravity parameter and then find the sweet spot for the error_scale ?

stefanscherzinger, 2022-09-28:
> Would you recommand to keep the gravity parameter and then find the sweet spot for the error_scale ?

I'm thinking of dropping the gravity compensation until you can confirm that this really balances the controller's drift during orientations. I'm not sure if we need [these suggestions](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/39#issuecomment-1238889112) before things work as expected. Unfortunately, I didn't have time so far to test these suggestions.

captain-yoshi, 2022-09-29:
So tested the `error_scale` with no gravity compensation.

Started to `0.0` and incremented by `0.1` to a final value of `2.2`. After each increment, I manually moved the robot and it was moving as appropriate (returning to the initial pose when not pushed by me). I think I could have gone up more, but did not test past `2.2`.

So I did a little script that starts the error_scale at 0.0 (could have set this directly in the config file), then switches to 1.0 after the start of the compliance controller:
```cpp
  // initialization
  ros::ServiceClient sensor_client =
      nh.serviceClient<robotiq_ft_sensor::sensor_accessor>("/robotiq_ft_sensor/robotiq_ft_sensor_acc");
  ros::ServiceClient controller_client =
      nh.serviceClient<std_srvs::Trigger>("/my_cartesian_compliance_controller/signal_taring");
  ros::ServiceClient ctrl_manager_client =
      nh.serviceClient<controller_manager_msgs::SwitchController>("/controller_manager/switch_controller");

  // dynamic reconfigure
  dynamic_reconfigure::Client<ControllerConfig> dyn_reconf_ctrl_client("/my_cartesian_compliance_controller/solver");

  ...

  // reset fts
  ...

  // reconfigure the cartesian controller
  ControllerConfig ctrl_dyn_config;
  ctrl_dyn_config.error_scale = 0.0;

  if (!dyn_reconf_ctrl_client.setConfiguration(ctrl_dyn_config))
  {
    ROS_ERROR("Failed to reconfigure the cartesian configuration");
    return 0;
  }

  // tare signal from force controller
  ...

  // start force controller
  ...

  // reset the error_scale to default value
  ctrl_dyn_config.error_scale = 1.0;

  if (!dyn_reconf_ctrl_client.setConfiguration(ctrl_dyn_config))
  {
    ROS_ERROR("Failed to reconfigure the cartesian configuration");
    return 0;
  }
```

The above works which means there is something wrong when first calculating the error, either the robot drive or the controller... I could print out some debugging info, but not sure where exactly, can you provide any hints ?

Using the https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/commit/696cbb334a0b76a2d18e26994adc158768ddba0b for the UR3.

stefanscherzinger, 2022-09-29:
> The default error_scale = 1.0 is not the problem here, I can make it work as long as I don't use the Freedrive mode.

That's interesting. The controller seems to fetch outdated joint states at some point. I have the feeling it could be gravity compensation. @KKSTB mentioned something [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/39#issuecomment-1238889112):

> In signalTaringCallback(), there are calls to Base::displayInTipLink() and Base::displayInBaseLink(). These functions use m_ik_solver->getPositions(). So the tare function does not get the correct joint positions if the controller is not running. So if (!Base::isRunning()), it should call Base::m_ik_solver->setStartState(Base::m_joint_handles, Base::m_joint_handle_map); to update joint positions.

--- 

> How could I debug further ?

It would be of great help if you could test these fixes. I'll see if I can get my hands on a URe and test this as well.

captain-yoshi, 2022-09-29:
I'm on a mission and will be able to tackle this today!

I first tested https://github.com/KKSTB/cartesian_controllers/commit/894a341ccbeaefcf68faa9e77538081e6710dbb9 which made things worse for me, e.g. the robot was oscillating without manually moving the robot. Without it, the controller is in equilibrium.

I will first test by removing the gravity compensation and then try the fix proposed by @KKSTB.

captain-yoshi, 2022-09-29:
This is the first sample of errors when starting the controller.

| error        | x            | y            | z           | x'        | y'       | z'           |
|--------------|--------------|--------------|-------------|-----------|----------|--------------|
| motion       | 0            | 0            | 0           | 0         | 0        | 0            |
| force        | -0.0589155   | -0.00968888  | -0.0416557  | -0.911579 | 0.518378 | -0.00264155  |
| gravity comp | -6.43946e-15 | -1.24102e-14 | 2.84217e-14 | -0.918453 | 0.53953  | -1.37694e-17 |
| compliance   | -0.0589155   | -0.00968888  | -0.0416557  | -0.911579 | 0.518378 | -0.00264155  |

It seems that the gravity compensation is not valid for torques...

captain-yoshi, 2022-09-29:
I can confirm that by disabling gravity compensation the system works. Moving manually and after an e-stop.

When using the `force controller` it was all good, but with `compliance controller` the gravity compensation made the robot move which rised the error of motion and eventually the system becamed unstable.

KKSTB, 2022-09-30:
I finally got a chance to create a PR of #39. Lets see if it works. Thanks.

captain-yoshi, 2022-09-30:
Thanks @KKSTB. I tried your solution but when moving the robot through the teach pendant before starting the controller, the robot oscillates as if trying to reach the previous starting pose. So I think my error, has nothing to do with gravity compensation... and cannot test your proposed fix at the moment.

> That's interesting. The controller seems to fetch outdated joint states at some point.

@stefanscherzinger Yeah it really seems that way. Even though the motion errors are all equal to zero ! I will double check that (not sure if it's the force or the motion controller that introduces the vibrations) and plot some graphs.

captain-yoshi, 2022-09-30:
Ok so I can correlate that the large spikes of forces comes from the `ur driver` loss of connection to the reverse interface. I do my tests on a ros-noetic PC without a real-time kernel. Maybe related to https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/507.

[image]

I got too check why it's dropping (wired gigabit interface). 

How should this driver behave when there is such a drop, should it stop ? Can it retrieve this information ?

captain-yoshi, 2022-09-30:
There is hope! 

So actually I had the `Universal_Robots_Client_Library` not up to date and another one. After fixing this there were no more interface drops.

The second problem was fixing the `error_scale = 0.01`. It was to low for my setup. The sweet-spot is around `0.5`. With this setting, the robot is always in equilibrium (must have tested 20 times and manually moving the arm and from e-stops).

ZZWang21, 2022-10-08:
Excuse me @captain-yoshi , I am doing the similar test. But can I know how you config the Moveit? Or how did you start the "cartesian_force_controller"? 

I am trying to start the controller, but the error:"Waiting for my_cartesian_force_controller/follow_joint_trajectory to come up
[ERROR] [1665219331.933675743]: Action client not connected: my_cartesian_force_controller/follow_joint_trajectory
"

Thank you very much.

ZZWang21, 2022-10-09:
Excuse me @captain-yoshi , how to find the "target_frame_topic" please. Thank you very much.   @stefanscherzinger @kurshakuz

captain-yoshi, 2022-10-09:
@ZZWang21 Did not try with MoveIt yet, but you can view #4 for more details...

 

> how to find the "target_frame_topic"

You can find this topic by using `rostopic list` in your terminal. By default, it should be `/my_compliance_controller/target_frame_topic`.

ZZWang21, 2022-10-10:
Thank you @captain-yoshi  for the reply!

so, I just need to set in the controllers.yaml :" target_frame_topic: "target_frame" "?

And, as you mentioned, there is no data through " /my_cartesian_compliance_controller/target_frame,  neither /my_cartesian_compliance_controller/target_wrench", why? 

as I started the my_cartesian_complian_controller, the error came out "[ERROR] [1665385510.916529066]: Pipeline producer overflowed! <RTDE Data Pipeline> "

and the robotic arm is shaking and then go to protection mode. emergency stop. I would like to sort this out. 

Thank you very much.

stefanscherzinger, 2022-10-11:
@ZZWang21 

I saw you opened other issues with questions on setting up Moveit and the `cartesian_compliance_controller`. Let's discuss solutions there (and in dedicated issues that you further open). This thread is quite packed and touches several topics already.

stefanscherzinger, 2022-10-11:
@captain-yoshi 

> The second problem was fixing the error_scale = 0.01. It was to low for my setup. The sweet-spot is around 0.5. With this setting, the robot is always in equilibrium (must have tested 20 times and manually moving the arm and from e-stops).

Many thanks for your continuous motivation and effort in getting your setup running and for the debugging (with plots)! Interesting to see how loss of connection to the driver could cause instability.

You mentioned that your use case is running now, right? Concerning the gravity compensation, are things working as expected and was the fix by @KKSTB required? (Unfortunately I didn't find the time to test this myself yet).

captain-yoshi, 2022-10-12:
> Concerning the gravity compensation, are things working as expected and was the fix by @KKSTB required?

I'm planning to do more tests this week and the gravity compensation will be on my list. Will update after testing.

I guess forcing the controller to rotate with the `target_frame` will be enough to see if the compensation will work ?

stefanscherzinger, 2022-10-12:
> I guess forcing the controller to rotate with the target_frame will be enough to see if the compensation will work ?

Yes, that should show if the rotation generates offsets. I might be a little difficult to recognize this with the compliance controller and a high stiffness, though. An alternative is to use the force controller and move the robot by hand. If it starts drifting after rotation, then the compensation does not yet work. If it drifts only a little, but otherwise seems to improve things, then maybe the specification of the weight is not accurate enough.

LordAcrobaticTurtle, 2022-10-14:
Hi guys

I've been using cartesian-controllers for my undergrad thesis @stefanscherzinger , it is wonderful thank you. 

@captain-yoshi, just wondering how you modified the kinematic tree for your UR robot? I haven't been able to work it out nor find any resources to help.
Hopefully I attached this image correctly, but this shows the current Tf tree. My TCP position is linked to "tool0_controller", so I assume I need to move it to the end of the other branch, but I am unsure how. 
I'd greatly appreciate any advice.
Thanks
[frames.pdf](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/files/9782909/frames.pdf)

stefanscherzinger, 2022-10-14:
@LordAcrobaticTurtle 

I frequently specify `tool0` for the end-effector on UR robots if nothing else is mounted. Here's [a config](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/76#issuecomment-1247028129) for force control. If your UR robot is calibrated correctly, `tool0_controller` and `tool0` should be identical as mentioned [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/458#issuecomment-909194525). The difference is that you can use the latter as a valid URDF link to build your robot description with additional tools, mounts, etc.

LordAcrobaticTurtle, 2022-10-19:
@stefanscherzinger 

Thanks for that. 
I'm actually using the compliance controller but I believe the controller yaml file I have matches what you've shown me. 
I have run ur_calibration and I pass the generated yaml file to the ur_robot_driver when I launch the launch it using this command
roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.0.100 kinematics_config:="/home/mtrn4230/repos/Thesis_robotic_drilling/calibration.yaml"

However when I inspect the frames using rosrun tf2_tools echo.py, tool0 to tool0_controller does not give an identity transform. 

Just for context, I'm using a UR5e arm. 

Thanks for helping out, do you have any further advice?

`

    my_cartesian_compliance_controller:
        type: "position_controllers/CartesianComplianceController"
        end_effector_link: "tool0"    # All links below must come before this link
        robot_base_link: "base_link"
        ft_sensor_ref_link: "tool0"
        compliance_ref_link: "tool0"
        target_frame_topic: "target_frame"
        joints:
         - shoulder_pan_joint
         - shoulder_lift_joint
         - elbow_joint
         - wrist_1_joint
         - wrist_2_joint
         - wrist_3_joint
        
        stiffness:  # w.r.t. compliance_ref_link
            trans_x: 500
            trans_y: 500
            trans_z: 500
            rot_x: 100
            rot_y: 100
            rot_z: 100
        pd_gains:
            trans_x: {p: 0.05}
            trans_y: {p: 0.05}
            trans_z: {p: 0.05}
            rot_x: {p: 0.01}
            rot_y: {p: 0.01}
            rot_z: {p: 0.01}
        solver:
            error_scale: 0.05
            iterations: 10

`

[image]

[image]

stefanscherzinger, 2022-10-19:
@LordAcrobaticTurtle

> Thanks for helping out, do you have any further advice?

I would try specifying the TCP that you taught in the pendant also as URDF Link with respect to `tool0`. Maybe it's the exact transform that you get when looking up TF between `tool0_controller` and `tool0` (also check the other way round), but that's a guess at the moment.

Your controller config seems a little outdated. [Here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/master/cartesian_compliance_controller#example-configuration)'s the new default. If you get oscillations in stiff contacts, decrease the `error_scale` a little bit. You can also do this online via dynamic reconfigure (it's a service call and easy to call from e.g. python scripts).

---

If you have follow-up questions, I suggest opening a new issue and re-post your initial question. This thread is quite packed already and won't be a good reference for future users.

stefanscherzinger, 2022-10-19:
@captain-yoshi Please open a new issue if/when there's further updates from your side on the gravity compensation. I'll close this thread here since we addressed the initial questions.

closed 2022-10-19 by stefanscherzinger

KKSTB, 2024-10-17:
Hi @shrutichakraborty the parameter `hand_frame_control` should do what you want. If it is True, the `target_wrench` topic force direction follows `end_effector_link` frame direction (i.e. Tool frame). If it is False, the `target_wrench` topic force direction follows `robot_base_link` frame direction. `hand_frame_control` is True by default, so maybe you have other problem?
Then you have to ensure FT sensor data is fed to `ft_sensor_wrench` topic, and force command is fed to `target_wrench` topic, and enable the force controller in controller manager.

shrutichakraborty, 2024-10-18:
> Hi @shrutichakraborty the parameter `hand_frame_control` should do what you want. If it is True, the `target_wrench` topic force direction follows `end_effector_link` frame direction (i.e. Tool [...]

Thanks a lot! When publishing target wrench it should be published in end effector frame then? I think that was the error I made..