fzi-forschungszentrum-informatik/cartesian_controllers #77: Arm moving unexpectedly when starting Cartesian Compliance/force controllers

mbs-tahir, 2022-09-06:
I have integrated an XARM6 with Cartesian Controllers and Robotiq Force Torque Sensor FT300. The problem I am facing is the arm is moving unexpectedly and sometimes crashing into the pallete. For this reason I have set smaller values for error_scale and PD. I have tried tuning the parameters alot but nothing helped this startup issue. I have recorded a some videos and a rosbag [here](https://my.hidrive.com/share/k9g37p4n4e). In controller folder is the initial behavior and in parameter tuning is the video for smaller values of PD and error_scale. The rosbag is for the same video i.e. parameter tuning. Any idea on how this problem can be solved? 
To me I think something is wrong with the force torque sensor values? May be I need to zero them at the start of controller for reference purpose?
Any ideas will highly be appreciated.

mbs-tahir, 2022-09-12:
Hi @stefanscherzinger 
Thank you so much for your reply and helping out. 

> Could you zero the sensor and report if the effect still happens?
It worked!

Robotiq FT300 have a service, so after calling: 

```
rosservice call /robotiq_ft_sensor_acc  "command_id: 8 command: ''" 
```

and 

` rosservice call /cartesian_compliance_controller/signal_taring "{}" `

Than initiating the controller the robot never moves and it works fine. 

I have one more question about controller behavior and configuration. When an external force is applied the robot reacts against this applied force applied and after force removal maintains its position. In my use case scenario the robot has to move the end effector on a surface which shouldn't be smooth. For example human arm touching from one end to another which can have different up and downs from one point to another. 

So the robot behavior is fully opposite I want the robot to get back from an uneven terrain however the robot exerts force on that. Can you give some points by following which I can have this expected behavior?

stefanscherzinger, 2022-09-12:
Hi @mbs-tahir 

> It worked!

I'm glad to hear that!

> Can you give some points by following which I can have this expected behavior?

The sensitive touch use case that you describe is certainly challenging with an industrial robot. The force-torque measurements will suffice, but the robot control frequency is relatively low and will require *slow* motion for stability and safety reasons. Moving up and down a human arm sounds to me like following a path compliantly and being especially compliant in the orthogonal (pushing downwards to the human arm) dimension. You could try setting a low stiffness in that direction and apply an additional pushing force downward of few newtons, for instance `z = 10 N`. You'll need some careful testing, but I'm excited to sew how that goes!

> I want the robot to get back from an uneven terrain however the robot exerts force on that. 

That should be doable with a low stiffness in that direction and an additional downward force as described above. Force reference setpoints are just overplayed with the virtual spring. If you have a low virtual stiffness, the applied target forces will compensate for the uneven terrain. The challenge is that this will work relatively slowly, though. You might also want to mitigate that the end-effector gets stuck somehow in the tissue of the sweater or similar by designing a suitable *brush* (no sharp edges, possibly a round underside?).

stefanscherzinger, 2022-09-16:
Hi @mbs-tahir
Thanks for posting your progress on this!

It's some time that I have used this feature (turning joint trajectories into a moving target for the `cartesian_controllers`). I remember, though, that they were not entirely robust towards arbitrary controller changes. I ran into freezes sometimes and wanted to fix this in #41 . 

Concerning the zeros that get published in the target frame from time to time: That's definitely incorrect behavior and a severe safety risk. I want to fix this, but need to reproduce this for debugging in a simple setup. Thanks to @matthias-mayr for spotting the [dangling robot chain](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/commit/6500df9c3d7e00a9cdcf31b41f20bc8616b64f33). Maybe that's somehow related.

@mbs-tahir Could you try again on the [improve moveit compatibility branch](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/improve-moveit-compatibility)?

I'll rethink this architecture. The `joint_to_cartesian_controller` should *always* publish the current robot pose as default. Something seems to initialize incorrectly.

stefanscherzinger, 2022-09-17:
**Update**:
The `joint_to_cartesian_controller` already updates its internal representation in [starting](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/joint_to_cartesian_controller/src/joint_to_cartesian_controller.cpp#L178), so the published target *should* be valid, provided the robot driver fills those buffers with meaningful values.

I'll try to reproduce this somehow.

Maybe also #78 is related to this issue.

stefanscherzinger, 2022-09-20:
@mbs-tahir

You might try again on the [improve-moveit-compatibility](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/improve-moveit-compatibility) branch. I rebased onto #78.

mbs-tahir, 2022-09-20:
> You could try setting a low stiffness in that direction and apply an additional pushing force downward of few newtons, for instance z = 10 N.

@stefanscherzinger I tried setting low stiffness and published a constant push in upward direction but still its not that compliant. Even I tried with 1N force in z-axis. Just to confirm I need to publish to `cartesian_compliance_controller/target_wrench` right?

Moreover to add here I am not planning in the frame of `ft_sensor` but in `link_eef` but they both are same. For reference I have recorded a bag file and a video which can be found [here](https://my.hidrive.com/share/k9g37p4n4e#$/) in `controller_parameter_tuning`. Any comments/guidelines will highly be appreciated.

stefanscherzinger, 2022-09-20:
> I need to publish to cartesian_compliance_controller/target_wrench right?

I'm not sure of your current remappings. If you didn't remap anything, then it looks correct. You could do a `rostopic list | grep target_wrench` and check how many `target_wrench` are there and if they are connected.

> but still its not that compliant. 

Your `error_scale` seems very low. Could you carefully increase that a little bit? Your system could be much more responsive, the only limiting factor would be oscillations in contact. Unfortunately, that's a classic trade-of. You might also lower your `trans_x` and `trans_y` stiffness a little bit. The rotational stiffness is also quite high.

I updated the [exemplary compliance controller](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2/cartesian_compliance_controller#example-configuration) config yesterday after testing a little on a UR5e (it's for ROS2, but that should be similar).
Maybe you want to try that. But start with your small `error_scale = 0.02` and gradually increase that because that's hugely different from robot to robot.

stefanscherzinger, 2022-09-20:
> Is it normal behavior if so can you please explain why does this happen?

Yes, that's normal behavior. The compliance controller computes the overall net force, consisting of `target_wrench`, `ft_sensor_wrench` and wrenches induced by offsets to the `target_pose`. This behavior is a design choice. I like its simplicity and the possibility to do some force control in directions orthogonal to the nominal motion. The fact that the controller's stiffness *balances* the `target_wrench` until an equilibrium is established, is fine for most use cases.

closed 2022-09-24 by stefanscherzinger