fzi-forschungszentrum-informatik/cartesian_controllers #92: How to config the Moveit for real robot for a cartesian_force_controller

ZZWang21, 2022-10-08:
Hi Experts, 

I am using ROS1, trying to control a real UR5e robot with the cartesian_force_controller. 

I am trying to realise a test of surface scanning with constant contact force of 5N. 

I tried to catkin_make the controller into my workspace. insert the controller in my robot launch file. But the Moveit launch is also neede.  the example: 

"joint_state_controller:
  type: joint_state_controller/JointStateController
  publish_rate: 50
controller_list:
- name: "scaled_pos_joint_traj_controller"
  action_ns: follow_joint_trajectory
  type: FollowJointTrajectory
  joints:
    - ......"

So, what is the "action_ns" and "type" of cartesian_force_controller? 

Thank you everyone. 

@fzi-admin  @fmauch @stefanscherzinger

ZZWang21, 2022-10-09:
And I tried the cartesian_force_controller. It seems I can only set the target_wrench, which is the target force. Can I set waypoints of the robotic arm using python? 

I used to use "scaled_pos_joint_traj_controller/follow_joint_trajectory" which has moveit config, so that I can use python code to control the trajectory. But since the cartesian controller is not configed in Moveit, I cannot use my original code. 

Thank you.

fmauch, 2022-10-10:
The cartesian_force_controller is a controller for controlling a certain force on a robot's tcp. If you want to do motions, you'd have to use the cartesian_compliance_controller. This will take a pose as an input, which you will have to publish yourself. It does not work out of the box with MoveIt, since it is basically missing the trajectory interpolation and pose publishing step that would be needed to fill the gap.

ZZWang21, 2022-10-11:
Thank you very much @fmauch  for your reply. I have tried the compliance controller, and I had a python code which is based on Moveit, so I would like to use cartesian+moveit. I saw @stefanscherzinger  has done some work on combining cartesian controller with moveit. I am trying to figure it out.

stefanscherzinger, 2022-10-11:
Hi @ZZWang21 

> I saw @stefanscherzinger has done some work on combining cartesian controller with moveit. I am trying to figure it out.

Yes, there's a workaround with [this mechanism](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/improve-moveit-compatibility/joint_to_cartesian_controller) that turns joint-based motion into a moving target for the `cartesian_compliance_controller`. You can use that and apply a contact force of `5 N` on top of your motion, provided you manage to move the robot above some surface. Please use [this branch](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/improve-moveit-compatibility) while working with the `joint_to_cartesian` controller. We fixed a dangling reference that might have caused segfaults and haven't merged that yet.

> And, how should the "xx-moveit_config" package be modified? I tried to set up the joint-to-cartesian controller, but the moveit cannot find any controller.

That's a good question. I'll try to set-up a documentation for this (need to get Moveit running myself and see how things work). It's probably a matter of namespaces. The `joint_trajectory_controller` that Moveit should connect to, will be the one from the `joint_to_cartesian` controller.

Note that turning joint-based trajectories into Cartesian targets is *not* robust to singularities as mentioned [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/4#issuecomment-714649920). Try to combine force control and Moveit for local motion in which you don't expect drastic joint configuration changes and move with the normal `scaled_pos_joint_traj_controller/follow_joint_trajector` instead for wider workspace motion.

ZZWang21, 2022-10-12:
@stefanscherzinger  Hi, Stefan. Thank you very much for your reply!

After two days work, I have got the "my_joint_to_cartesian_controller/joint_trajectory_controller" on my rqt. However, I cannot move the real robot within the rqt-Joint Trajectory Controller, but I can move it with the other joint_trajectory_controller in normal namespace "controller_manager". I do not know where the setup is wrong. Do I have to load "my_cartesian_motion_controller“ and "my_motion_control_handle"? 

Could other expert @captain-yoshi  help? 

Thank you very much.

stefanscherzinger, 2022-10-12:
> Do I have to load "my_cartesian_motion_controller“ 

Yes, you also need a controller that follows the target that the adapter publishes. Try to achieve that in the simulation environment first. Here's a screenshot from the example setup. Note which controllers are running.

[image]

And here's the `joint_trajectory_controller` in the `my_joint_to_cartesian_controller`'s namespace:
[image]

ZZWang21, 2022-10-12:
Thank you for your information.  I will try this tomorrow morning. I tried a whole afternoon but I cannot move my robot with "joint_to_cartesian_controller". no matter  the robot is moving or static. 

But I noticed that you only used "motion_controller" "joint_to_cartesian_controller" "joint_state_controller". Is it possible to use "joint_to_cartesian"+"compliance_controller"? 

Thank you.

ZZWang21, 2022-10-13:
> Yes, you also need a controller that follows the target that the adapter publishes. Try to achieve that in the simulation environment first. Here's a screenshot from the example setup. Note which [...]

Hi @stefanscherzinger , I tried the example, to use the controllers shown in your figure. But the "joint_to_cartesian_controller" is still not working. And when I turn it on, the end-effector automatically went to the (0,0,0) point. You can see in the screenshot: 

[image]

Looking forward to your reply. Thank you.

stefanscherzinger, 2022-10-13:
Could you git pull the latest [master branch](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/master)? I pushed [this fix](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/commit/8a4ec9340fd2a170148d49ea48e6f1adb1487c5a) yesterday. That should resolve it.

ZZWang21, 2022-10-13:
> Could you git pull the latest [master branch](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/master)? I pushed [this [...]

Thank you. I have tried, BUT it is the same. Not working. I will attache the situation as below for you to check: 

[image]

stefanscherzinger, 2022-10-13:
> Is it possible to use "joint_to_cartesian"+"compliance_controller"?

Yes, that's also possible. Both the `cartesian_motion_controller` and the `cartesian_compliance_controller` follow the target frame as published by the `joint_to_cartesian_controller`.

ZZWang21, 2022-10-13:
> ```shell
> git rev-parse HEAD
> ```

Thank you. Yes, it works now. 

"git rev-parse HEAD".   It is : 8a4ec9340fd2a170148d49ea48e6f1adb1487c5a . 

Maybe since the last time it was not pulled, so the  "git diff" gave me no feedback. 

Yes, I have rebuilt the workspace with catkin_make .  Now the example works, I will work on my real robot now.

ZZWang21, 2022-10-13:
Dear experts, 

I met a problem that I do not know if other people had before. If I disconnect the communication between robot and PC, I move the robot manually, then I reconnect, with "my_cartesian_motion_controller" on, the robot will harshly go back to the previous position. That is dangerous and I checked several times, it happens all the time. 

[image]

Could any expert explain? @stefanscherzinger  @captain-yoshi

stefanscherzinger, 2022-10-13:
> There is an error: " Unexpected error: No trajectory defined at current time. Please contact the package maintainer."

I found that error message [here](https://github.com/ros-controls/ros_controllers/blob/noetic-devel/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L381) in the `joint_trajectory_controller`. Without being 100% certain, I could imagine that the `joint_trajectory_controller` that feeds into the `joint_to_cartesian_controller` for goal publishing is not stopped correctly when you interrupt the program on the teach pendant for moving the robot by hand. When playing the program again, the `joint_trajectory_controller` seems to be confronted with an overly long duration (the time of manual interference) and causes the error.

In addition, if the `joint_to_cartesian_controller` isn't stopped and started for manual intervention, then this controller probably continues to publish the old goal and the `cartesian_motion_controller` (if running), directly drives to that goal. Only starting and stopping should fetch the current joint state from the real hardware.

stefanscherzinger, 2022-10-14:
Does it still happen with the new fixes on [this branch](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/improve-moveit-compatibility)?

ZZWang21, 2022-10-14:
Thank you @stefanscherzinger , but the internet of our lab is down. I can only try the new branch later. 

1. I have another issue. I am trying compliance+joint_to controller. But when the target force is 10N in z direction, the end-effector stopped. I also tried 20N. It only moved to the table surface when the target is 30N. I modified the gravity compensation, -9.81 and 0. Nothing changed. 

2. And, according to yesterday's test, I can use moveit with the cartesian controller. But the movement of the robot is strange, for example, I set goal in x-axis direction, but it went to y-axis. I think this needs more tests. Let us figure the first question.

Thank you.

stefanscherzinger, 2022-10-14:
Regarding your `1.` question:
The `cartesian_compliance_controller` uses a virtual stiffness to balance motion offsets (from the target frame) with forces and torques (either from `ft_sensor_wrench` or from `target_wrench`). This means that the controller only moves into the force direction until the stiffness generates equally high restoring forces. You have now two options:

- Drive with the robot exactly into contact and apply `10N`. 
- Lower the `stiffness.z` so that your target forces is not *consumed* when compensating offsets of several centimeters.

Try guiding the robot a little by hand when using the `cartesian_compliance_controller`. This gives you a *feeling* for the stiffness that you specify and could also give hints concerning your `2.` question. For example, if you push in `x` and the robot moves in `y`, then something is wrong.

ZZWang21, 2022-10-15:
Hi @stefanscherzinger , firstly, I would say sorry that I posted in another issue. I will not do that again. 

For the suggestions you mentioned in last post. I have tried the 1. I drove the robot till it contact the table. Then, I relaunch the bringup launch file, and applied 10N to the target. It worked but the force was oscillating. Anyway, this is kind of not practical, the bringup file needs to be relaunched every time. 

 2. I have lowered the stiffness.z from 10 to 0, but there is no change. Set target_wrench force 10N, 20N, the robot stopped in the midway. Only when applying 30N, the robot can move to the surface. 

In compliance mode, the orientation of the end-effector will change when applied with target force. I will try to fix this by modifying the stiffness of rot_x,y,z. 

And, Is it possible to combine force_controller+joint_to_cartesian_controller? 

Thank you.

ZZWang21, 2022-10-17:
@captain-yoshi 

Hi, I am trying to combine force_controller+joint_to_cartesian_controller, to realise a surface scanning (travel on the surface according to the path I planned) with constant contact force in z-axis. 

1. I saw you have completed something(as shown in [video](https://drive.google.com/drive/folders/1kNhk7iVsJjqg-gAHRThBnH4p1LU73pz-?usp=sharing)), but as I discussed with stefan, only compliance_controller can be used with joint_to_cartesian_controller. Can I ask how did you realize that? 

2. Could you explain to me, what is "signal_taring"? I only see this in force_controller. 

Thank you very much!

stefanscherzinger, 2022-10-17:
@ZZWang21 

> I have lowered the stiffness.z from 10 to 0, but there is no change. Set target_wrench force 10N, 20N, the robot stopped in the midway. Only when applying 30N, the robot can move to the surface.

I assume that's because the stiffness specification in the `.yaml` is not yet correct. Try doing that in `rqt` dynamic reconfigure. This makes sure that you do not misspell the parameters (wrong parameter names will be ignored).
For instance, `stiffness.z` is not a valid parameter.
Here's an example config for your use case:
```yaml
    stiffness:  # w.r.t. compliance_ref_link
        trans_x: 500
        trans_y: 500
        trans_z: 10   # 10 N will move the robot 1 m until reaching an equilibrium
        rot_x: 20
        rot_y: 20
        rot_z: 20
```
> It worked but the force was oscillating

Try decreasing `error_scale` a little bit until the oscillations flat out. If tracking the `target_frame` lacks to much behind, then increase the stiffness in the respective axis a bit.

ZZWang21, 2022-10-18:
@stefanscherzinger 

Hi, Stefan, I have spent some time to tune the parameters in the controller.yaml, in Dynamic Reconfigure. I have got the optimised parameters for my ur5e robot, as you listed in the post. Stiffness: trans_x: 500, trans_y: 500, trans_z: 100, rot_x: 100, rot_y: 100, rot_z: 100. 

However, even with the optimised parameters, the contact force is not so stable. For example, I set the target as 10N, the actual force ranges from 8N to 11N. I am wondering, if I can make it better. 

And, I am trying to make the robot to draw a circle with the radius of 0.1m. However, it is not moving according to the plan-- it is moving in a circle, but smaller than the plan. Have you tested cases like this before? Is there any way to improve? 

Thank you very much.

stefanscherzinger, 2022-10-18:
> For example, I set the target as 10N, the actual force ranges from 8N to 11N. I am wondering, if I can make it better.

This might be due to oscillations in contact, which might not be directly visible to the human eye. Try decreasing `error_scale` a little bit more and check if this helps smoothing the force reference of `10 N`.

> However, it is not moving according to the plan-- it is moving in a circle, but smaller than the plan.

That's interesting. You could add a visualization for `geometry_msgs/PoseStamped` (marker style coordinate system) in RViz for the compliance controller's `target_frame`. You can then check if the controller drifts behind the target, which could explain the smaller circle. If that's the case, try increasing the `trans_x` and `trans_y` stiffness a little (e.g. `1000`). 

Could you post your complete compliance controller config here? I feel there's room for improvement.

ZZWang21, 2022-10-19:
@stefanscherzinger 

The config is like this: 

type: "position_controllers/CartesianComplianceController"
  robot_base_link: "base_link"
  end_effector_link: "tool0"
  ft_sensor_ref_link: "tool0"
  compliance_ref_link: "tool0"
  target_frame_topic: "target_frame"     
  joints: *robot_joints

solver:
    error_scale: 0.01
    iterations: 1

  stiffness:
    trans_x: 1000    #original 100 10:20221014 20221017:0 200. all 100. 
    trans_y: 1000
    trans_z: 100
    rot_x: 150     #10   2
    rot_y: 150
    rot_z: 150

  pd_gains:
    trans_x: {p: 1.0} #0.5  0.05 20221017:0.5,0.1. 
    trans_y: {p: 1.0}
    trans_z: {p: 2.0, d: 0.3}
    rot_x: {p: 10.0} #0.1  1.50
    rot_y: {p: 10.0}
    rot_z: {p: 10.0}

  gravity:
    x: 0
    y: 0
    z: 0  #-9.81

  tool:
    mass: 2.12
    com_x: -0.0242
    com_y: 0.0
    com_z: 0.0923

stefanscherzinger, 2022-10-24:
Concerning your `pd_gains`: I would recommend decreasing `trans_z` a little bit more. I think it should work better, if its `p` gain is lower than the other axes, for instance `trans_z : {p: 0.5, d: 0.05}`.

### General suggestions
The reason is the following: As a rule of thumb, the *higher* the gains, the *more responsive* is the system in that axes, but also tends to be more *unstable* in stiff contacts. If I understood your use case correctly, you plan to follow a moving target (x,y axes) mostly on a flat surface. So, while motion tracking needs higher gains to follow the target and not drift too much, the force control dimension does not regulate much motion offset, it mostly regulates the target wrench but else stays in contact with the surface. So, it makes sense to decrease the gains in the force dimension (z axes). This allows to increase the `error_scale` a little until the motion tracking is sufficient (or increase their p gains). When the gains in the force dimension are higher, then the system might become unstable *before* good motion tracking is achieved. In the same argumentation, the contact of the surface in z is more challenging for the system (because of the stiffness of the contact) than the other axes in which you move. The stiffer the contacts, the lower should the gains be, and the more freely the system can move, the higher can the gains be.

stefanscherzinger, 2022-10-25:
@ZZWang21 please reopen if things are still unclear.

closed 2022-10-25 by stefanscherzinger

ZZWang21, 2022-10-25:
@stefanscherzinger  thanks. These days I am working on another project. So I did not try out the robot. I will reopen if things are still not clear.