fzi-forschungszentrum-informatik/cartesian_controllers #94: Questions about setting "joint_to_cartesian_controller"

ZZWang21, 2022-10-11:
Hello @stefanscherzinger , I have read the document about the "joint_to_cartesian_controller". Thank you for your work, this controller can surely solve problems for a lot of researchers. 

 but I am kind of lost since the document is not very clear. 

1. where should I put this: "my_joint_to_cartesian_controller/joint_trajectory_controller:
    type: "position_controllers/JointTrajectoryController"
    joints:
    - shoulder_pan_joint
    - shoulder_lift_joint
    - elbow_joint
    - wrist_1_joint
    - wrist_2_joint
    - wrist_3_joint "   ? In controllers.yaml  or somewhere else? 

2. Where should I put these? "   <node name="controller_spawner" pkg="controller_manager" type="spawner" args="--stopped my_joint_to_cartesian_controller" /> "  AND " <group ns="my_joint_to_cartesian_controller" >
                <node name="controller_spawner" pkg="controller_manager" type="spawner" args="joint_trajectory_controller" />
        </group>"?   Should I put them together in one launch for the new controller? In a seperated single launch file?

Thank you very much.

stefanscherzinger, 2022-10-11:
Hello @ZZWang21 

> Thank you for your work, this controller can surely solve problems for a lot of researchers.

Thanks! I'm not aware how many people are actually using this, so thanks for the feedback.

> where should I put this:

You can put the controller config in the config file of your driver, or you could also put that into a separate .yaml file and load that. Just make sure that you *load* the config file to the ROS parameter server as is done [here in the example](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_examples/launch/examples.launch#L38). You will then see a `joint_trajectory_controller` under the `my_joint_to_cartesian_controller`'s controller manager (namespace).

`rqt`'s [controller manager](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/45#issuecomment-1007307823) is really helpful there. You can inspect the available controllers in each namespace and can *load/unload* and *start/stop* everything via the GUI.

ZZWang21, 2022-10-12:
> Hello @ZZWang21
>
>
> Thanks! I'm not aware how many people are actually using this, so thanks for the feedback.
>
>
> You can put the controller config in the config file of your driver, or you could also [...]

Hi, @stefanscherzinger , after some hours check, I found that the /target_fram is always 0,0,0,0,0,0,  no matter how I change the "my_joint_to_cartesian_controller" in the rqt. I make the controller.yaml according to the example. the target_frame_topic is "/my_cartesian_motion_controller/target_frame", then, remap the target_frame from  "my_cartesian_motion_controller" to "target_frame".  But it is not moving.

 Could you please share some ideas? Thank you.

stefanscherzinger, 2022-10-13:
This is in parallel to your questions in #92. Let's keep the discussions there.

closed 2022-10-13 by stefanscherzinger