fzi-forschungszentrum-informatik/cartesian_controllers #95: How to control a real robot for a cartesian_force_controller

mouguo123, 2022-10-12:
Hi Experts,

I am using ROS1, trying to control a real  robot with this package.

I would like to know how to make the robot move by hand.

stefanscherzinger, 2022-10-12:
Hi @mouguo123 

You can use the [cartesian_force_controller](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/master/cartesian_force_controller#example-configuration) for this. When active, the robot moves by external forces and torques that are measured by the end-effector sensor. I frequently use that to guide the robot by hand. Increasing the `solver/error_scale` parameter (possibly in dynamic reconfigure) makes the system follow your motion faster.

To get started, put [this example config](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/master/cartesian_force_controller#example-configuration) into a `.yaml` file and load it to the ROS parameter server in one of your launch files. You can then *load* and *start* the controller via [rqt](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/45#issuecomment-1007307823).

You will also need to connect the controller's `ft_sensor_wrench` topic to the measurements of the sensor, e.g. with
```xml
<remap from="my_cartesian_force_controller/ft_sensor_wrench" to="wrench" />
```
for the *Universal Robots*.

ZZWang21, 2022-10-14:
Hi, @stefanscherzinger . After some tests, I found that with the cartesian_force_controller, the "joint_to_cartesian_controller" will not function. The robot cannot move under the force_controller. It can only move under the "compliance_controller", is that right? Or something is wrong with my setup. Thank you.

stefanscherzinger, 2022-10-14:
@ZZWang21 
Please open a new issue or stay in #92. The `joint_to_cartesian-controller` is not related to what @mouguo123 asked here. If these threads cover all sorts of topics, then it's difficult for users to find answers to their specific problems when searching the issues.

---
> it can only move under the "compliance_controller", is that right?

Yes, it works with the `cartesian_compliance_controller` because that takes `geometry_msgs/PoseStamped` as inputs.

stefanscherzinger, 2022-10-17:
@mouguo123

Please reopen if something is still unclear.

mouguo123, 2022-10-25:
Thank you for your guidance.、
I can Publish a geometry_msgs/WrenchStamped to /target_wrench to let robot move,
 but I can't guide the robot by hand.

stefanscherzinger, 2022-10-25:
That's probably because the sensor input is not yet configured (remapped) correctly.
Could you check the output of
```bash
rostopic list | grep wrench
```
and post that here in the chat? I use that followed by `rostopic info ...` to see if the `cartesian_force_controller` listens to the correct input topics.

mouguo123, 2022-10-25:
<remap from="robotic_ft_sensor_wrench" to="target_wrench" />
should I remap like this ?

mouguo123, 2022-10-25:
`<remap from="robotiq_ft_sensor_wrench" to="target_wrench" />`
should I remap like this ?

stefanscherzinger, 2022-10-25:
No. There are two different input topics: the `ft_sensor_wrench` which shall receive data from the force-torque sensor, and the `target_wrench`, which takes force-torque setpoints by the user. In the use case of manual guidance, `target_wrench == 0`, i.e. you don't explicitly publish to that topic.

Could you post the output of `rostopic list | grep wrench`? That gives us the names for correct remapping.

mouguo123, 2022-10-26:
/my_cartesian_compliance_controller/ft_sensor_wrench
/robotiq_ft_wrench
/target_wrench
/wrench

stefanscherzinger, 2022-10-26:
Then I would remap like this:

```yaml
<remap from="my_cartesian_compliance_controller/ft_sensor_wrench" to="robotiq_ft_wrench" />
```
I assume here that `rostopic echo /robotiq_ft_wrench` gives the data from your force-torque sensor. If you are using a `URe` robot (e-Series), I recommend to use their build-in sensor:
```yaml
<remap from="my_cartesian_compliance_controller/ft_sensor_wrench" to="wrench" />
```

[Here](https://github.com/stefanscherzinger/cartesian_controllers_ur_robots/blob/main/launch/driver.xml#L16) is an example that I use on *URe* robots.

gavanderhoorn, 2022-10-27:
off-topic, but:

@mouguo123 wrote:
> I am using ROS1, trying to control a real robot with this package.

and:

> [VID_20221027_204748.zip](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/files/9879796/VID_20221027_204748.zip) I did you said , but when I start cartesian_force_controller , robot will directly move like above video. it is very strange.

I hope that's not your real robot in that video :)

stefanscherzinger, 2022-10-27:
@mouguo123 

Could you paste your controller configuration here? I assume controller gains that are too high. Try decreasing the `error_scale` (e.g. via dynamic reconfigure) and see if this helps. Can you set the force-torque sensor signals to `0` before switching on the `cartesian_force_controller`? The controller will react immediately to whatever the sensor publishes when switching-on.

stefanscherzinger, 2022-11-04:
@mouguo123 

You could retry with the latest `master` branch. There's a dynamic reconfigure parameter `link_mass` that stabilizes joint oscillations when the robot is fully stretched. The callback was fixed [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/pull/102) and now works as expected.

You can change this value either in dynamic reconfigure online, or once at startup via the controller's  `.yaml` file:

```yaml
my_cartesian_force_controller:
    type: "position_controllers/CartesianForceController"
    end_effector_link: grasping_center
    robot_base_link: base
    ft_sensor_ref_link: tool0
    joints:
     - shoulder_pan_joint
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint

    solver:
        error_scale: 0.7
        iterations: 1
        forward_dynamics:
            link_mass: 0.10  # Higher values decrease oscillations in fully stretched configurations

    pd_gains:
        trans_x: {p: 0.05, d: 0.005}
        trans_y: {p: 0.05, d: 0.005}
        trans_z: {p: 0.05, d: 0.005}
        rot_x: {p: 1.50}
        rot_y: {p: 1.50}
        rot_z: {p: 1.50}
```

stefanscherzinger, 2022-11-04:
Please reopen if something is still unclear.

mouguo123, 2022-12-09:
Now I can guide the robot by hand, but  It moves to different direction compared to applied wrench,  is  robot kinematics description wrong?  Any suggestions on how to fix it?

this is my tf tree.
[image]

below is the definition of the controller.

  `my_cartesian_force_controller:
      type: "position_controllers/CartesianForceController"
      end_effector_link: "tool0"
      robot_base_link: "base_link"
      ft_sensor_ref_link: "sensor_link"
      joints:
      - joint1
      - joint2
      - joint3
      - joint4
      - joint5
      - joint6

    pd_gains:
        trans_x: {p: 0.05}
        trans_y: {p: 0.05}
        trans_z: {p: 0.05}
        rot_x: {p: 0.01}
        rot_y: {p: 0.01}
        rot_z: {p: 0.01}`

How shold I adjust it?

stefanscherzinger, 2022-12-09:
@mouguo123 

Please provide the following information:
 - **Commit**: What's the output of `git rev-parse HEAD` in the `cartesian_controllers` package?
 - **OS**: What's your Operating System?

mouguo123, 2022-12-10:
1. ba73b9798ea4a6d4b8729cb03a5573fe27088c20
2. ubuntu20.04

stefanscherzinger, 2022-12-10:
Please update to the newest `master` branch. There's a new feature ([via this PR](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/pull/75)) that would have caught the error at startup. 

The problem is that `sensor_link` is not part of the kinematic chain from `tool0` to `base_link`. Try to attach the subchain `Flange2 - Gripper - tool0` to the `sensor_link`. That should solve the issue.

closed 2022-12-12 by mouguo123