fzi-forschungszentrum-informatik/cartesian_controllers #105: Error while cloning this repo

marioesteban11, 2022-11-07:
Hi everyone,

Trying to clone your repository [cartesian controllerss](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers) the terminal says:

´´´
Cloning into 'cartesian_controllers'...
[git@github.com](mailto:git@github.com): Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
´´´

And I dont know if my github is wrongly configurated or i need any permision to clone it

stefanscherzinger, 2022-11-07:
Hi @marioesteban11 

That's interesting. I haven't heard that before and assume some issue with your ssh configuration. Here's a quick [checklist provided by github](https://docs.github.com/en/authentication/troubleshooting-ssh/error-permission-denied-publickey). If one of the answers covers your case, you could post your solution here to help other users.

stefanscherzinger, 2022-11-09:
Please reopen if this still not works for you.

closed 2022-11-09 by stefanscherzinger