fzi-forschungszentrum-informatik/cartesian_controllers #107: Build error

LucaBross, 2022-11-09:
**Problem description**
When i try to build `cartesian_controllers` i get a error with `cartesian_controller_tests`
```
Finished <<< cartesian_controller_base [26.3s]                        
Starting >>> cartesian_force_controller
Starting >>> cartesian_motion_controller                   
Starting >>> cartesian_controller_handles
Finished <<< cartesian_controller_handles [10.6s]
Finished <<< cartesian_motion_controller [14.1s]
Finished <<< cartesian_force_controller [14.4s]                         
Starting >>> cartesian_compliance_controller
Finished <<< cartesian_compliance_controller [11.4s]                         
Starting >>> cartesian_controller_tests
[52.649s] ERROR:colcon.colcon_cmake.task.cmake.build:Failed to find the following files:
- /home/luca/workspace/ur_control/install/cartesian_controller_simulation/share/cartesian_controller_simulation/package.sh
Check that the following packages have been built:
- cartesian_controller_simulation
Failed   <<< cartesian_controller_tests [0.03s, exited with code 1]
```

**Software versions**
 - **OS**: Ubuntu 22.04
 - **ROS version**: ROS 2 - Humble
 - **Robot**: UR3e

stefanscherzinger, 2022-11-09:
True, that makes no sense. The `cartesian_controller_tests` need the simulation for integration tests.
Try installing the [cartesian_controller_simulation](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2/cartesian_controller_simulation#build-and-install) package with the instructions. It should not be a big deal.
Thanks for pointing this out.

stefanscherzinger, 2022-11-09:
I updated the [install instructions](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2#installation).

closed 2022-11-09 by LucaBross