fzi-forschungszentrum-informatik/cartesian_controllers #109: How to configure cartesian_motion_controller with Panda arm

AnlageM, 2022-11-15:
**Problem description**
How to configure cartesian_motion_controller with Panda arm

**Software versions**
ROS Noetic, Franka Emika Panda

**To Reproduce**
I inserted the cartesian_motion_controller specific configuration into the .yaml file in franka_control package (franka_ros repo), setting panda_link0 as base and panda_link8 as end effector. I am publishing the desired pose to /my_cartesian_motion_controller/target_frame.
If I bring up the robot spawning for the cartesian_motion_controller, it seems to be loaded correctly, but a communication_constraints_violation error comes out. This error happens in case the robot misses more than 20 packages in a row, and the robot control frequency is 1 KHz. The network is not an issue because the original controllers of panda works.
Any suggestion?

Thank you in advance

stefanscherzinger, 2022-11-25:
@AnlageM

Any updates regarding this?

AnlageM, 2022-11-28:
I was already building in release mode unfortunately. And the workaround did not worked for me.
As soon as I can switch the system on I'll edit this comment adding the error I get.
Thank you for following up!

stefanscherzinger, 2022-12-21:
I just saw that you edited your last response and added the error output (for which I didn't get notified, sorry).

Unfortunately, I can't spot anything unusual either. Just to be sure that it's built in release mode, could you navigate into the root of the workspace and put the output of
```bash
cat build/CMakeCache.txt | grep -i build_type
```
here?
You should be getting a `CMAKE_BUILD_TYPE:STRING=Release`. 

It seems to be running for about *2202 seconds* (*36 minutes*)
between `[INFO] [1669749200.981695]: Started controllers: my_cartesian_motion_controller` and `[ERROR] [1669751403.458882815]: libfranka: ...`.

The error is also mentioned [here](https://github.com/frankaemika/libfranka/issues/60).

> The network is not an issue because the original controllers of panda works.

I'm wondering if you are getting this error with the original controllers if you wait long enough. But that's of course unhandy to debug...

stefanscherzinger, 2022-12-21:
Following Franka's [trouble shooting section](https://frankaemika.github.io/docs/troubleshooting.html#motion-stopped-due-to-discontinuities-or-communication-constraints-violation), this might also be an issue with the [joint limits](https://frankaemika.github.io/docs/control_parameters.html#limit-table). Could you post the configuration for your `my_cartesian_motion_controller`? I could imagine that using relatively high gains and fast workspace motion could exceed the robot's torque limits in the joints.

matthias-mayr, 2023-02-08:
For the Panda it could be worthwhile to point out that we have now open-sourced a [Cartesian impedance controller package](https://github.com/matthias-mayr/Cartesian-Impedance-Controller) that can send torque commands. At least with the iiwa we experience it to be very stable and it also works at 1 kHz update rates. The feature set is a bit different. We do not have workpiece/tool compensation for example.

(Not to draw people away from this implementation. I [recommend this controller package in my repository](https://github.com/matthias-mayr/Cartesian-Impedance-Controller/issues/3#issuecomment-1398570315) to everyone with position- or velocity-controlled robots & will most likely use it a lot this summer with a UR project)

stefanscherzinger, 2023-02-28:
@AnlageM Please reopen if issues still exist or post your solution here if available.

closed 2023-02-28 by stefanscherzinger