fzi-forschungszentrum-informatik/cartesian_controllers #110: Force Measurement of Force/Compliance controller

ZZWang21, 2022-11-15:
Hi Experts, 

currently, I am using the controller to drive UR5e to approach a rigid flat surface. The end-effector is 45 degree to the surface. I used Force controller and Compliance controller to control it. But the 3-axis force measurement is not as I imagined. 

I put my measurement here:
[image]

As I thought, the x- and y- axis force should increase while the z-force increases. 

Please tell me what is wrong here. Is the stiffness setup? 

Thank you very much.

stefanscherzinger, 2022-11-15:
@ZZWang21 

Thanks for providing a plot of the force components! Could you additionally give info about:
 - **Commit**: What's the output of `git rev-parse HEAD` in the `cartesian_controllers` package?
 - **OS**: What's your Operating System?
 - **ROS version**: What's your ROS1/2 version?

> Please tell me what is wrong here.

I have a theory, but could you post an image or a sketch (with coordinate frames) of the end-effector in contact? I'm not 100% sure what you mean by 45 degree to the surface.

> Is the stiffness setup?

Maybe. Let's see how your end-effector orientation looks like before I start speculating.

ZZWang21, 2022-11-16:
@stefanscherzinger 

Thanks for your reply. 

-  Commit: 8a4ec9340fd2a170148d49ea48e6f1adb1487c5a
- LINUX, UBUNTU 18.04
- ROS Melodic, ROS1

The sketch:
[image]

Thanks.

stefanscherzinger, 2022-11-16:
It might be a matter of coordinate systems. Here's an illustration of possible effects:

[image]

By default, I think `target_wrench` is applied in the `end_effector_link`, that's frame **{2}** in the sketch above. See [these explanations](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/ros2/cartesian_controller_simulation/config/controller_manager.yaml#L37) (That's ROS2 but also applies to ROS1). I assume that you measured the force-torque signals directly without any transformation. So, they will be given in frame **{1}**.  If the surfaces has sufficient friction, the robot could find an equilibrium without traveling above the surface and it might appear that there's only a resistance in the `z-axis` (the sensor's z-axis).

There's an option to command the `target_frame` in the `robot_base_link`. There's a boolean parameter `hand_frame_control`. You'll find that in dynamic reconfigure in the force controller. Untick that checkbox and the robot will apply the `target_wrench` in the orientation of **{3}**. This can also be used with the compliance controller (because that inherits from the force controller).

ZZWang21, 2022-11-16:
@stefanscherzinger 

Thank you for your reply. 

My plan is to use the measured forces to estimate the normal vector of the surface, so frame {3} is not my option. 
Yes, the forces are real measured value by the sensor in {1}. 

I also tried using "my_cartesian_motion_controller" to 'simulate' the contact with the desk surface. The force curve is different. 
[image]

I think the value in the picture above is more like the real situation. (since the gripper is not vertical to the surface, the Fx and Fy should NOT be 0- which was shown in my previous post, very small Fx, Fy values. Furthermore, in previous post, they are not changing with the Fz, which makes no sense) Can you explain why the motion_controller has more realistic force measurement?

and, How can I get the contact force in {2}? 

Thank you.

stefanscherzinger, 2022-11-17:
## Some basics

I'll try to explain what you probably observe. I assume that the sensor is working properly and I don't see any result in these plots that would make me think otherwise. If you believe that the sensor signals are *wrong*, please contact your OEM and report your concern. Or, if you wish, contact the driver's developers about details how they propagate the OEM driver's signals.

> ... the motion_controller has more realistic force measurement?

The `cartesian_force_controller` and the `cartesian_compliance_controller` don't have a *less* realistic force measurement. Nor does the `cartesian_motion_controller` have a *more realistic* one. The sensor measurements are what the sensor measures. The controllers read those values and react upon them, but they don't change them.

## Friction

> since the gripper is not vertical to the surface, the Fx and Fy should NOT be 0- which was shown in my previous post

This really depends on the friction of your surface. If you push in the `end_effector_link` (which I assume you do, because that's the default behavior), and in the `z-axis` only, then it's probable that you are inside the *friction cone* of the surface's material. I indicated this with the orange arrow in my sketch above. As a result, the sensor will mostly measure the exact reaction force for the `target_wrench` that you apply. Since you seem to be expecting [the plot from this post](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/110#issuecomment-1316980897) this would only be the case if the gripper was moving in contact. Like sliding. But even then would you measure something different because of friction. That's why estimating the surface normal is challenging.

## Experiments

> Can you explain why the motion_controller has more realistic force measurement?

The difference could be that you behave differently than does the surface. I assume that you took some form of object (for instance a plate) and pushed upwards while maintaining contact with the gripper under an approx. 45° angle. See sketch below:

[image]

The pushing force (purple) is split into its components (red and blue) and measured in the sensor's coordinate frame axes.

When pushing with the force controller, what happens might be like this:

[image]

Due to high friction on the surface, the gripper does not slide and the applied pushing force (purple) is measured mostly in the sensor's z-axis.

If you command your pushing force in negative z-direction in the `robot_base_link` as `target_wrench`, you will observe the [plot that you expect](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/110#issuecomment-1316980897).

---

> and, How can I get the contact force in {2}?

That should be a static transform from **{1}** to **{2}**. But that's not published by the controller at the moment.

ZZWang21, 2022-11-21:
@stefanscherzinger 

Thank you very much for your detailed reply!

I donot think I would go to the OEM of the sensor, since the robot is quite new. I believe the sensor has been calibrated before installation. So that is why I am enquiring experts like you at Github. 

> The cartesian_force_controller and the cartesian_compliance_controller don't have a less realistic force measurement. Nor does the cartesian_motion_controller have a more realistic one. The sensor [...]
--Thank you for that. 

For the curve in [this post](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/110#issuecomment-1316980897), it was not measured in a motion. I set the orientation of the gripper as 45 degree, then I used the motion_controller to control the gripper move verticallly down(keep the orientation) to the surface, and recorded the data. No sliding at all. 

All my data in previous posts was measured when the gripper reached the desk surface. Nothing to do with my hand.

stefanscherzinger, 2022-11-25:
Please reopen if things are still unclear.

closed 2022-11-25 by stefanscherzinger