fzi-forschungszentrum-informatik/cartesian_controllers #115: Stopping the force controller does not stop robot movements

captain-yoshi, 2022-12-05:
**Problem description**
Trying to stop the `force controller`. Using the SwitchController service client, the manager stops the controller but the force controller seems to continue to operate. E.g. falling and oscillating when in contact after enabling control to ros from the ur teach pendant. I don't see anything that stops the force controller in [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_force_controller/include/cartesian_force_controller/cartesian_force_controller.hpp#L137-L141).

Is this a bug and are there other ways to stop the force controller programatically (without having access to the controller object) ?

```cpp
  // start force controller
  controller_manager_msgs::SwitchController switch_ctrl;
  switch_ctrl.request.start_controllers = { cartesian_ctrl_name };
  switch_ctrl.request.timeout = 1.0;
  switch_ctrl.request.strictness = switch_ctrl.request.STRICT;

  if (!ctrl_manager_client.call(switch_ctrl) || !switch_ctrl.response.ok)
  {
    ROS_ERROR("Failed to start the %s controller '%s'", CONTROLLER_TYPE_STR,
              switch_ctrl.request.stop_controllers[0].c_str());
    return 0;
  }

  // do something with a wrench

  // stop controller
  switch_ctrl.request.stop_controllers = { cartesian_ctrl_name };
  switch_ctrl.request.timeout = 1.0;
  switch_ctrl.request.strictness = switch_ctrl.request.STRICT;

  if (!ctrl_manager_client.call(switch_ctrl) || !switch_ctrl.response.ok)
  {
    ROS_ERROR("Failed to stop the %s controller '%s'", CONTROLLER_TYPE_STR,
              switch_ctrl.request.stop_controllers[0].c_str());
    return 0;
  }
```

stefanscherzinger, 2022-12-06:
Based on my understanding of ROS-control, the `controller_manager` should call `update` only on the *running* controllers. So, the position-based controllers are fine with not doing much in `stopping()`. The last joint positions will remain in the command buffer of the corresponding `hardware_interface`. Note that the [velocity-based controllers](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_force_controller/include/cartesian_force_controller/cartesian_force_controller.hpp#L145) do set zero velocities to avoid a continuous drift after stopping.

Since

> the force controller seems to continue to operate

, let's confirm this assumption. Could you perform the following steps?

1. Checkout the latest [master branch](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers), navigate into the `cartesian_controllers` package and post the output of `git rev-parse HEAD` here. That's for me to confirm that you have [this feature](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/commit/0815d44184940ea134f83def125a9487116490fb) available.

2. Activate state publishing in the force controller as explained [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/resources/doc/Solver_details.md#solver-parameters):
   ```yaml
     your_cartesian_force_controller:
       # Type, link and joints specification here
       # ...

       solver:
           publish_state_feedback: True

       # Further specification
       # ...
   ```
3. Now, run your experiment again and check whether you receive messages on the controller's `current_pose` topic. If you do, then someone is in fact calling the Cartesian controller's `update()` method.
4.  Could you also post the output of `git rev-parse HEAD` in the [Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) package?

stefanscherzinger, 2022-12-21:
@captain-yoshi I'll close this since it seems outside the scope of the `cartesian_controllers`. You might reopen this issue or post your resolution here once there's an update.

closed 2022-12-21 by stefanscherzinger

captain-yoshi, 2023-02-01:
**Resolution**
The PR has now been merged in `ros_control`. Adding the same ctrl name for start and stop had undefined behavior. The controller manager will now reject this case and warn the user.