fzi-forschungszentrum-informatik/cartesian_controllers #116: multiple targets behaviuor

muttistefano, 2023-01-25:
Hi, thanks for the good work.
I am using the cartesian motion controller, with ros-rolling.
I noticed that when I publish multiple target frames, the controller won't follow the last one, but rather "blends" the movements and create unstable behaviuors.
Am i using it wrong? 
Is there a way to stop the controller before sending more targets?
Thanks

stefanscherzinger, 2023-01-27:
Hi @muttistefano 
Thanks for using the Controllers with `rolling`. I'm not sure whether that's intended behavior either. Could you check whether that happens in simulation as well? The controllers do interpolate. That's a feature. You can change that with the solver's parameters. However, it should not become unstable. 

> Is there a way to stop the controller before sending more targets?

Starting an stopping the controllers is up to you, and is done as usual with the respective services. When starting the controllers, they should start with the robot's current state and then immediately react to newly published targets.

stefanscherzinger, 2023-02-28:
Please reopen if things are still unclear.

closed 2023-02-28 by stefanscherzinger