fzi-forschungszentrum-informatik/cartesian_controllers #120: UR5e Gravity Compensation of Payload with Zeroed FT Sensor

ian-chuang, 2023-03-16:
**Problem description**

I have a UR5e with a Robotiq 2F-85 gripper attached to its end effector. I am trying to use the cartesian force controller, but I am running into issues when the end effector rotates and the weight of the gripper starts changing the values of the force torque sensor. I've seen that there gravity compensation parameters for solving this issue, however the nature of the force torque readings in the current UR E-Series ROS driver complicate things.

It seems like the UR5e driver zeros the force torque sensor on launch, and then you can keep calling a service to zero it again. It seems like I can't get the raw, unzeroed values of the sensor. I could try diving deeper into the code for their driver, but I just wanted to ask here first to see if there is a way around this? 

Essentially, how can I do gravity compensation and rotate the end effector with force control while dealing with a force torque sensor that is off-center due to being zeroed out?

gavanderhoorn, 2023-03-16:
> It seems like the UR5e driver zeros the force torque sensor on launch [..]

As I was curious about this myself, I looked at the implementation.

The ROS service you mention (`zero_ftsensor`, [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/395c0541b20d0da2cd480e2ad85b2100410fb043/ur_robot_driver/src/hardware_interface.cpp#L448-L449)) is implemented [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/395c0541b20d0da2cd480e2ad85b2100410fb043/ur_robot_driver/src/hardware_interface.cpp#L1140-L1157). This calls `this->ur_driver_->zeroFTSensor()`, which is provided by `Universal_Robots_Client_Library`. That's implemented [here](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/2e69bc165da0ad22fe9a1a3417e7eaa4b082ebac/src/ur/ur_driver.cpp#L236-L262) and basically calls `ScriptCommandInterface::zeroFTSensor()` ([here](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/2e69bc165da0ad22fe9a1a3417e7eaa4b082ebac/src/control/script_command_interface.cpp#L40-L57)).

Afaict, neither `zero_ftsensor` nor its callback nor `UrDriver::zeroFTSensor()` or the underlying `ScriptCommandInterface::zeroFTSensor()` seem to be called by at startup of the hardware interface.

Afaik, the sensor is not tared automatically, or at least the code does not seem to suggest it is.

ian-chuang, 2023-03-16:
Sorry, I assumed that the sensor was zeroed on startup because whenever I would restart the ros driver, it would start at 0 (however, I didn't move the robot before restarting the driver). It seems like then that the zeroing is done on the robot side and not on ROS then? I will keep playing around with the ROS driver once I get back to the lab on Friday.

It would be nice then if we had more control over the FT sensor for the UR E-Series ROS driver, similar to the ROS package for the [ATI sensor](http://wiki.ros.org/ati_force_torque). For example, having some options for unzeroing the sensor and reading raw data, doing gravity compensation, applying a low pass filter would be helpful.

Ideally, having a way to do gravity compensation and then having the option to zero the sensor after applying compensation would be the most robust solution. I will try looking into my own solution for now, but was wondering if other people were experiencing limitations with the UR E-series FT sensor in ROS and have already solved this problem.

Apologies if this is the wrong repo to be posting these things.

gavanderhoorn, 2023-03-17:
> Sorry, I assumed that the sensor was zeroed on startup because whenever I would restart the ros driver [..]

I'm not dismissing your observation.

Just drawing attention to the relevant code paths.

stefanscherzinger, 2023-03-25:
@ian-chuang 

Thanks for reporting your issue!

> It seems like then that the zeroing is done on the robot side and not on ROS then?

I guess so. At least that's my cheap way of manually zeroing the sensor when I'm in the lab with the teach panel. I guide the robot with the `cartesian_force_controller` and just restart the driver whenever the sensor drifts too much. Thanks @gavanderhoorn for checking the driver's code.

@ian-chuang Could I ask you to open a feature issue at the driver package ([ROS1](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) or [ROS2](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver))?

> It would be nice then if we had more control over the FT sensor for the UR E-Series ROS driver, similar to the ROS package for the [ATI sensor](http://wiki.ros.org/ati_force_torque). For example, [...]

I absolutely like those ideas! Others might too, so that's a great start for a discussion.

---

Other than that, gravity compensation should still work with your current setup and the driver. This feature, however, hasn't been used much lately and I'm not 100% sure that it works as I remember. Here's how it is intended to be used:
1. Users specify the tool mass (everything that comes after the sensor flange) in their launch file
2. Then, they start their robot with a zeroed sensor in some configuration of their choice
3. When the end-effector rotates, the controller's gravity compensation should exactly compensate sensor's drift. That works as good as the user has specified the tool mass.
4. If, for some reason, the robot's sensor needs to be zeroed, users call the controller's `signal_taring` service right after to tell the controller that it should correct its gravity direction.
5. Proceed with step `2.`.

It has also been investigated in [this issue](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/100) and in [this issue](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/39), and there's also a [PR](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/pull/91). Unfortunately, I haven't had the time to look into this.

stefanscherzinger, 2024-01-10:
Please reopen if still relevant or when having new insights/suggestions.

closed 2024-01-10 by stefanscherzinger