fzi-forschungszentrum-informatik/cartesian_controllers #121: How can I apply this package to Gazebo model on ROS1?

soohwan-justin, 2023-03-22:
Hello @stefanscherzinger, I have tried to follow this [issue](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/47) with [UR5 gazebo model](https://github.com/ros-industrial/universal_robot/tree/kinetic).

This is my yaml and launch file

**ur_controllers.yaml**
```yaml
my_cartesian_motion_controller:
    type: "position_controllers/CartesianMotionController"
    end_effector_link: "tool0"
    robot_base_link: "base_link"
    target_frame_topic: "target_frame"
    joints:
    - shoulder_pan_joint
    - shoulder_lift_joint
    - elbow_joint
    - wrist_1_joint
    - wrist_2_joint
    - wrist_3_joint

    pd_gains:
        trans_x: {p: 10.0}
        trans_y: {p: 10.0}
        trans_z: {p: 10.0}
        rot_x: {p: 1.0}
        rot_y: {p: 1.0}
        rot_z: {p: 1.0}

my_cartesian_force_controller:
    type: "position_controllers/CartesianForceController"
    end_effector_link: "tool0"
    robot_base_link: "base_link"
    ft_sensor_ref_link: "tool0"
    joints:
    - shoulder_pan_joint
    - shoulder_lift_joint
    - elbow_joint
    - wrist_1_joint
    - wrist_2_joint
    - wrist_3_joint

    solver:
        error_scale: 0.5

    pd_gains:
        trans_x: {p: 0.05}
        trans_y: {p: 0.05}
        trans_z: {p: 0.05}
        rot_x: {p: 1.5}
        rot_y: {p: 1.5}
        rot_z: {p: 1.5}

my_cartesian_compliance_controller:
    type: "position_controllers/CartesianComplianceController"
    end_effector_link: "tool0"
    robot_base_link: "base_link"
    ft_sensor_ref_link: "tool0"
    compliance_ref_link: "tool0"
    target_frame_topic: "target_frame"
    joints:
    - shoulder_pan_joint
    - shoulder_lift_joint
    - elbow_joint
    - wrist_1_joint
    - wrist_2_joint
    - wrist_3_joint

    stiffness:  # w.r.t. compliance_ref_link
        trans_x: 500
        trans_y: 500
        trans_z: 500
[... 171 lines cut ...]
                <remap from="my_cartesian_force_controller/target_wrench" to="target_wrench" />
                <remap from="my_cartesian_compliance_controller/target_wrench" to="target_wrench" />

        </node>

	<arg name="paused" default="true"/>
	<arg name="use_sim_time" default="true"/>
	<arg name="gui" default="true"/>
	<arg name="headless" default="false"/>

	<include file="$(find gazebo_ros)/launch/empty_world.launch">
		<arg name="debug" value="$(arg debug)" />
		<arg name="gui" value="$(arg gui)" />
		<arg name="paused" value="$(arg paused)"/>
		<arg name="use_sim_time" value="$(arg use_sim_time)"/>
		<arg name="headless" value="$(arg headless)"/>
	</include> 
</launch>

```

But when I run this launch file, the robot model oscillates and doesn't work even I set the parameters on rqt, like this:
[image]

I don't know what mistake have I made, I just want to sync the rviz and gazebo model. I hope you help me, please.

stefanscherzinger, 2023-03-25:
Hi @soohwan-justin 

Thanks for reporting this. I'm not completely sure what's happening. Could you do two things?

1. Spawn the robot a little above the floor. It looks a little as if it was colliding initially. You can achieve that with `-z 0.1` in this line:
   ```xml
      <node name="urdf_spawner" pkg="gazebo_ros" type="spawn_model" respawn="false" output="screen"
            args="-urdf -param robot_description -model robot -z 0.1 -unpause">
   ```

   You seem to be working on `Kinetic`. The more recent documentation for `Melodic` mentions this [in one of its launch files](https://github.com/ros-industrial/universal_robot/blob/melodic/ur_gazebo/launch/inc/ur_control.launch.xml#L29).
2. Could you copy and paste the error messages from the terminal here? Maybe something's wrong with remapping.

stefanscherzinger, 2023-04-25:
@soohwan-justin 

Please re-open if your issue still exists.

closed 2023-04-25 by stefanscherzinger