fzi-forschungszentrum-informatik/cartesian_controllers #122: Stable compliance controller configuration in simulation

matthias-mayr, 2023-04-18:
**Problem description**
I am setting up a UR5e setup in simulation and reality right now. With an older version of this controller package we found a simulation and real-robot configuration that we were happy with.

Now with the latest version of this package, I am getting very unstable behavior in simulation (haven't tried the real robot yet) and none of the gain & solver configurations that I saw here in the issues worked for me.

**Software versions**
 - **Commit**: 41a11e2efac7125c9795f1365ae158fa82b8f6c8
 - **OS**: Ubuntu 20.04
 - **ROS version**: ROS 1 Noetic
 - **Robot**: UR5e

**To Reproduce**
Start a UR5e Gazebo setup with the following configuration:
```yaml
cartesian_compliance_controller:
    type: "position_controllers/CartesianComplianceController"
    end_effector_link: "ur5e_tcp_link"      # All links below must come before this link
    robot_base_link: "ur5e_base_link"       # Which is the "world" frame of this controller
    ft_sensor_ref_link: "ur5e_wrist_3_link"
    compliance_ref_link: "ur5e_tcp_link"    # Specify based on which frame you want to be compliant
    target_frame_topic: "target_frame"      # Name of the topic where it receives commands
    joints:
     - ur5e_shoulder_pan_joint
     - ur5e_shoulder_lift_joint
     - ur5e_elbow_joint
     - ur5e_wrist_1_joint
     - ur5e_wrist_2_joint
     - ur5e_wrist_3_joint

    stiffness:  # w.r.t. compliance_ref_link
        trans_x: 200
        trans_y: 200
        trans_z: 200
        rot_x: 20
        rot_y: 20
        rot_z: 20
    solver:
        error_scale: 0.2
        iterations: 3
    pd_gains:
        trans_x:
          p: 0.05
          d: 0.015
        trans_y:
          p: 0.05
          d: 0.015
        trans_z:
          p: 0.05
          d: 0.015
        rot_x:
          p: 3
          d: 0.01
        rot_y:
          p: 3
          d: 0.01
        rot_z:
          p: 3
          d: 0.01
```

and the following gains:
```yaml
gazebo_ros_control/pid_gains:
  # from https://github.com/ros-industrial/universal_robot/pull/525/commits/77cc72fa846d0ce69bd4122769a5f8cd951cc517
  ur5e_shoulder_pan_joint: {p: 2000,  d: 20, i: 1, i_clamp: 1}
  ur5e_shoulder_lift_joint: {p: 6000,  d: 40, i: 1, i_clamp: 1}
  ur5e_elbow_joint: {p: 2000,  d: 20, i: 1, i_clamp: 1}
  ur5e_wrist_1_joint: {p: 500,  d: 1, i: 1, i_clamp: 1}
  ur5e_wrist_2_joint: {p: 500,  d: 1, i: 1, i_clamp: 1}
  ur5e_wrist_3_joint: {p: 500,  d: 1, i: 1, i_clamp: 1}
```

This configuration was stable until `90f8509: define high-level goals for the next development` and broke with `6fc2898: Merge pull request #44 from fzi-forschungszentrum-informatik/change-robot-synchronization-policy`.

Now the robot overshoots a lot. Sometimes it converges after some minutes, but most often it does do continuous large motions of 10-50cm and all sorts of rotations.

**Expected behavior**
Stable performance in simulation.

If someone has a configuration that does not make the robot "dance", I'd be very happy to get the configuration or some hints.

matthias-mayr, 2023-04-24:
A part of the issue are the PID gains. I tried these ones from the current configuration in the [universal_robots repo](https://github.com/ros-industrial/universal_robot/blob/melodic-devel/ur_gazebo/config/ur5e_controllers.yaml) as well:
```yaml
  gains: # Required because we're controlling an effort interface
    shoulder_pan_joint: {p: 4000,  d: 200, i: 1, i_clamp: 1}
    shoulder_lift_joint: {p: 10000,  d: 200, i: 1, i_clamp: 1}
    elbow_joint: {p: 2000,  d: 20, i: 1, i_clamp: 1}
    wrist_1_joint: {p: 500,  d: 1, i: 1, i_clamp: 1}
    wrist_2_joint: {p: 500,  d: 1, i: 1, i_clamp: 1}
    wrist_3_joint: {p: 10,  d: 0.1, i: 0, i_clamp: 1} 
```
They are also very unstable. Even for non-compliant position control with the standard "UR" scaled position trajectory controller. Not sure why. My understand was that the Gazebo PID gains do the same job as the ones specified in the `effort_joint_trajectory_controller` above, just in a different place.

As an addition: I rate-limit the Gazebo FT-sensor plugin values to a change of 100 N and 20 Nm a second, because of the noise.

When not using PID gains, current master branch is also very unstable, but it gets better with strongly rate-limited FT sensor feedback. Without the rate-limiter, this is not very surprising as the signal is very noise. About +-10 N on the z-Axis for example.

stefanscherzinger, 2023-04-25:
@matthias-mayr 

Thanks for reporting this! I haven't worked with Gazebo (classic) lately but I also think that it could be a stability problem of cascaded controller gains.

> They are also very unstable. Even for non-compliant position control with the standard "UR" scaled position trajectory controller. Not sure why. 

That's interesting. If not already done, could you open an issue there?

> (haven't tried the real robot yet) 

I recommend [these default](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_examples/config/example_controllers.yaml#L47) gains when working with real hardware. They feel natural on URe robots when manually guiding the end-effector and are stable in most of the robot's workspace on a stiff surface. You might fine tune from there. In your configuration from [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/122#issue-1672988936), the `d` gains might already be too high.

I'll see if I can have a look at Gazebo.

ZZWang21, 2023-06-05:
Similar here, will do the Gazebo simulation these days, please keep updating @matthias-mayr . Thanks!

ZZWang21, 2023-06-08:
Sorry, but @matthias-mayr what code do you use? Python or C++? I am wondering if I can use python code to use the cartesian controller.

ZZWang21, 2023-06-30:
@matthias-mayr  i checked the link you posted, but it is from ur_gazebo folder? how do you start the simulation? 

and, where comes the code:
"
gazebo_ros_control/pid_gains:
  # from https://github.com/ros-industrial/universal_robot/pull/525/commits/77cc72fa846d0ce69bd4122769a5f8cd951cc517
  ur5e_shoulder_pan_joint: {p: 2000,  d: 20, i: 1, i_clamp: 1}
  ur5e_shoulder_lift_joint: {p: 6000,  d: 40, i: 1, i_clamp: 1}
  ur5e_elbow_joint: {p: 2000,  d: 20, i: 1, i_clamp: 1}
  ur5e_wrist_1_joint: {p: 500,  d: 1, i: 1, i_clamp: 1}
  ur5e_wrist_2_joint: {p: 500,  d: 1, i: 1, i_clamp: 1}
  ur5e_wrist_3_joint: {p: 500,  d: 1, i: 1, i_clamp: 1}
" ？

In my opinion, if you use the cartesian controllers in this post, you should stop the original arm controllers, such as effort_joint_trajectory_controller? Correct me if i am wrong. Thanks.

ZZWang21, 2023-07-10:
Hi @matthias-mayr  I used your config in Gazebo. And I published a z-direction force 100N to the /target_wrench topic. But the robotic arm is not moving. the similar problem also happened on my own config, did your Gazebo model move using the compliance controller? Thanks.

ZZWang21, 2023-07-11:
> A part of the issue are the PID gains. I tried these ones from the current configuration in the [universal_robots [...]

sorry, my question is, when the cartesian_controller is used, the original controller is stopped, right? How are the PID parameters still working? Thanks.

ZZWang21, 2023-07-17:
sorry @matthias-mayr , but can I ask did you use the compliance controller with handle in RViz? Did you modify the .rviz file? Thank you.

ZZWang21, 2023-08-05:
@matthias-mayr  could you please share with me some of your experience setting up the Gazebo simulation with the cartesian controller? Thanks.

matthias-mayr, 2023-10-30:
Some small update: For the real robot, I implemented a low-pass filter, which made it much more stable.
I haven't tried the simulation with PID gains again and I will not have time to do it any time soon.

I didn't close this issue yet, because I think it would be a great help if some stable configuration for popular robot arms like the UR5e could be provided. However, since I don't have time to do this any time soon, you are also welcome to close this issue.

stefanscherzinger, 2024-01-10:
Please reopen if still relevant or when having more insights/suggestions.

closed 2024-01-10 by stefanscherzinger

shrutic12, 2024-04-08:
> @matthias-mayr
>
> Thanks for reporting this! I haven't worked with Gazebo (classic) lately but I also think that it could be a stability problem of cascaded controller gains.
>
>
> That's interesting. If [...]

Hello @stefanscherzinger, I am using the catesian compliance controller with UR10e robot on ROS2 Humble using the default values you suggest.  I am testing with different stiffnesses in XYZ translation direction and I am expecting that this will add "compliance" to the robot in these axes, and expect that the robot will be less stiff in these directions. For instance, reducing stiffness in Z, would mean I should be able to push the robot up (physically) and it should restabilize (like if it had a spring damper attached to the end effector). But this is not what happens in my tests. After reducing stiffness in XYZ, I command the robot to move vertically down to a position which has an obstacle, but instead of stopping on contact and stabilizing, the robot continue to move down. In simulation on rviz, the robot jitters heavily as well. 

Can someone help me out? I have reduced the stiffness to about 90 in X,Y,Z translation for the tests. Are there any recommended stiffness values for UR10e from your experience/tests? At recomended default values that you provided what kind of performance do you have on your setup, does the robot re-stabilize on external contact? Thanks a lot! Any help would be appreciated