fzi-forschungszentrum-informatik/cartesian_controllers #123: CartesianForceController/CartesianComplianceController stops working if wrench contains NaN values

firesurfer, 2023-04-24:
**Problem description**

The CartesianForceController stops working if it recieves  a NaN value in the ftSensorWrenchCallback unfortunately NaN is the default value of the mocked ros2control interface for the UR16e arms. 

**Software versions**
 - **Commit**:  6707afac2658a6b6353dad7f4df6fe33aff528e7
 - **OS**:  Ubuntu 22.04
 - **ROS version**: humble
 - **Robot**: Sim

**To Reproduce**

Publish a sensor wrench message containing a NaN value 

**Expected behavior**

Controller interprets received wrench as list of 0 values (perhaps prints a warning once) and is therefore just a position controller in the case of the CartesianComplianceController.

stefanscherzinger, 2023-04-25:
Thanks @firesurfer for reporting this.

I'll check where to catch that in the controller.

> unfortunately NaN is the default value of the mocked ros2control interface for the UR16e arms

Could you point me to some code? It might be interesting to additionally fix that over there. I'm not sure if `NaN` is a good default value for a force-torque sensor.

firesurfer, 2023-04-27:
Regarding catching the Nan you might want to start here: https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/41a11e2efac7125c9795f1365ae158fa82b8f6c8/cartesian_force_controller/include/cartesian_force_controller/cartesian_force_controller.hpp#L259

Regarding the ros2control interface for the UR arms. Not sure where to point at exactly. So if you take a look in the description package: https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/blob/464864acfe85f600c3d3438e025805b5296f5aa8/urdf/ur.ros2_control.xacro#L165

You can see that it has a ForceTorque sensor defined. When specifying `use_fake_hardware` a ros2control mockinterface will be used:
https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/blob/464864acfe85f600c3d3438e025805b5296f5aa8/urdf/ur.ros2_control.xacro#L34

According to the [documentation](https://control.ros.org/master/doc/ros2_control/hardware_interface/doc/mock_components_userdoc.html)  we can also specify `mock_sensor_commands` there to fake sensor values. 
So the way to go there would be to specify a new controller that pushes for example a 0 value to mocked sensor interface

(Apparently mock_sensor_commands was called fake_sensor_commands in previous version)

stefanscherzinger, 2023-05-25:
@firesurfer 

This is now fixed on the `ros2` branch.
- The controllers now discard inputs with `NaN` content
- They currently warn in a throttled way (3 sek)
- When `NaN`s do get into the internal dynamics, e.g. through controller gains or other solver parameters, then the controller shuts down. It never writes `NaN`s to the hardware.

Let me know how this works for you.

PR #126 (2023-05-25): Discard nan inputs
## Steps

Discard any `NaN`s in the controllers' input topics:

- [x] Target poses
- [x] Target wrenches
- [x] Force-torque sensor wrenches
- [x] Catch any `NaN`s before writing to the command handles

Fixes #123