fzi-forschungszentrum-informatik/cartesian_controllers #125: ROS2 Humble with physical UR10

Tahir-Rasheed-437, 2023-05-09:
**Software versions**
 - **OS**: UBUNTU 22.04
 - **ROS version**: ROS2-humble
 - **Robot**: Physical UR10 (no simulation)
 -  **Branch**: ros2-devel

**Problem description**
I have previously used the Cartesian Controllers in ROS1 melodic with actual UR10 and its works perfectly. However, i need to use the controller now in ROS2-humble. I have tested the simulation environment and it works great (ROS-devel branch), however, i cannot find a launch file that can link me with the real robot. I believe i used [this](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_examples/launch/examples.launch) launch file to control my actual UR in ROS1 but i believe it hasn't been exported to ROS2 yet? Is there any package/launch file that I can apply to control the Cartesian position/forces of the actual UR10?

Secondly, i was wondering, if there is any Cartesian twist controller (to command the Cartesian velocity) exists for UR10? I remember something in ROS1 in [cartesian_ros_control](https://github.com/fzi-forschungszentrum-informatik/cartesian_ros_control) repo but does it exist in ROS2? Please do guide me about it.

Thank you

stefanscherzinger, 2023-05-11:
Hi @Tahir-Rasheed-437 

> I believe i used [this](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_examples/launch/examples.launch) launch file to control my actual UR in ROS1 but i believe it hasn't been exported to ROS2 yet?

The ROS2 version of this file is [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/ros2/cartesian_controller_simulation/launch/simulation.launch.py) and the configuration for a the `controller_manager` is [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/ros2/cartesian_controller_simulation/config/controller_manager.yaml). You can use that as basis for controlling the actual UR10. I'll see if I can provide you with a working example later.

> Secondly, i was wondering, if there is any Cartesian twist controller (to command the Cartesian velocity) exists for UR10?

I'm not completely sure. The `cartesian_ros_control` interface hasn't been ported yet, but there might be other controller interfaces or controllers available somewhere. If you want to control the robot somehow *directly*, e.g. with teleoperation, you could also consider using wrenches instead. [Here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2/cartesian_controller_utilities#force-control) is a nice method to control the robot with a 6d space mouse. I use that frequently for teaching tasks.

Tahir-Rasheed-437, 2023-05-11:
Hello @stefanscherzinger, thank you very much for your response. Actually i did tried this launch file, it works very well with in simulation but i am unable to run it with the UR10. if you can please provide me a working example that would be great. 

Yes true. I was searching for the ROS2 version of cartesian_ros_control repo but yes as you said it hasn't been imported to ROS2 yet. I found that [moveit_servo](https://github.com/ros-planning/moveit2/tree/main/moveit_ros/moveit_servo) can be used for such a task (twist control). My current goal is to use the UR directly in position or twist based on the visual servoying feedback but later force/wrench control as well.

Tahir-Rasheed-437, 2023-05-24:
@stefanscherzinger May I kindly inquire if you had the chance to take a look at setting up an example? Thanks

stefanscherzinger, 2023-05-30:
@Tahir-Rasheed-437 

Sorry for not responding earlier! Unfortunately very busy these days.

> May I kindly inquire if you had the chance to take a look at setting up an example?

Kind of. I set up a `ROS2 Humble` version [here](https://github.com/stefanscherzinger/cartesian_controllers_universal_robots/tree/ros2). I tried to test that with `ursim`, which behaves just like the real hardware. Unfortunately, the driver is currently crashing and I'm running out of time for today.
You could see if this setup already clarifies some of your questions.

If I'm lucky, I'll get my hands on an actual UR10e tomorrow and test this.

stefanscherzinger, 2023-05-31:
@Tahir-Rasheed-437 

I just tested this on `Humble` and it works as expected.

When using this on your UR robot, remember to
- specify your robot [here](https://github.com/stefanscherzinger/cartesian_controllers_universal_robots/blob/a6b95437c8a11480600c609d4b0c9260581fe214/urdf/setup.urdf.xacro#L16)
- use your robot-specific calibration file. [This one](https://github.com/stefanscherzinger/cartesian_controllers_universal_robots/blob/a6b95437c8a11480600c609d4b0c9260581fe214/config/test_calibration.yaml#L2) is just an example.
- Use your robot's `robot_ip` [here](https://github.com/stefanscherzinger/cartesian_controllers_universal_robots/blob/a6b95437c8a11480600c609d4b0c9260581fe214/launch/robot.launch.py#L21)

Let me know if this works for you.

stefanscherzinger, 2023-06-12:
@Tahir-Rasheed-437 

Did that work out?

Tahir-Rasheed-437, 2023-06-12:
@stefanscherzinger 

Thank you very much for the proposed solution. Apologize I was on parental leave for past 2 weeks. I Just joined back today and saw our UR10 not working. Something went wrong and its not starting up. We will try to fix it in the coming days and will then test your proposed solution as soon as our UR10 is fixed.

By the time being, I can try your proposed solution in coppeliasim simulator hopefully before the end of the week.

stefanscherzinger, 2023-07-13:
Please re-open if anything is still unclear.

closed 2023-07-13 by stefanscherzinger

Tahir-Rasheed-437, 2023-07-31:
@stefanscherzinger Apologize for the late reply. Due to Summer vocations and other commitments, i couldn't timely test your proposed solution. I have tested today and it worked very well. Thank you very much again for setting up a working example. 
It would be also great if you can mention the  cartesian_controllers_universal_robots package in the readme so people can directly link to it with the UR for running the controllers.

Tahir-Rasheed-437, 2023-07-31:
@stefanscherzinger Apologize for the late response. Due to summer vocations and other important commitments, i couldn't timely test your proposed solution. I finally been able to test today and it worked very well. Thank you once again for the setting up the working example. 
It would be great if you can mention the package  [cartesian_controllers_universal_robots](https://github.com/stefanscherzinger/cartesian_controllers_universal_robots) in the readme so anyone who wants to test the cartesian controllers with UR's can directly link to the package.