fzi-forschungszentrum-informatik/cartesian_controllers #127: [Feature Request]  - Allow to change compliance_ref_link in a loaded controller

firesurfer, 2023-05-25:
## Problem statement and motivation

Currently it is not possible to change to compliance_ref_link in an already loaded controller as it is not possible to reconfigure a controller once it is configured ( as far as know - see https://github.com/ros-controls/ros2_control/issues/1033)

For larger setups with multiple robot arm which also have multiple tools which have to be compliant at some point the current solution of defining a controller per compliant link becomes messy quite easily. 

Therefore it would be nice to just have a compliance controller per arm and being able to change to ref_link.

## Goal of this feature

Being able to set the `compliance_ref_link` , deactivate -> active the controller and then use the changed ref_link

## Approach and next steps

Simply move:
https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/752a6a928f6584597ea4b8bace047ec7fdc2314d/cartesian_compliance_controller/src/cartesian_compliance_controller.cpp#L103

into the beginning of on_activate. 

If desired I can also provide a PR for that.

stefanscherzinger, 2023-05-25:
@firesurfer 

Thanks for this feature request. I like the idea and will have a look.

firesurfer, 2023-05-26:
@stefanscherzinger  Great that you see it the same way. Give that I need this feature rather sooner then later I will try to workout a solution and create a PR.

srsidd, 2025-06-06:
Hi all, any updates on this feature? This seems like a core / fundamental feature to have.