fzi-forschungszentrum-informatik/cartesian_controllers #132: Velocity command interface ROS2 foxy

fafux, 2023-05-31:
**Problem description**
Hi,
I am trying to control an UR robot both in simulation (both on Rviz and Webots) and with the real hardware with the command interface in velocity with ROS2 foxy. The values of the joint velocities states, in simulation, are changing but the robot is not moving. While if I try to move the robot using the position command interface I can do it.
Is it possible to move the robot only using the velocity command interface? 
Thanks.

**Software versions**
 - **OS**: Ubuntu 20.04
 - **ROS version**: ROS2 foxy
 - **Robot**: Universal Robot

fafux, 2023-05-31:
Hi @stefanscherzinger 
Below you can see the first part of _ur_controllers.yaml_ file:
```
controller_manager:
  ros__parameters:
    update_rate: 600  # Hz

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    io_and_status_controller:
      type: ur_controllers/GPIOController

    speed_scaling_state_broadcaster:
      type: ur_controllers/SpeedScalingStateBroadcaster

    force_torque_sensor_broadcaster:
      type: ur_controllers/ForceTorqueStateBroadcaster

    joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    scaled_joint_trajectory_controller:
      type: ur_controllers/ScaledJointTrajectoryController

    cartesian_motion_controller:
      type: cartesian_motion_controller/CartesianMotionController

cartesian_motion_controller:
  ros__parameters:
    end_effector_link: "tool0"
    robot_base_link: "base_link"
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint

    # Choose: position or velocity.
    command_interfaces:
      #- position
      - velocity

    solver:
        error_scale: 1.0
        iterations: 10
        publish_state_feedback: True

    pd_gains:
        trans_x: {p: 1.0}
        trans_y: {p: 1.0}
        trans_z: {p: 1.0}
        rot_x: {p: 0.5}
        rot_y: {p: 0.5}
        rot_z: {p: 0.5}

# [...] the other controllers params
```
The speed data both from the publish_state_feedback (_/cartesian_motion_controller/current_twist_) and from the _/joint_states_ are meaningful and are changing and different from 0 but the robot is not moving.

stefanscherzinger, 2023-06-01:
@fafux 

The controllers' specification looks alright. Let's clarify two things before we go into further debugging:

1. Why do you need velocity-based control as opposed to position-based control?
2. > but the robot is not moving.

   Simulation or the real hardware?

fafux, 2023-06-01:
@stefanscherzinger 

1. Because afterwards I would like to do something on top of your controller.
2. Both simulation and real hardware.

stefanscherzinger, 2023-06-02:
Alright, I'll see if I can reproduce this on a real setup.

Concerning `1.`: The `cartesian_controllers` take force-torque vectors, poses, or both as input. On the lower level, they command what is most suitable for the respective hardware. But the inputs for the users do not change. What is it that you want to do on top that you cannot do with the position-based version?

fafux, 2023-06-05:
First, I would like to avoid the numerical integration from joint velocities to joint positions based on a fixed control loop rate (in Foxy) that, depending on the hardware/application could be not respected precisely.
Then, I would like to add some null-space task that I would like to test directly with the velocity commands.

fafux, 2023-06-06:
Thanks for your time. Universal Robots answered my issue and declared that velocity commands are not implemented in foxy.
https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/694

closed 2023-06-06 by fafux