fzi-forschungszentrum-informatik/cartesian_controllers #135: How to use handle and compliance controller on UR5 robot?

ZZWang21, 2023-07-01:
I am trying to implement compliance controller+handle on UR5 robot in GAZEBO. But the handle cannot move the robot. 

error message:  ERROR] [1688235663.301442237, 86.602000000]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 1.655419 seconds). Stopping trajectory.

I am using ROS NOETIC. ros1 version. UR5 ROBOT+ ROBOTIQ 85 Gripper. 

I started the robot in gazebo with original "arm_controller", then switch to compliance controller and handle. but the handle controller cannot move the robot. AND, compliance force controller can apply force on the end effector, but compliance controller seemed not working. 

I expected the robot moves according to the target frame of the handle and apply the compliance on the end effector. 

Thanks.

ZZWang21, 2023-07-02:
the yaml:

"arm_controller_force:
        type: "position_controllers/CartesianForceController"
        end_effector_link: "tool0"
        robot_base_link: "base_link"
        ft_sensor_ref_link: "wrist_3_link"
        joints:
          - shoulder_pan_joint
          - shoulder_lift_joint
          - elbow_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint

        solver:
          error_scale: 0.5

        pd_gains:
          trans_x: {p: 0.05}
          trans_y: {p: 0.05}
          trans_z: {p: 0.05}
          rot_x: {p: 1.5}
          rot_y: {p: 1.5}
          rot_z: {p: 1.5}
          
    arm_controller_compliance:
        type: "position_controllers/CartesianComplianceController"
        end_effector_link: "tool0"
        robot_base_link: "base_link"
        ft_sensor_ref_link: "wrist_3_link"
        compliance_ref_link: "wrist_3_link" #tool0
        target_frame_topic: "target_frame"
        joints:
          - shoulder_pan_joint
          - shoulder_lift_joint
          - elbow_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
          
        stiffness:  # w.r.t. compliance_ref_link
          trans_x: 500
          trans_y: 500
          trans_z: 500
          rot_x: 20
          rot_y: 20
          rot_z: 20

        solver:
          error_scale: 0.5

        pd_gains:
          trans_x: {p: 0.05}
          trans_y: {p: 0.05}
          trans_z: {p: 0.05}
          rot_x: {p: 1.5}
          rot_y: {p: 1.5}
          rot_z: {p: 1.5}
          
    arm_controller_handle:
        type: "cartesian_controllers/MotionControlHandle"
        end_effector_link: "tool0"
        robot_base_link: "base_link"
        target_frame_topic: "/my_cartesian_motion_controller/target_frame"
        joints:
          - shoulder_pan_joint
          - shoulder_lift_joint
          - elbow_joint
          - wrist_1_joint
[... 103 lines cut ...]
    

    
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" respawn="true" output="screen" />

    
    <include file="$(find ur5erobotiq_moveit_config)/launch/planning_context.launch">
        <arg name="load_robot_description" value="true" />
    </include>
    <include file="$(find ur5erobotiq_moveit_config)/launch/move_group.launch">
    </include>
    <group if="$(arg gui)">
        <include file="$(find ur5erobotiq_moveit_config)/launch/moveit_rviz.launch">
            <arg name="rviz_config" value="$(arg rviz_config)"/>
        </include>
    </group>

    

    
    <node pkg="ur5_robotiq_85_simulation" type="robot_control_node" name="robot_control_node" output="screen" />

    
    <node pkg="controller_manager" type="spawner" name="joint_position_launcher" args="--stopped joint_position_controller arm_controller_force arm_controller_compliance arm_controller_handle" />
</launch>"

ZZWang21, 2023-07-09:
Hello experts @bmagyar  @stefanscherzinger  could you give me some hints? I am tring to use the controller to control Gazebo model, and from there, I can apply some machine learning controls. But the first thing is to use the compliance controller on robot simulation model and move it to target positions. Thanks!

ZZWang21, 2023-07-11:
Excuse me, experts @fmauch  @fzi-admin @stefanscherzinger @bmagyar  could you help me out here? Thanks.

fmauch, 2023-07-11:
Please do not at-mention people (especially people not directly affiliated with this repo). This is considered very rude.

I understand your frustration and that you want to get things done, but please understand that sometimes answering an issue can take some time. We are a very small team on this and especially in summer time the relevant developer(s) might just not be available for a certain period of time.

ZZWang21, 2023-07-11:
> Please do not at-mention people (especially people not directly affiliated with this repo). This is considered very rude.
>
> I understand your frustration and that you want to get things done, but [...]

Sorry about that. I did not realise that since when I typed the names automatically came out. I Have re-editted the post. Sorry.

stefanscherzinger, 2023-07-13:
@ZZWang21 

I'm not completely sure what's happening, but there are some parts in your launch file that look suspicious to me:

1. Using both gazebo *and* the `ros_control_boilerplate` seems wrong. I would drop these lines from your config:
   ```xml
    <node name="sim_hardware_interface" pkg="ros_control_boilerplate" type="sim_hw_main" output="screen" launch-prefix="$(arg launch-prefix)">
    <!-- Control motion and compliance controller with one handle -->
            <remap from="my_motion_control_handle/target_frame" to="target_frame" />
            <remap from="arm_controller_compliance/target_frame" to="target_frame" />
            <!-- Control wrenches via one topic -->
            <remap from="arm_controller_force/target_wrench" to="target_wrench" />
            <remap from="arm_controller_compliance/target_wrench" to="target_wrench" />
    </node>
   ``` 

2. The remappings from above might, however, be important. Let's check if your `arm_controller_handle` is connected to your `arm_controller_compliance`. When your stack is running, could you paste the output of  `rostopic list | grep target` here? Both handle and controller should communicate via some `target_frame` topic.

3. You should change `ft_sensor_ref_link: "wrist_3_link"` to `ft_sensor_ref_link: "tool0". That's where the sensor is in the UR robots.

ZZWang21, 2023-07-14:
Hi @stefanscherzinger 

Thanks for your reply. 

I editted the launch file and the yaml file as you suggested. 

currently, the Gazebo model is the same as before, it is not moving in compliance mode. 

The output of  "rostopic list | grep target" is :"

/arm_controller_compliance/target_frame
/arm_controller_force/target_wrench
/my_cartesian_motion_controller/target_frame
/target_wrench
"

I also tried compliance_force_controller, I set force_z as -100, the robot is moving, but the measured force is very small. I will keep tuning the parameters, if you have any ideas, please let me know. Thank you very much!

ZZWang21, 2023-07-14:
I tried force_controller, set target_wrench to -90N, but the actual is -25N, see pic: 
[image]

the controller.yaml is :"
arm_controller_force:
        type: "position_controllers/CartesianForceController"
        end_effector_link: "tool0"
        robot_base_link: "base_link"
        ft_sensor_ref_link: "tool0" #wrist_3_link
        joints:
          - shoulder_pan_joint
          - shoulder_lift_joint
          - elbow_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint

        solver:
          error_scale: 0.7
          iterations: 1
          forward_dynamics:
              link_mass: 1.0  # Higher values decrease oscillations in fully stretched configurations
                               #error_scale: 0.5

        pd_gains:
          trans_x: {p: 1,d: 0.05} #p 0.05
          trans_y: {p: 1,d: 0.05}
          trans_z: {p: 1,d: 0.05}
          rot_x: {p: 1.5}
          rot_y: {p: 1.5}
          rot_z: {p: 1.5}
          
    arm_controller_compliance:
        type: "position_controllers/CartesianComplianceController"
        end_effector_link: "camera_link"
        robot_base_link: "base_link"
        ft_sensor_ref_link: "toool0" #wrist_3_link
        compliance_ref_link: "camera_link" #ORI: tool0, wrist_3_link 
        target_frame_topic: "target_frame"
        joints:
          - shoulder_pan_joint
          - shoulder_lift_joint
          - elbow_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
          
        stiffness:  # w.r.t. compliance_ref_link
           trans_x: 800 #200
           trans_y: 800
           trans_z: 890
           rot_x: 40 #20
           rot_y: 40
           rot_z: 40
        solver:
           error_scale: 3.2 #0.2
           iterations: 1
        pd_gains:
           trans_x:
             p: 0.5 #0.5
             d: 0.015
           trans_y:
             p: 0.5
             d: 0.015
           trans_z:
             p: 0.5
             d: 0.015
           rot_x:
             p: 3 #3
             d: 0.01
           rot_y:
             p: 3
             d: 0.01
           rot_z:
             p: 3
             d: 0.01"

stefanscherzinger, 2023-07-14:
Regarding [this post](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/135#issuecomment-1635238276):
Thanks for the output. I see that the `target_frame` topic is not connected. Try setting this in your launch file:
```yaml
arm_controller_handle:
    type: "cartesian_controllers/MotionControlHandle"
    end_effector_link: "tool0"
    robot_base_link: "base_link"
    target_frame_topic: "/arm_controller_compliance/target_frame"   #  "/my_cartesian_motion_controller/target_frame"

   # ...
```
You should then be able to steer the robot with the handle.

Note: I usually leave the `target_frame_topic` of the controller configurations out and use remapping to steer all of my motion-related controllers with a single handle. That's what the `ros_control_boilerplate` re-mappings achieved. That would be something like

```yaml
       <remap from="arm_controller_handle/target_frame" to="target_frame" />
       <remap from="arm_controller_compliance/target_frame" to="target_frame" />
```

---

Regarding [this post](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/135#issuecomment-1635292770):
That's interesting. Intuitively, I would also expect `-90 N` for the force measurement. Could you change this in the force controller's config and try again?
```yaml
arm_controller_force:
type: "position_controllers/CartesianForceController"
end_effector_link: "tool0"
robot_base_link: "base_link"
ft_sensor_ref_link: "tool0" #wrist_3_link

# This is new
hand_frame_control: False  #  The commanded target wrench is now with respect to the robot's base link
```

I'm also confused by the negative sign. I would expect `+90 N` to drive into the table with `tool0` as sensor frame and end-effector frame. Could you plot the sensor's other measurements?

ZZWang21, 2023-07-15:
Thanks for your reply!
1. I modified the force_controller in yaml:"
     arm_controller_force:
        type: "position_controllers/CartesianForceController"
        end_effector_link: "tool0"
        robot_base_link: "base_link"
        ft_sensor_ref_link: "tool0" 
        joints:
          - shoulder_pan_joint
          - shoulder_lift_joint
          - elbow_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
        # This is new:
        hand_frame_control: False "

    and the force measurement is like follows:
[image]

2. for the compliance controller and handle. 
    I have modified the code according to your suggestions. 
    - the robot is not moving under compliance_controller. 
    - the robot is not moving using the handle. 
         the output of "rostopic list | grep target": 
             /arm_controller_compliance/target_frame
  /arm_controller_compliance/target_wrench
  /arm_controller_force/target_wrench
  /target_wrench "

        when I tried to move the robot under "handle", it showed error:"[ERROR] [1689406442.879355286, 75.219000000]: 
Invalid Trajectory: start point deviates from current robot state more than 0.01
joint 'wrist_2_joint': expected: -1.55639, current: -1.57367"
        and, I have a question, in RVIZ, the controller controls "arm" is the "arm_controller", but not "handle". 
        How to control the robot with "handle?". 
[image]

       Thanks.

stefanscherzinger, 2023-07-16:
Concerning **1.**:
Could you explain a little more what force-torque sensor you are using in Gazebo? I haven't used that in a while and there was nothing that really worked for me back then. I'm wondering why there's such a big difference.

Concerning **2.**:
That seems to be mixed-up with Moveit somehow. Make sure that you have both the `arm_controller_compliance` and the `arm_controller_handle` loaded via the `controller_manager` (service call). Check with
```bash
rosservice call /controller_manager/list_controllers
```
and print the output here. Both are controllers and need to be active so that you can steer the robot with the handle. I would also recommend using a different RViz configuration, such as [this one](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/blob/master/cartesian_controller_examples/etc/examples.rviz) for debugging. Not the Moveit one.

> when I tried to move the robot under "handle", it showed error:"[ERROR] [1689406442.879355286, 75.219000000]:
Invalid Trajectory: start point deviates from current robot state more than 0.01

That's a common error message from Moveit. It has nothing to do with the compliance controller or its handle. The handle should look [like this](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2/cartesian_controller_handles#cartesian-controller-handles) instead (That's a link to ROS2 documentation, but it helps clarify how the mechanism is meant to be used. I think you are erroneously using Moveit's handle in their RViz window.

ZZWang21, 2023-07-17:
1. for the FT sensor, I am using a ROS plug-in called "ft-sensor" - plugin filename="libgazebo_ros_ft_sensor.so" . I think it is commonly used in ROS UR5 application. And in my case, I think the sensor is working since when I move the robot with position (not compliance controller), the force can reach -90N or even more. It seemed is that the compliance controller is not working well in my model, maybe it is the parameter problem, but I tuned the parameters these days, the force measurement never met the set target using compliance controller or force controller. 

2.  The output of "rosservice call /controller_manager/list_controllers":
 
      "- 
    name: "arm_controller"
    state: "stopped"
    type: "position_controllers/JointTrajectoryController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::PositionJointInterface"
        resources: 
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
  - 
    name: "joint_position_controller"
    state: "initialized"
    type: "ur5_robotiq_85_simulation/JointPositionController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::PositionJointInterface"
        resources: 
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
  - 
    name: "gripper_controller"
    state: "running"
    type: "position_controllers/JointTrajectoryController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::PositionJointInterface"
        resources: 
          - simple_gripper_gripper_finger1_joint
  - 
    name: "arm_controller_force"
    state: "running"
    type: "position_controllers/CartesianForceController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::PositionJointInterface"
        resources: 
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
  - 
    name: "joint_state_controller"
    state: "running"
    type: "joint_state_controller/JointStateController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::JointStateInterface"
        resources: []
  - 
    name: "arm_controller_compliance"
    state: "initialized"
    type: "position_controllers/CartesianComplianceController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::PositionJointInterface"
        resources: 
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
  - 
    name: "arm_controller_handle"
    state: "running"
    type: "cartesian_controllers/MotionControlHandle"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::JointStateInterface"
        resources: []
"

For the RViz, I think I understand what you mean, the RViz should be like the RViz of the controller_examples one. I would try modify the rviz file later and see what going on.

ZZWang21, 2023-07-17:
I am thinking of implementing the controller_examples.rviz to ur5 model, but it seems it is a bit tricky. since in controller_examples, only rviz is moving, and there is no Moveit planning. But in my case, RVIZ is running along with GAZEBO and there is MOVEIT planning in RViz. I am not sure if it will work.  Should I try without moveit?

ZZWang21, 2023-07-18:
Sorry it is just weird, I used force_controller, the robot at least can move, but when I switch to compliance_controller, the robot just stopped moving... why?

ZZWang21, 2023-07-18:
Sorry to disturb, but in cartesian_controller, if I set the tran_z and solver to very big, for example, p in tran_z to 5, gain in solver to 3, the robot will rush into the table, and then drift away, I tried to tune the parameters but till now it has no satisfactory movement. Thanks.

stefanscherzinger, 2023-07-18:
> Sorry it is just weird, I used force_controller, the robot at least can move, but when I switch to compliance_controller, the robot just stopped moving... why?

The `cartesian_compliance_controller` implements a virtual stiffness that relates offsets from its `target_frame` with forces and torques. That means whenever you activate the compliance controller, it will feel compliant about its `compliance_ref_frame` and will try to get back to it after external disturbances. When sending force-torque commands over the `target_wrench` topic, then you temporarily deviate the controller from its equilibrium pose. How much the offset is depends on your setting of the translational and rotational stiffness.

>  for example, p in tran_z to 5, gain in solver to 3, the robot will rush into the table, and then drift away,

I think you observe instability here. Try lowering the controller gains and start from the recommended default ones. We did still not clarify why you experience the offset to your reference force you mentioned [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/135#issuecomment-1635292770). That won't go away by increasing the controller gains. There must be some other reason that's causing the offset. You were working with the force controller for that experiment, right? Because the compliance controller would behave as I just described.

ZZWang21, 2023-07-19:
So you mean in compliance_controller, the robot will not move. But in the controller_examples, when I applied compliance_controller, the end-effector moved a bit. I am now sure if this is normal:

https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/assets/55466401/307b510e-844d-46f4-86b2-164a171d2691

According to your description, I think I can move the robot with force_controller and when the robot touches the surface, change to the compliance_controller. I will try this later today. 

But I still do not know how to move robot under compliance_controller. Since I am using Moveit to plan trajectory, I cannot use the RViz like the "controller_examples". When using "motion_control_handle", I do not know how to code the movement. Should I try joint_to_trajectory_controller?

ZZWang21, 2023-07-21:
I am trying to use "joint_to_cartesian_controller" instead of "handle", but I got error like this:

"ERROR] [1689913355.526819980, 36.034000000]: Exception thrown while initializing controller 'my_joint_to_cartesian_controller'.
Cannot enforce limits for joint 'shoulder_pan_joint'. It has no velocity limits specification.
[ERROR] [1689913355.531104066, 36.035000000]: Initializing controller 'my_joint_to_cartesian_controller' failed"

I checked the limits, I think the velocity limits are there.

stefanscherzinger, 2023-07-21:
@ZZWang21 

Could you open a new issue for [your last post](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/135#issuecomment-1644969291)?
This issue's topic is about the *handle*. Let's not mix too much. That's easier for others who follow these conversations.

stefanscherzinger, 2023-07-21:
Regarding [this post](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/135#issuecomment-1641392679):
> So you mean in compliance_controller, the robot will not move.

Yes, the robot will move! It moves with `dx = f / cx`.
`dx` is the amount it moves
`f` is force in that direction (either from `target_wrench` or from `ft_sensor_wrench`)
`cx` is stiffness specified in the compliance controller for that direction.

ZZWang21, 2023-07-31:
Sorry, but I also tested on the real robot yesterday, that using compliance controller, the robot was not moving. I am not sure if it is the Ubuntu version problem? I used ROS MELODIC before, the controller can be used on real robot. Now I am using ROS NOETIC, but it is not working.    

AND, when I used the compliance controller in simulation, instead of moving towards the target, the robot moved backwards, moved away from the target surface. I am not sure where the setting is wrong or something else went wrong. Sorry about the issue, but I would like to make it work.

ZZWang21, 2023-08-02:
I tried to ignore the compliance_controller problem and trying to use joint_to_cartesian_controller+compliance_controller to move the robot in Gazebo. However, I got error like this:

"Can't accept new action goals. Controller is not running.
[ WARN] [1690951248.614989374, 323.472000000]: Controller 'arm_controller' failed with error INVALID_GOAL: 
[ WARN] [1690951248.615583555, 323.472000000]: Controller handle arm_controller reports status FAILED
"

I used python code to set the moving target, and the MoveGroup is "arm" and the "arm" group was controlled by controller "arm_controller". This was set up in Moveit. I do not know how to switch it to joint_to_cartesian_controller. 

python: 
“rospy.init_node("moveit_cartesian_path", anonymous=False)

        rospy.loginfo("Starting node moveit_cartesian_path")

        rospy.on_shutdown(self.cleanup)

        # Initialize the move_group API
        moveit_commander.roscpp_initialize(sys.argv)

        # Initialize the move group for the ur5_arm
        self.arm = moveit_commander.MoveGroupCommander('manipulator')”

Could anyone help? Thanks.

ZZWang21, 2023-08-05:
I tried to modify the moveit settings, changed the controller of move_group 'arm' from 'arm_controller' to 'joint_to_cartesian_controller/arm_controller'. Then, I tried to control the robot with joint_to_cartesian_controller+cartesian_motion_controller+joint_state_controller, just as the example. But the robot did not move in GAZEBO. Could anyone help me out here? Thanks.

ZZWang21, 2023-08-05:
I think the problem is that the "arm_controller" under group ns "joint_to_cartesian_controller" can receive the actual positions, target positions, error, but it cannot move the robot. The controller is correctly set. I am not sure why. Still checking.

ZZWang21, 2023-08-06:
OK, after two days debug, I kind of sort this out. Now I can use "joint_to_cartesian_controller" to move the robot. Next step, I will use "joint_to"+compliance controllers to control the robot.

ZZWang21, 2023-08-08:
I tried set target_wrench to -30N, and use compliance+joint_to controllers to control the UR5 in Gazebo with Moveit. But the actual measurement is -60N around. I tried tune the parameters, but no change. 

[image]

ZZWang21, 2023-08-09:
I tuned the parameters of compliance controller again, and it seems better but the control is not there.
target: -100N:
[image]

target: -70N:
[image]

ZZWang21, 2023-08-10:
Sorry to disturb you, but @stefanscherzinger  @matthias-mayr , could you help me here? Any suggestions? Thanks.

stefanscherzinger, 2023-08-10:
@ZZWang21 Thanks for sticking to this and for reporting your progress. Unfortunately, I'm confused about the results and unsure about where you got stuck. Could you answer the following questions?

1. Were you successful in moving the robot with Moveit + *joint_to_cartesian_controller* + *compliance controller* in Gazebo?
2. Were you successful in moving the robot with the *compliance controller* and the controller's handle?
3. When commanding a `target_wrench` to the compliance controller, is your robot in contact with the surface?
4. What does your force-torque sensor measure when in compliance control and you command `target_wrench = 0`?

Concerning both plots from [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/135#issuecomment-1670688137): That's not what I would expect. I'm particularly interested in your answer to my question `4.`.

ZZWang21, 2023-08-11:
Thanks for your reply, @stefanscherzinger .

    1. No.  I can move with Moveit + joint_to + motion_controller in Gazebo, but **not** Moveit+joint_to+compliance in Gazebo. 
    2. No, I cannot move with compliance+motion_control_handle. In RVIZ, I cannot move the robot with the interactive marker when the compliance_controller is in use. 
    3. Yes, as you can see in the last post, the wrench/force is -70N, it was contacting the surface. 
    
[image]

    5. I set the target_wrench from -10N to 0N with compliance controller. The force_z measurement is like this: (it seemed like the force is not changing)
    
[image]

ZZWang21, 2023-08-13:
Sorry to disturb, I am also trying to implement the compliance controller on UR5e real arm. But I cannot realize it, the robot is not moving as it should be. I am using ROS NOETIC. Ubuntu 20.04.

The ROS bringup package is as link:  https://github.com/UniversalRobots/Universal_Robots_ROS_Driver 

and the moveit setup:   roslaunch ur5e_moveit_config moveit_planning_execution.launch 
RVIZ:                                roslaunch ur5e_moveit_config moveit_rviz.launch config:=true

But the robot is not moving as compliance controller. I realized this before in ROS Melodic Ubuntu 18 and old package. But now it is not working. Please also help if possible. 

Thanks.

ZZWang21, 2023-08-14:
I tried to log the measured force and target_wrench in Gazebo, it is like this:
[image]

ZZWang21, 2023-08-15:
another thing to mention, when using force_controller, it can be seen that the joint states are moving in the rqt window, but when using compliance controller, the joints are stable, which is not the case in "cartesian_controller_examples". 
[image]

ZZWang21, 2023-08-16:
Plus, I found that when I use "joint_to"+"motion" controller, the robot moved itself maybe due to gravity, in Gazebo. Has anyone noticed that? Sorry to disturb, but @stefanscherzinger @matthias-mayr  could you share some help.. Thank you very much.

stefanscherzinger, 2023-08-16:
@ZZWang21 

I'm afraid most of what you experience is because the compliance controller is not connected correctly. The motion that you plan with Moveit seems to get lost somewhere in the chain from `Moveit` to `joint_to_cartesian_controller` to `compliance_controller`.
That's also the case why your force-torque measurements don't change [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/135#issuecomment-1674203065) nor [here](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/135#issuecomment-1676664091).

The first thing that we should fix is that you can move the robot in free space with Moveit + `joint_to_cartesian_controller` + `compliance_controller`. Leave the contacts aside for now until we really have a working pipeline.

You seem to be close, though, since you managed to combine Moveit with the `cartesian_motion` controller. The remapping for the `cartesian_compliance_controller` should be very similar. Both controllers follow the moving target that the `joint_to_cartesian_controller` builds from the incoming Moveit trajectory. Could you again check the difference in the configuration between the two controllers and why one works while the other doesn't?

ZZWang21, 2023-08-17:
@stefanscherzinger  Thanks for your reply! You are absolutely right! The first target to get joint_to+compliance+Moveit right! 

I will check the remapping and the config for the "cartesian_compliance_controller". I will update later.

ZZWang21, 2023-08-17:
After check, I think I do not have /target_frame rostopic originally. But in the "cartesian_controller_examples", it has target_frame and target_wrench, automatically. But since I do not have target_frame, I am now remapping the /compliance/target_frame to /joint_to/target_frame, to see if it works.

ZZWang21, 2023-08-18:
I remapped the motion_controller and compliance_controller's target frame to joint_to's target frame, now the robot cannot move with joint_to+motion. It seemed like the problem is the target_frame rostopic. But I do not have a target_frame rostopic, in the example, the topic is published via rviz? I am not sure how to publish a target_frame rostopic, please help. Thanks. @stefanscherzinger

ZZWang21, 2023-08-18:
I tried several places to put the remap code, but the "target_frame" and "target_wrench" rostopics seemed like not published. 

" <node name="spawn_urdf" pkg="gazebo_ros" type="spawn_model" args="-param robot_description -urdf -model ur5 -x 0 -y 0 -z 0.6" >
        <remap from="my_cartesian_motion_controller/target_frame" to="target_frame" />
        <remap from="arm_controller_compliance/target_frame" to="target_frame" />
                
        <remap from="arm_controller_force/target_wrench" to="target_wrench" />
        <remap from="arm_controller_compliance/target_wrench" to="target_wrench" />    
    </node>" 

I do not know where is the problem. the info of rostopic "my_cartesian_motion_controller/target_frame", is publisher: /gazebo, subscriber: /gazebo.

ZZWang21, 2023-08-19:
Sorry, I tried to move the "remap" part code to everywhere in the launch file, it just did not do the job. /target_frame rostopic is just not there...

ZZWang21, 2023-08-20:
After check, I found the compliance/target_frame was subscribed by /gazebo rosnode. In the /gazebo rosnode, remap all the */target_frame to the /target_frame and it worked. I also did this to all the */target_wrench. The next target is to control force target. 

[image]

ZZWang21, 2023-08-21:
I checked the performance of the compliance controller in Gazebo. When I applied target_wrench -20N, it leaves the surface. When I applied target_wrench to 20N, it touches the surface, measured force range: -5N ~ -100N, but the measured force is very unstable. Later I will upload the curve. I tuned the parameters but seemed like no improvement.

ZZWang21, 2023-08-22:
I updated the controller config, and managed to move the robot with force_controller. then I tried the compliance controller, but the control is not ideal. target is -20. the actual is -100 to 10 N. 

[image]

ZZWang21, 2023-08-25:
currently, the compliance controller is working, but the control on small target_wrench is not good. for example, if the target_wrench is set to -15N, the measured force in Gazebo is like this: 
[image]

I tuned all of the parameters in rqt, but till now, no good results.

ZZWang21, 2023-08-28:
I switched to another laptop,

Old laptop: vmware Ubuntu 20.04

New laptop: Ubuntu 20.04 and Windows dual system. 

I restored the system to be the same, But the control.yaml should be modified. I have to tune the parameters again.

ZZWang21, 2023-09-03:
I tried to update the code and managed to realise the compliance+joint_to controllers control. it was moving, but the pose of the end effector is not fixed. Why? I have set the pose fixed in the python code. See the pics before moving and after moving. 
before: 
[image]
after:
[image]
Thank you. @stefanscherzinger

ZZWang21, 2023-09-03:
And, I am trying to apply this on real UR5 robot. When applying compliance+joint_to (I unloaded both of them), the robot moved randomly. I tried to control the robot with each of them, they were fine. Just does not work with the "Compliance+Joint_to" combination. Why?

stefanscherzinger, 2024-01-10:
Please reopen when having more insights/suggestions.

closed 2024-01-10 by stefanscherzinger