fzi-forschungszentrum-informatik/cartesian_controllers #142: Warning: Uncontrolled movement with UR's latest `Humble` release

stefanscherzinger, 2023-08-10:
**Problem description**
There seems to be a problem with the cartesian controllers and the latest UR ROS2 driver. After upgrading today from
```bash
> $ cat /var/log/dpkg.log | grep ros-humble-ur-                                                                                                                                                                             
2023-08-10 14:21:59 upgrade ros-humble-ur-robot-driver:amd64 2.2.6-1jammy.20230327.202833 2.2.8-1jammy.20230722.001651
2023-08-10 14:21:58 upgrade ros-humble-ur-client-library:amd64 1.3.1-1jammy.20221201.233731 1.3.2-1jammy.20230714.001128
```

the `cartesian_force_controller` completely freaked out upon activation. I'm currently not sure what's happening, but I could pinpoint it down a little. After removing the system dependencies and after installing from source
- [Universal_Robots_ROS2_Driver - 2.2.7](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/releases/tag/2.2.7)
- [Universal_Robots_Client_Library - 1.3.2](https://github.com/UniversalRobots/Universal_Robots_Client_Library/releases/tag/1.3.2)

everything behaves normal.

So, something happens in the driver between `2.2.7` and `2.2.8` whose consequence I need to check in the controllers. The only commit that seems to touch relevant parts of the code was introduced with [#719](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/719). Let's see what that is..

**Software versions**
 - **Commit**: a7bbee50faf4ef3fed5ef2fe2857683fb15d14d3
 - **OS**: Ubuntu 22.04
 - **ROS version**: ROS2 Humble, ros-humble-ur-robot_driver 2.2.8, ros-humble-ur-client-library 1.3.2
 - **Robot**:  UR5e

Daan0103, 2024-01-23:
When building  this 'safe' versions from source in ROS2 humble, I get errors. How to use these cartesian controller safe on the UR5e?

```
/ros2_ws/src/Universal_Robots_ROS2_Driver-2.2.7/ur_controllers/src/scaled_joint_trajectory_controller.cpp:160:7: error: ‘traj_point_active_ptr_’ was not declared in this scope
  160 |   if (traj_point_active_ptr_ && (*traj_point_active_ptr_)->has_trajectory_msg()) {
      |       ^~~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:223: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [4.69s, exited with code 2]

Summary: 10 packages finished [10.6s]
  1 package failed: ur_controllers
  1 package had stderr output: ur_controllers
  5 packages not processed
```

shrutichakraborty, 2024-04-22:
Hi @stefanscherzinger, is this still the case? I have been experiencing issues with the cartesian motion controller where the robot vibrates upon reaching commanded position, and sometimes even experiences random jerks along the motion, is this related to this issue or could that be fixed by tuning the controller? Any ideas would be helpful thanks.

yg19930918, 2024-06-18:
I just want to report I met the same situation using ROS iron/ur ros driver 2.3.7/UR5e. It is not only cartesian_force_controller, cartesian_motion_controller is also completely freaked out when I try to activate the controller. I did tried to adjust the iteration number and p/d, it did not work.

yg19930918, 2024-06-28:
A small update, after downgrading my ROS and ur-driver version following  @stefanscherzinger advice， the compliance controller works. Thanks for your open-source controller, this is quite helpful for me.