fzi-forschungszentrum-informatik/cartesian_controllers #144: Using my_cartesian_motion_controller in gazebo, the UR5 joints start to move when activate my_cartesian_motion_controller.

caimingxue, 2023-09-08:
Using my_cartesian_motion_controller in gazebo, the UR5 starts to move when activate my_cartesian_motion_controller. However, I don't send any message. The pose of UR5 do not change. However, some joints are moving. Here is the video.

https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/assets/45029643/52fe0137-358a-4ab9-b783-e670cf603930

stefanscherzinger, 2023-09-21:
Hi @caimingxue 

Thanks for the video. I can see the drift. Does this happen on the newest *master* branch? What's the output of `git rev-parse HEAD` in a sourced terminal for the `cartesian_controllers`?

stefanscherzinger, 2023-10-20:
Please reopen if this is still an issue.

closed 2023-10-20 by stefanscherzinger