fzi-forschungszentrum-informatik/cartesian_controllers #151: [Cartesian Compliance Controller] Massive issue with controller stability

firesurfer, 2023-10-05:
**Problem description**

We are currently trying to properly setup the cartesian compliance controller in a setup where we need a rather fast response to a changing force applied to the end effector.

In order to track a change faster we need to increase the gains - unfortunately the controllers becomes unstable quite easily. I was hoping to gain some further insight into tuning the controller as I am kind of having a hard time to understand to exact influence of all the parameters on the controller behavior from the documentation. 

**Software versions**
 - **Commit**: 03057bde4e1b04ff59e9bc1393f3fb47db2548e2
 - **OS**:  Ubuntu 22.04
 - **ROS version**:  Ros2 Iron - binary install
 - **Robot**: UR16e

**Information about the setup**

Entry in the controller.yaml
Basically we have a virtual center point in our gripper where the wrench will be applied to later on. The y axis is the one that should be compliant in the end. 
The hardware setup consists of an UR16e with a robotiq 2f-85 gripper.  The arm is connected via a thread with another moving part in direction of the y axis. 
The controller runs at 500Hz
```
cartesian_compliance_controller_gripper_center:
  ros__parameters:
    end_effector_link: "gripper/center"
    robot_base_link: "base_link"
    ft_sensor_ref_link: "tool0"
    compliance_ref_link: "gripper/center"
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint

    # Choose: position or velocity.
    command_interfaces:
      - position

    stiffness:  # w.r.t. compliance_ref_link
        trans_x: 5000.0
        trans_y: 20.0
        trans_z: 5000.0
        rot_x: 500.0
        rot_y: 500.0
        rot_z: 500.0

    solver:
        error_scale: 0.5
        iterations: 2

    pd_gains:
        trans_x: {p: 0.011, d: 0.0001}
        trans_y: {p: 0.011, d: 0.0001}
        trans_z: {p: 0.011, d: 0.0001}
        rot_x: {p: 0.3}
        rot_y: {p: 0.3}
        rot_z: {p: 0.3}
```

When we now apply a force in the y-direction it takes the controller quite some time to track the change in force. (Because the arm follows very slow).
In order to solve this we tried to increase the gains. But as soon as one goes like above p: 0.019 the controller will start to vibrate. The vibration / instability is always in the x and z axes. (That do not get any force applied on)

Things we tried to improve the stability:

1. Tune p gain until a becomes slightly unstable -> add d gain. Works only if arm does not move
2. Increase stiffness in x and z axis. If I understand this correctly this should reduce the influence of the force sensor inputs
3. Set the `link_mass` correctly - couldn't see any difference so far
4. Filter the sensor values - made it worse
5. Decrease the controller frequency to 100Hz - made it worse
6. Playing around with the iterations -> Did not seem to change anything

**Expected behavior**

I would expect the controller be stable if I set the stiffness rather high, and to track the position accurately if I set the gains for these axes high enough. 

Let me know if you need more information - I hope I described our setup well enough in order to understand what is going on.

**Additional context**
Add anything that you feel is relevant to tackle this issue.

matthias-mayr, 2023-10-24:
When I used the controller this summer in the ROS 1 branch, I found it necessary to apply a filter on the FT signals to achieve stability while having gains that allowed for a reactive behavior.

After some tuning I ended up in a configuration that gave me pretty stable contact, no vibrations and reasonable, but not super fast motion tracking. I'll add it to my to-do list to put the filter into a gist.

FYI, there is also an [admittance controller in the ROS2 control repository](https://github.com/ros-controls/ros2_controllers/tree/master/admittance_controller), but I have not used it and am not aware about the features.

firesurfer, 2023-10-31:
Thanks for your answers. We actually found some parameters which work stable (even though not as reactive as we would wish for) . As I said I think in our case the main issue is that we have two moving components tied together. 

@matthias-mayr I already tried a simple lowpass filter on the FT signals but it actually made it worse. But that makes sense in our case at it will reduce the control bandwidth. 

We will definitely come back to this in the future.

stefanscherzinger, 2024-01-10:
Please reopen if still relevant or when having new insights/suggestions.

closed 2024-01-10 by stefanscherzinger